zephyrproject-rtos/zephyr #80234: Nominate alexsven as collaborator

aescolar, 2024-10-22:
# Nomination

## GitHub User

1. Alexander Svensen
1. [alexsven](https://github.com/alexsven)
1. Nordic Semi

## Supporting Documents

* https://github.com/zephyrproject-rtos/zephyr/pull/79505
* https://github.com/zephyrproject-rtos/zephyr/pull/78130
* https://github.com/zephyrproject-rtos/zephyr/pull/69085
* https://github.com/zephyrproject-rtos/zephyr/pull/68611
* https://github.com/zephyrproject-rtos/zephyr/pull/67835

stephanosio, 2024-10-22:
@alexsven is currently not listed as a collaborator in the MAINTAINERS file.

aescolar, 2024-10-22:
@stephanosio you are right, and his contributions are quite spread around zephyr overall. I'll close this until this changes.

closed 2024-10-22 by aescolar