zephyrproject-rtos/zephyr #80241: Telnet shell sporadically locks up on client disconnect

madscientist159, 2024-10-22:
**Describe the bug**
#69612 converted the Telnet shell to use network sockets, and inadvertently introduced a race condition on client disconnect.

The socket API is capable of generating an error condition for a single network "fault" (client disconnect) in two paths, one synchronous and one asynchronous.   The Telnet shell listens to the asynchronous path in `telnet_server_cb()`, and the synchronous path in `telnet_command_send_reply()` / `telnet_send()` / `telnet_recv`.

As a result of these two paths, a race condition exists.  If the timing of a client disconnect and subsequent reconnect is exactly right, the following sequence is observed:

1.) An active `telnet_send()` / `zsock_send()` is interrupted (`Failed to send 128, shutting down`)
2.) The fd is reset (`fd = -1`) in `telnet_end_client_connection()`
3.) A new client is accepted (`fd` is set back to a valid value, typically the same value as that used by the disconnected clien)
4.) `telnet_server_cb()` finally fires for the first disconnect (`Telnet socket 8 event 8 error (104)`), calling `telnet_end_client_connection()` and _partially_ shutting down the just-accepted new connection.

The Telnet shell is now in an unrecoverable state.  It accepts connections but will pass no traffic as the shell thinks there is no active connection, nor will it reset its faulty internal state on disconnect for the same reason.

**To Reproduce**
Set up an example to log lots of text.  Repeatedly connect and disconnect using the standard Linux `telnet` client until the lockup is observed.

**Expected behavior**
The Telnet shell should always allow disconnect and reconnect without locking up.

**Impact**
Remote shell access is unusable in practice due to the chance of needing a system reboot after disconnect.

**Environment (please complete the following information):**

 - OS: Debian Bookworm
 - Toolchain: GCC 12
 -  Zephyr 3.7

jukkar, 2024-10-22:
I wonder if this is related to this fix #78976

madscientist159, 2024-10-22:
#78976 does not fix the problem, that was one of the first patches I applied before diving deeper and finding the race condition above.

jukkar, 2024-10-24:
I have been trying to replicate the issue but it seems to be very difficult. @madscientist159 do you possibly use a script in client side to trigger the issue?

madscientist159, 2024-10-28:
@jukkar Unfortunately I have been unable to reproduce the issue in a scripted environment despite trying a few options.  I have sore fingers from `Ctrl+]` + 'quit` in the `telnet` client :wink: 

I suspect the `telnet` client interrupts the connection in a way that e.g. the Python telnet library will never do, e.g. while data transmission is in process.  The target device CPU speed probably plays a role too, so if you can crank the CPU speed down or otherwise load the CPU with other work the race condition I identified above will be easier to hit.

jukkar, 2024-10-30:
I managed to see similar errors although the telnet stayed fully functional after the issue.
Steps:
In Linux shell
```
while [ 1 ]; do telnet 192.0.2.1 <<EOF
net stats
EOF
done
```
Then I am seeing this in zephyr console:
```
[00:16:36.096,923] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.101,440] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.106,109] <err> shell_telnet: Telnet accept error (-113)
[00:16:36.108,917] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.114,471] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.121,246] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.125,427] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.129,791] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.134,338] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.138,610] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.141,845] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.148,468] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.155,059] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.159,454] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.163,970] <err> shell_telnet: Failed to send 128, shutting down
[00:16:36.166,046] <err> shell_telnet: Network disconnected, shutting down
[00:16:36.168,151] <err> shell_telnet: Failed to send command -1, shutting down
[00:16:36.172,607] <err> shell_telnet: Failed to send command -1, shutting down
```
After stopping the script, I could still connect to the device using `telnet 192.0.2.1`.

All this with latest upstream (4.0-rc1) with USB NCM profile.
I try next with 3.7.

carlescufi, 2024-11-13:
@madscientist159 should we close this issue, given how it doesn't seem to be reproducible on the maintainers' machines?

carlescufi, 2024-12-04:
Closing due to reporter's inactivity

closed 2024-12-04 by carlescufi