zephyrproject-rtos/zephyr #80251: Test tests/subsys/llext/simple on mps2/an385 fails

tbursztyka, 2024-10-22:
**Describe the bug**
The test ends with a fatal error.

**To Reproduce**
1. west build --board mps2/an385 tests/subsys/llext/simple
2. west build -t run

**Expected behavior**
Test should end successfully

**Impact**
It's a CI blocker on other PRs 

**Logs and console output**
```D: Linking ELF...
D: relocation section .rel.text (2) acting on section 1 has 13 relocations
D: relocation 2:0 info 0x1502 (type 2, sym 21) offset 88 sym_name ztest_relative_filename sym_type 0 sym_bind 1 sym_ndx 0
I: found symbol ztest_relative_filename at 0x88a1
I: writing relocation symbol ztest_relative_filename type 2 sym 21 at addr 0x20003858 addr 0x88a1
D: 2 20003858 88a1 ztest_relative_filename
D: relocation 2:1 info 0x802 (type 2, sym 8) offset 92 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x2000385c addr 0x20003b00
D: 2 2000385c 20003b00 
D: relocation 2:2 info 0x1602 (type 2, sym 22) offset 96 sym_name printk sym_type 0 sym_bind 1 sym_ndx 0
I: found symbol printk at 0x87a1
I: writing relocation symbol printk type 2 sym 22 at addr 0x20003860 addr 0x87a1
D: 2 20003860 87a1 printk
D: relocation 2:3 info 0x1702 (type 2, sym 23) offset 100 sym_name vprintk sym_type 0 sym_bind 1 sym_ndx 0
I: found symbol vprintk at 0xe61
I: writing relocation symbol vprintk type 2 sym 23 at addr 0x20003864 addr 0xe61
D: 2 20003864 e61 vprintk
D: relocation 2:4 info 0x802 (type 2, sym 8) offset 104 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x20003868 addr 0x20003b00
D: 2 20003868 20003b00 
D: relocation 2:5 info 0x1802 (type 2, sym 24) offset 108 sym_name ztest_test_fail sym_type 0 sym_bind 1 sym_ndx 0
I: found symbol ztest_test_fail at 0x1fd5
I: writing relocation symbol ztest_test_fail type 2 sym 24 at addr 0x2000386c addr 0x1fd5
D: 2 2000386c 1fd5 ztest_test_fail
D: relocation 2:6 info 0x1902 (type 2, sym 25) offset 236 sym_name test_entry sym_type 2 sym_bind 1 sym_ndx 1
I: writing relocation symbol test_entry type 2 sym 25 at addr 0x200038ec addr 0x20003871
D: 2 200038ec 20003871 test_entry
D: relocation 2:7 info 0x1a02 (type 2, sym 26) offset 240 sym_name test_dependency sym_type 0 sym_bind 1 sym_ndx 0
I: found symbol test_dependency at 0x20002f41
I: writing relocation symbol test_dependency type 2 sym 26 at addr 0x200038f0 addr 0x20002f41
D: 2 200038f0 20002f41 test_dependency
D: relocation 2:8 info 0x802 (type 2, sym 8) offset 244 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x200038f4 addr 0x20003b00
D: 2 200038f4 20003b00 
D: relocation 2:9 info 0x802 (type 2, sym 8) offset 248 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x200038f8 addr 0x20003b00
D: 2 200038f8 20003b00 
D: relocation 2:10 info 0x802 (type 2, sym 8) offset 252 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x200038fc addr 0x20003b00
D: 2 200038fc 20003b00 
D: relocation 2:11 info 0x802 (type 2, sym 8) offset 256 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x20003900 addr 0x20003b00
D: 2 20003900 20003b00 
D: relocation 2:12 info 0x802 (type 2, sym 8) offset 260 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x20003904 addr 0x20003b00
D: 2 20003904 20003b00 
D: relocation section .rel.exported_sym (7) acting on section 6 has 2 relocations
D: relocation 7:0 info 0x802 (type 2, sym 8) offset 0 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: writing relocation symbol  type 2 sym 8 at addr 0x200031c0 addr 0x20003b00
D: 2 200031c0 20003b00 
D: relocation 7:1 info 0x1902 (type 2, sym 25) offset 4 sym_name test_entry sym_type 2 sym_bind 1 sym_ndx 1
I: writing relocation symbol test_entry type 2 sym 25 at addr 0x200031c4 addr 0x20003871
D: 2 200031c4 20003871 test_entry
D: sym 0x20003871 name test_entry in 0x200030c4
D: loaded module, .text at 0x20003800, .rodata at 0x20003b00
I: Loaded extension export_dependen
E: ***** MPU FAULT *****
E:   Instruction Access Violation
E: r0/a1:  0x00000001  r1/a2:  0x00014b10  r2/a3:  0x0001499c
E: r3/a4:  0x00000166 r12/ip:  0x00000000 r14/lr:  0x000008a1
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x20003870
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200001e8 (test_inter_ext)
E: Halting system```

tbursztyka, 2024-10-22:
nevermind, mixed up command as it seems.

closed 2024-10-22 by tbursztyka

teburd, 2024-10-22:
yeah the test is a bit interesting, the testcase.yml really matters in setting things up correctly