zephyrproject-rtos/zephyr #80260: Documentation workflow stuck blocking PR CI completion

dleach02, 2024-10-22:
The documentation build workflow is stuck without a path to restart. 

Speculation is that it will be stuck if the PR has no reason to do documentation generation.

Behavior shows CI waiting on completion of all tasks but the documentation workflow has the following error:

```
Invalid workflow file: .github/workflows/doc-build.yml#L264
error parsing called workflow ".github/workflows/doc-build.yml" -> "./.github/workflows/ready-to-merge.yml" : failed to fetch workflow: workflow was not found.
```

See #80016 as a reference.

dleach02, 2024-10-22:
ping @nashif

kartben, 2024-10-22:
Looking at recent jobs it's only ever happened once on another PR https://github.com/zephyrproject-rtos/zephyr/actions/runs/11427402099
This is really odd and while I investigate, or at least keep an eye on future failed doc build jobs, please try retriggering CI on the linked PR. I am almost certain this will do the trick (looks like it did for the other PR I just mentioned since issue seems to have magically fixed itself for @faxe1008 as I didn't hear from him :))

kartben, 2024-10-22:
> Speculation is that it will be stuck if the PR has no reason to do documentation generation.

Something's off, for sure, but it's definitely not what's happening as *many* PRs in that case ran just fine, ex. https://github.com/zephyrproject-rtos/zephyr/actions/runs/11469657074.

dleach02, 2024-10-23:
> Looking at recent jobs it's only ever happened once on another PR [zephyrproject-rtos/zephyr/actions/runs/11427402099](https://github.com/zephyrproject-rtos/zephyr/actions/runs/11427402099) This is really odd and while I investigate, or at least keep an eye on future failed doc build jobs, please try retriggering CI on the linked PR. I am almost certain this will do the trick (looks like it did for the other PR I just mentioned since issue seems to have magically fixed itself for @faxe1008 as I didn't hear from him :))

I don't know how to retrigger the CI in this case. There is no "restart failed jobs" path.

kartben, 2024-10-23:
> I don't know how to retrigger the CI in this case. There is no "restart failed jobs" path.

Like I said in the linked PR force-push (after a rebase or, less disruptive, a simple commit amend) will typically do just that (and without dismissing existing reviews)

dleach02, 2024-10-30:
Discussed in release meeting that this may not be so broad of an issue. @kartben will continue to monitor.

closed 2025-01-13 by github-actions[bot]