zephyrproject-rtos/zephyr #80262: System tree

bjarki-andreasen, 2024-10-22:
## Introduction

This RFC provides a solution which completely separates software (drivers, subsystems, RAM/ROM layout etc.) from hardware (physical devices, interrupts, cores, etc.)

This solves:
* Removes need to define "software" properties in devicetree (like file systems, buffer sizes, partition layouts etc. etc.)
* Removes need to use devicetree to instanciate drivers and subsytems using node statuses, chosen nodes, compatibles, etc.
* Adds single source of truth for all things software, allowing for completely deterministic init orders for all software (drivers, subsystems etc.)

### Problem description

We currently have a really tight coupling between drivers and subsystems, and the devicetree, to the point where we continuously discuss the degree to which we want "software" configurations in the devicetree. On top of this, we only have Kconfig, a flat configuration language, to instantiate and configure instances of software components, which results in the devicetree being used to instanciate software components.

### Proposed change

We should create a tree like configuration in a simple to parse language, which allows for configuration of all software components, including device drivers.

For a hard and clear example of what this would change:
```
DT_INST_FOREACH_STATUS_OKAY()
```
would change to something like:
```
SYS_SCHEMA uart_nrfx_uarte
SYS_INST_FOREACH_STATUS_OKAY()
```
since the driver would not be instanciated based on the devicetree nodes status, but the inclusion of the driver (matched by scema) in the system tree:
```
uart0:
  schema: uart_nrfx_uarte
  dt-nodelabel: uart0
```

This provides full flexibility in defining the system tree schema just as we want, for example adding some buffer size to the specific driver instance of uart0:
```
uart0:
  schema: uart_nrfx_uarte
  dt-nodelabel: uart0
  rx-buf-size: 255
```

We can even create pure software components, in a tree like structure:
```
telnet_shell:
  schema: telnet_shell
  port: 21
  line-buf-size: 64
```

### Concerns and Unresolved Questions

This RFC is clearly an immense amount of work, to the degree that even though going with it would be fantastic IMO, finally adding that layer of separation between DT and software, while providing complete freedom for zephyr defined schema for "multi instance" definition and inclusion.

## Alternatives

Just abuse the devicetree and bindings as we do now, try to alleviate some pressure here and there with partial solutions, until the end of time :D

oooor, abuse the devicetree to the nth degree by adding software components in there, like shells, bluetooth HCI etc. (which we kind of do with chosen nodes), and make it a "system tree" with everything sort of related to hardware in there, using the strange and limited dt language and bindings.

decsny, 2024-10-22:
This is good, but what I would like to clarify is how this differs structurally from what has developed with what @fgrandel has put 
forward. Because, it seems to be mostly in line with the same idea, from the point of view of the configuration files, i.e.:
* it has tree like structure with nodes representing software instances or properties
* the nodes would have a schema property to indicate which schema to match to
* the configuration language can reference information from the DT

So those are the similarities, but from what I can tell from the current content posted here, is that there is one thing  which might not necessarily be a difference but a further step, which is in the instantiation of device drivers specifically. In the current revision of @fgrandel proposal, he doesn't go this far and still device drivers get instances based on what is status okay in DT. It seems to me like this proposal is basically the same except for this extra step. 

And I think that there is some value in this extra step, because currently, and even with the @fgrandel proposal, the only way to control driver selection is by Kconfig, which would mean all nodes of the certain DT compat use the same driver that is configured for that compat in the Kconfig, even if there are multiple options. Probably in many cases to save ROM space that is ideal, but maybe in some cases like timers/counter, which have many possible API in zephyr (PWM, counter, time aware GPIO, kernel timer, RTC etc), that is not feasible. This is actually a problem specifically with hardware timers and their representation that I have been thinking how to deal with from an enablement perspective for NXP parts for a while, but it seems like most ideas that dont involve reworking the whole zephyr config system are just dirty hacks that could explode into confusion or follow bad design patterns. 

But  like you said, it would require a large rework because now every single driver and corresponding DT overlays or base platform defaults that would pull in that driver will need rework. So the question is would we be willing to do this rework as a project and how would it be done. And since this is not much in opposition and actually very similar to many components of @fgrandel PoC, I wonder if it can use that somehow, since this RFC is really more about the idea of how we could USE a configuration system, and not necessarily suggesting any specific method of schemas/language/tooling. But since this RFC involves reworking how we do the actual static creation of devices, the scope expands greatly, so I am not clear on the details of if that would be possible within the limits of the same design goals that he had, so I'll let him comment about this

@fgrandel The absolute key point /requirement/ design goal that @bjarki-andreasen has hit upon here is how do we handle something like timers which maybe should have one hardware description representation in something like DT, but maybe we want to use instances of them in completely different ways, with even entirely different programming models within even the same build, like mentioned above in my comment and in this RFC. Do you have any idea about this from your vantage point of having worked on this topic for a while ?

benediktibk, 2024-10-23:
@bjarki-andreasen If I understood this RFC correctly you are proposing to move everything which is definitely not hardware related and rather driver specific to a different configuration system/file? I suppose then only things like memory addresses for peripherals would be left over in the devicetree, the rest should end up in the other system?

If this is the case, this proposal is very similar to the one @fgrandel presented in #77638. The real differences are:

- different language for the software configuration (JSON vs YAML)
- more stuff moved from the hardware configuration to the software configuration

I'm pretty convinced that the actual format we choose to use for software configuration is a matter which we can discuss separately. On a high level view the same information can be represented in devicetree, YAML or JSON. Therefore, I think we should not focus that much on the specific format.

More interesting to me is your shift of most configuration items into software. I think this is a reasonable idea, but: It would absolutely be possible to do the same thing with the PoC from @fgrandel.

bjarki-andreasen, 2024-10-23:
@decsny @benediktibk A notable benefit of this RFC is that it is not a tree, since software is generally not defined in trees :) 

Consider the following arbitrary config in some vague schema:
```
foo0:
  schema: foo
  value: 2
```
It has no parent, no path, no child, it doesn't reference anything, its just a config. 

compare this to an example extracted from @fgrandel s POC [commit](https://github.com/fgrandel/zephyr/commit/25c19c683a7f951e3d5f7ad6c67be8feb7c6190e#diff-3bfbfd28afaa532991c4254be2019097431bd3fdf674d59d4141bb6ab4bdf954R16) :
```
/{
	alpha {
		compatible = "zephyr,sample-settings";

		angle {
			1 = <90>;
		};
        };
};
```
Why is it a tree? (notice root node `/`) why is it using DTS syntax? What's with the compatible?

I believe we should focus on the design and usability of the "markup language" in favor of code reuse or familiarity. We don't have this "markup language" yet, and it should differ drastically from DTS given its describing zephyr specific software, not hardware...

benediktibk, 2024-10-23:
> @decsny @benediktibk A notable benefit of this RFC is that it is not a tree, since software is generally not defined in trees :)

I kind of disagree on this point, because software architecture can be described as tree:

    firmware
        http server
            ethernet device
        logging system
            uart
        CAN server
            CAN bus 1
            CAN bus 2

Probably this example is not the best architecture in the world, but it should get the point across. But in the end I honestly do not care that much if the software configuration is a tree or something else. The key point is collecting hardware and software configuration together in a common intermediate format, validating during build time references from one into another and generating macros, c-structs, binary images, ... whatever from this intermediate format. Implementing such a thing is definitely easier when hardware and configuration formats are closer matched, but I would not see this as a hard requirement. On the other hand a point can be made why users should be forced to learn something additionally to a tree-based system, when they are already used to something similar in devicetree.

benediktibk, 2024-10-23:
Btw, just thought about a better tree example: A complete linux system is described as a tree, including files, statistics, devices, ...

fgrandel, 2024-10-23:
@bjarki-andreasen 
```
/{
	alpha {
		compatible = "zephyr,sample-settings";

		angle {
			1 = <90>;
		};
        };
};
```

Bjarki, what you're citing is not in HEAD. You're picking out an intermediate experiment of mine that will not make it into any PR. This is at least careless and superficial reading of my code and at worst intentional ill-natured misinformation. At the very least you're abusing of the trust I put into the community by publishing an intermediate result that is not perfect. I think I made it very, very clear that the commit history is to be disregarded because I didn't see any use in cleaning it up unless we decide to at least partially go ahead.

How could I have dreamed of someone abusing my trust in such a way? **I very much hope this was just a mistake of yours and therefore I hope you confirm and correct this mistake as quickly as possible.**

Some background: I was using this to experiment with the NVS settings provisioning to generate some arbitrary data to check whether I was capable to build an image from it. The focus was in that case in no way on the source format which is completely irrelevant to the outcome. The provisioning part wasn't presented yet for a reason.

Here is the actual solution from HEAD.

https://github.com/fgrandel/zephyr/blob/c7c2780859ae818c84592f688a78f3f4a5b455f7/samples/subsys/settings/app.config.yaml

```
# SPDX-License-Identifier: Apache-2.0
# Sample NVS settings provisioning configuration.

alpha:
  compatible: zephyr,sample-settings

  angle:
    1: 90

  length:
    1: 41
    2: 58

  beta:
    voltage: 380
    source: AC

some:
  other:
    labeled_path:
      schema: zephyr,sample-settings
      gamma: 45
```

And even them I'm not saying that this is good configuration data. Having keys 1 and 2 is bullshit and I know that. I have to rewrite the settings sample to make this work but I just didn't see any reason to do this at the current point in time because it is not contributing any relevant risk reduction or extra option to the PoC.

fgrandel, 2024-10-23:
> A notable benefit of this RFC is that it is not a tree, since software is generally not defined in trees

This statement is completely the opposite of what the modeling community has proven to be true. All relevant large-scale configuration systems out there can be reduced to trees from Kubernetes across all kinds of IoC containers to the very well established Spring framework which has probably seen all kinds of software that has ever been invented on this globe. This is not a practical experience in the first place but is built on top of very sound theoretical concepts coming from several areas of (rather simple) mathematics.

As I've stated several times: I think a well-founded theoretical knowledge about state-of-the-art modeling and architecture is fundamental to not be confused to a point of saying something like this.

bjarki-andreasen, 2024-10-23:
> > A notable benefit of this RFC is that it is not a tree, since software is generally not defined in trees
> 
> This statement is completely the opposite of what the modeling community has proven to be true. All relevant large-scale configuration systems out there can be reduced to trees from Kubernetes across all kinds of IoC containers to the very well established Spring framework which has probably seen all kinds of software that has ever been invented on this globe. This is not a practical experience in the first place but is built on top of very sound theoretical concepts coming from several areas of (rather simple) mathematics.
> 
> As I've stated repeatedly: I think a well-founded theoretical knowledge about state-of-the-art modeling and architecture is fundamental to not be confused to a point of saying something like this. Unfortunately this knowledge seems to be rare and far apart in the embedded programming world still. But I'm confident this will change soon.

Thanks for your input as always @fgrandel If you could actually address the RFC as well that would be really helpful. Thanks :)

bjarki-andreasen, 2024-10-23:
@fgrandel regarding the example I picked from your branch, how am I supposed to know what is "just experimental" and actually meaningful? I have been pointed to that branch as a POC of what you are describing, if its not relevant, remove it.

decsny, 2024-10-23:
> Thanks for your input as always @fgrandel If you could actually address the RFC as well that would be really helpful. Thanks :)

Dude, he did? Why don't you read it then instead of saying he didn't? I'm not even the one being slighted here and I am frustrated by these things you're saying. He spoke a lot about this RFC in these lengthy comments above, how can you say that's not the case is absolutely baffling to me. I even pinged him about one specific aspect of it that I wanted to hear his thought on and he answered about that too. How can we work as a collaborative project if people just are playing make believe like this, I am so frustrated by how I have seen the discourse around this topic has been.

> @fgrandel regarding the example I picked from your branch, how am I supposed to know what is "just experimental" and actually meaningful? I have been pointed to that branch as a POC of what you are describing, if it (the commit) is not relevant, remove it.

First of all, we've talked about this across many arch WG calls, he's presented, you've asked this question MULTIPLE TIMES about why is he using DTS syntax which he explained he's not, he shared on his screen for about THIRTY MINUTES in the last arch WG the syntax which is very clearly yaml and NOT DTS, how many more times does this need to be repeated for you before you stop using this as a lazy claim to argue against this approach which you very clearly don't understand and are even more clearly not even attempting to.

@carlescufi I am absolutely so fed up with the unfaithful interactions about this topic between ARCH WG members, I have not even been following this topic until last week when I was first introduced to this idea in the meeting. I thought it was a pretty good idea but didn't necessarily expect it will be the last idea, but what I am surprised about is the absolute unfaithful and destructive, repeatedly misleading discourse surrounding it put forth by many of the participants, and the mode of engagement by many members who are supposed to be participating in a working group, not a fighting pit, this is completely unacceptable behavior and is making me want to disengage from even participating in arch WG if this is how people are going to act, by completely being ignorant and repeating falsehoods 100 times over that have been repeatedly been dispelled or even taken and addressed only because they want to be right and not actually interested in collaboration. What do we do about this to improve the situation, because until this meta problem is discussed, I don't even think it's worth going forward and discussing this topic anymore, we simply CANNOT come to any agreeable and collaborated solution while people are acting like children like this.

jfischer-no, 2024-10-23:
@bjarki-andreasen This RFC is a kind of trolling, no? (Also, are you aware that at this point in the project, the way we use devicetree and devicetree bindings, we cannot just break with that and introduce something new? We challenge the users (customers) with every release. So far, based on what I have seen, the smoothest way to "new configuration system" is Florian's proposal. So please do not behave not smart.)

bjarki-andreasen, 2024-10-23:
> @bjarki-andreasen This RFC is a kind of trolling, no? (Also, are you aware that at this point in the project, the way we use devicetree and devicetree bindings, we cannot just break with that and [...]

Have you seen the state of the comments on this RFC? not sure sarcasm or light snark in any way will be well met :)

fgrandel, 2024-10-23:
@decsny I'm so sorry to see your frustration and disappointment - especially as you've contributed a lot of deep thought to the debate. I've learnt a lot from you and others on this wild journey. But to some extent I think it's a healthy development that some of the more human problems are so clearly out in the open now so that we can talk about them and make our choices. And I think you'll agree that we do not need to single out anyone - it's just examples that could be found in many other interactions, too - nothing special or personal at all. I saw that many contributors "feared" interactions with some of the stronger commenters and I witnessed on several occasions how above all "free" community contributors tended to lose interest because they didn't feel safe. It's impressive how much "silent" support I got in private and which is the main reason why I'm still here to defend my case - the technical one and the human one.

As I'm someone who's not easily frustrated and even less fearing difficult interactions or conflict, I thought it would benefit our community to point the finger to where psychological safety and constructive collaboration was at danger. I very much believe that each and every honest feedback I gave was well founded in that sense. And therefore I am totally willing to talk it over with the people concerned - especially if it's done with the goal to define a debating culture that really furthers psychological safety for everyone and takes the bias away in favor of the over-confident to encourage the less extrovert to contribute their opinion, too, without fear.

This means that those taking too much room sometimes or not taking others' less outspoken needs seriously or defending strong opinions on the basis of personal beliefs and preferences rather than sound community-oriented argument need to be challenged in the open. And that's what I tried to do on purpose. I didn't confront the weak - even if they showed questionable behavior, too. I purposefully chose the strong because I know that they can defend themselves properly and won't be hurt in the long run. That's probably why we see long-standing community members show such behaviors especially in this context. IMO it just makes a much more general point about human behaviors and needs and is in now way linked to the specific people concerned, I'm very sure about this.

So while on one side I find it important to give honest and open feedback, it's is extremely important to me that at no point this should become something personal. And I think you see this very similarly. It's about behaviors, not about people. I feel very much with everyone involved and I want those people to be as successful as they were in the past - but not at the expense of others.

It's hard to accept truly open and honest feedback. I also got a lot of feedback over the last weeks. And on several occasions people were completely right and I acknowledged that in the open. (You gave me very good feedback once in that sense!) I hope that I was able to accept that feedback gracefully, though, where it was constructive. And I hope others will do so, too, even if it really, really hurts now. This goes away. And then we're all better off. I can already see a lot of positive development in that direction over the last few weeks.

rruuaanng, 2024-11-19:
> Here is a POC for this RFC https://github.com/bjarki-andreasen/zml

Hmmm. It's look like binding-API, Why can't we use the devicetree directly?

bjarki-andreasen, 2024-11-19:
> > Here is a POC for this RFC https://github.com/bjarki-andreasen/zml
> 
> Hmmm. It's look like binding-API, Why can't we use the devicetree directly?

The binding-API is great, but the devicetree syntax is not:
```
quak {
        foo = <3>;
        bar = <&intc 23 32>;
        baz = <-3>; /* does not work since DT spec does not support negative integers */
};
```

We can reuse most of the binding-API syntax and create something nicer like:
```
quak:
  foo: 3
  # bar = <&intc 23 32>; this makes no sense since phandles are a devicetree quirk
  baz: -3
};
```

rruuaanng, 2024-11-19:
In this way, I think it is more suitable as a meta description of the network subsystem or about service configuration. Because as pure software, they need a higher level of abstraction and may not need to be close to any specific transaction. But if it really lands, most developers may not be willing to go out of devicetree (at least I am not so willing to absorb too much new content because it is difficult)

de-nordic, 2024-11-19:
> 
> The binding-API is great, but the devicetree syntax is not (for software description, its great for hardware):
> 
> ```
> quak {
>         foo = <3>;
>         bar = <&intc 23 32>;
>         baz = <-3>; /* does not work since DT spec does not support negative integers */
> };
> ```
> 

Stupid question: since the DTS in current form does not fulfill our expectations why not take it and start making it into ZDTS, with our custom extensions and improvements?
Why we want to go into creating a new tool from scratch and going through all the pains of synchronizing the two specs to work together.
DTS as describes device configuration, that is half true, because when you have device with status = "disabled", it is not clearly that device is not there, it is rather clear that software (i.e. driver) does not have to bother supporting it.
Same thing with many device configuration options, like for example enabling deep sleep on SPI NOR devices: there is DTS param for that but it does not state whether device has the feature or not, it states whether driver on given instance of device should support it or not - that is software configuration, not hardware configuration. Hardware is bound by its specification not DTS.

The problem of tight coupling, you mention, is something that annoys me incredibly, but it is not caused by dts: it is caused by people reaching to devices via DT macros, bypassing layers subsystems and coupling themselves directly with things.
You will not solve this with new tool or spec file for configuration, you will just probably move part of that behaviour to that other file.
Reviewers need to be able to say "NO" when people start digging in not directly relates subsystems with DT macros.

bjarki-andreasen, 2024-11-19:
> Stupid question: since the DTS in current form does not fulfill our expectations why not take it and start making it into ZDTS, with our custom extensions and improvements? Why we want to go into [...]

Its a good question: Because I for one would like to move to [dtschema](https://pypi.org/project/dtschema/) for devicetree bindings and validation, allowing us to reuse Linux bindings and validation tooling directly. No chosen nodes, no zephyr,user :)

> DTS as describes device configuration, that is half true, because when you have device with status = "disabled", it is not clearly that device is not there, it is rather clear that software (i.e. [...]

That is the case now. One of the points of this RFC is to move stuff like status out of the devicetree. A driver should be selected by some software description language, where the driver would be pointed to a device to use, which in turn sets the status "okay" to indicate device is ready. status should not be a static value in DT.

> The problem of tight coupling, you mention, is something that annoys me incredibly, but it is not caused by dts: it is caused by people reaching to devices via DT macros, bypassing layers subsystems [...]

True, but we don't have a good alternative outside of CONFIG_<option>_<instance> which is not super scalable :)

de-nordic, 2024-11-19:
> 
> Its a good question: Because I for one would like to move to [dtschema](https://pypi.org/project/dtschema/) for devicetree bindings and validation, allowing us to reuse Linux bindings and validation tooling directly. No chosen nodes, no zephyr,user :)

Not sure if I will cry, need to read about the dtschema thing first.

> That is the case now. One of the points of this RFC is to move stuff like status out of the devicetree. A driver should be selected by some software description language, where the driver would be [...]

That just makes status software controller parameter, it is becoming software variable. You may as well not have it here at all, as it belongs to subsystem that is willing to support the instance or not; this reduces the DTS entry for a device to just a node that describes its position in hierarchy and address/reg space; everything else is becoming software configuration.

> 
> True, but we don't have a good alternative outside of `CONFIG_<option>_<instance>` which is not super scalable :)

No that is not my intention: Kconfig only to turn on/off driver wide stuff, if you turn something on in a driver that is available for each instance that is able to supports it, if you turn it off then even instances that support the feature will not have it - to remove, driverwide, stuff your solution does not use.

Btw. Nordic has a flash-pcd bridge device that passes operations on flash device via PCD to other CPU: this is completely software device, same as partitions, how does it fit into a new structure. Flash Simulator is similar thing.

benediktibk, 2024-11-20:
> and why limit ourselves to tree like structures:
> 
> ```
> foo: -3
> bar: "text"
> baz: 0
> ```

How do you intent to avoid name clashes when you get rid of the hierarchy?

bjarki-andreasen, 2024-11-20:
> > and why limit ourselves to tree like structures:
> > ```
> > foo: -3
> > bar: "text"
> > baz: 0
> > ```
> 
> How do you intent to avoid name clashes when you get rid of the hierarchy?

See https://github.com/bjarki-andreasen/zml/blob/07bac50a4e9adfc0efeae3dc0e5758a6c075249c/zml.yaml#L1-L17

benediktibk, 2024-11-20:
> > > and why limit ourselves to tree like structures:
> > > ```
> > > foo: -3
> > > bar: "text"
> > > baz: 0
> > > ```
> > 
> > 
> > How do you intent to avoid name clashes when you get rid of the hierarchy?
> 
> See https://github.com/bjarki-andreasen/zml/blob/07bac50a4e9adfc0efeae3dc0e5758a6c075249c/zml.yaml#L1-L17
> 
> There is still some hierarchy to describe the node schema and such, but the instances themselves are on the same level :)

I see, so basically it can still be described as a tree just with a depth of 1.

bjarki-andreasen, 2024-11-20:
> I see, so basically it can still be described as a tree just with a depth of 1.

Yeah :)

benediktibk, 2024-11-20:
> > I see, so basically it can still be described as a tree just with a depth of 1.
> 
> Yeah :)

So at this point you have solved the issue of name clashes for items which are part of instances. Which obviously leads to the next question: How are you planning on solving name clashes with instance names?

Maybe it would be better to just allow users to structure their applications in a tree, like basically almost everybody does, citing florian:

> This statement is completely the opposite of what the modeling community has proven to be true. All relevant large-scale configuration systems out there can be reduced to trees from Kubernetes [...]

I still do not get the point of structuring a software configuration not as a tree. In my opinion this approach has only disadvantages.

de-nordic, 2024-11-21:
> To be clear, the solution in this RFC allows you to do both, structure it as a tree, or flat, whichever is needed. There is no limitation on the "shape", I'm just pointing out that both are possible...

If I may suggest an approach: can we start some PR, where we put a file where we present a final description of something real, not something big, but representing various problems we already have being solved in context? Kinda overlay or the new description language in action?
Something that we can slowly shape and comment on, maybe even propose PRs to, according to our expertise and fields we care about.
This way users can propose PRs to that file with their problems to solve and which we can debate on, shape and bring in and evaluate the solution for their problem, how it looks, in the context of the generic solution that is delivered?

And once we have something that looks like a thing that we can really use, go back from that point and try to define requirements from what clearly see we want to achieve?

Because right now I look at things under my "control" and wonder whether after we are done that I would have now two sets of macros cutting everything here and there (DT_ and whatever), or even whether I would have to significantly rewrite things, because seemingly they no longer are devices.