zephyrproject-rtos/zephyr #80277: Issue with Intel Raptor Lake and MEC1521 SPI Image Flashing:

jayeshyantram6386, 2024-10-23:
I'm working with Intel Raptor Lake and using the open-source ECFW Zephyr (ecfw-zephyr) to generate the ksc.bin file. Here’s what I did
Cloned the CPGZephyrDocs repository and added the necessary SPI image generation files.
Set the environment variable for everglades_spi_gen_RomE.
Successfully built the project using west build -c -p auto -b mec1501_mtl_p, which generated the ksc.bin file.
Stitched the ksc.bin with Intel BIOS and flashed it using Dediprogrammer onto the Intel Raptor Lake S board.
After flashing, both the CPU fan and ATX fan started working, and the power sequence voltages were active, but the system did not boot. However, these signals (CPU fan, ATX fan, and power sequence) only activate when I manually touch the RSMRST signal using a multimeter.

Any insights on why the system isn't booting after flashing?

nashif, 2024-10-23:
@jayeshyantram6386 this issue should be opened on https://github.com/intel/ecfw-zephyr

closed 2024-10-23 by nashif

jayeshyantram6386, 2024-10-24:
> @jayeshyantram6386 this issue should be opened on https://github.com/intel/ecfw-zephyr

Thanks