zephyrproject-rtos/zephyr #80284: mcuboot+application don't start on mimxrt1050_evk

blemouzy, 2024-10-23:
**Describe the bug**

With default flash SPI partitions layout, flashing an application seems to erase mcuboot on mimxrt1050_evk.

Tested with Zephyr v3.7.0 and Zephyr main (39917ae5e6423fe74edce399b927ddff38670817).

**To Reproduce**

1. Flash mcuboot

```bash
cd bootloader/mcuboot/boot/zephyr/
echo "CONFIG_BOOT_SIGNATURE_TYPE_NONE=y" >> prj.conf
west build -b mimxrt1050_evk@hyperflash
west flash --runner=jlink --erase
```

==> mcuboot starts as expected:

```
*** Booting MCUboot v2.1.0-rc1-36-gfb2cf0ec3da3 ***
*** Using Zephyr OS build v3.7.0 ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
W: Failed reading image headers; Image=0
E: Unable to find bootable image
```

2. Flash application

```bash
cd zephyr/samples/basic/blinky/
echo "CONFIG_BOOTLOADER_MCUBOOT=y" >> prj.conf
echo "CONFIG_MCUBOOT_GENERATE_UNSIGNED_IMAGE=y" >> prj.conf
west build -b mimxrt1050_evk@hyperflash
west flash --runner=jlink
```

==> nothing is printed on serial like application programming erased mcuboot.

**Expected behavior**

mcuboot and application start:

```
*** Booting MCUboot v2.1.0-rc1-36-gfb2cf0ec3da3 ***
*** Using Zephyr OS build v3.7.0 ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Image index: 0, Swap type: none
I: Bootloader chainload address offset: 0x100000
I: Jumping to the first image slot
*** Booting Zephyr OS build v3.7.0 ***
LED state: OFF
LED state: ON
```

**Additional context**

I made some tests and with the following flash SPI partition change, mcuboot and application start as expected:

```
zephyr/boards/nxp/mimxrt1050_evk/mimxrt1050_evk_mimxrt1052_hyperflash.overlay

partitions {
    compatible = "fixed-partitions";
    #address-cells = <1>;
    #size-cells = <1>;
    boot_partition: partition@0 {
        label = "mcuboot";
        reg = <0x00000000 DT_SIZE_M(1)>;
    };
    /* The MCUBoot swap-move algorithm uses the last 14 sectors
        * of the primary slot0 for swap status and move.
        */
    slot0_partition: partition@100000 {
        label = "image-0";
        reg = <0x00100000 DT_SIZE_M(3)>;
    };
    slot1_partition: partition@400000 {
        label = "image-1";
        reg = <0x00400000 DT_SIZE_M(3)>;
    };
    storage_partition: partition@700000 {
        label = "storage";
        reg = <0x00700000 DT_SIZE_M(58)>;
    };
};
```

I hope it is helpful.

butok, 2024-10-31:
Hi @blemouzy

Going to order the board.  
Are you using the IMXRT1050-EVKB board (https://www.nxp.com/part/IMXRT1050-EVKB)?

Meanwhile, could you try the the MCUBoot sysbuild sample:
- west build -p always -b mimxrt1050_evk --sysbuild samples/sysbuild/with_mcuboot
- west flash

blemouzy, 2024-10-31:
Hi @butok 

> Are you using the IMXRT1050-EVKB board (https://www.nxp.com/part/IMXRT1050-EVKB)?

Yes (revision B1).

> Meanwhile, could you try the the MCUBoot sysbuild sample:
> 
>     * west build -p always -b mimxrt1050_evk --sysbuild samples/sysbuild/with_mcuboot
> 
>     * west flash

Same issue.

danieldegrasse, 2024-11-08:
Looking at this more, I suspect the issue might be a bit deeper, but related to the partition alignment. The [datasheet](https://www.infineon.com/dgdl/Infineon-512-MB64_MB_256-MB32_MB_128-MB16_MB1_8_V3-DataSheet-v17_00-EN.pdf?fileId=8ac78c8c7d0d8da4017d0ed5ca77559f) for the S26KS512S hyperflash on the RT1050 claims erases may only be performed on 256 KB sectors, and the driver implementation aligns with this: https://github.com/zephyrproject-rtos/zephyr/blob/67980643d799f76aff93118346a8dabfb472a642/drivers/flash/flash_mcux_flexspi_hyperflash.c#L513

I believe the change in 03d1eabc32c74 is probably the root cause here. We reduced the size of the mcuboot boot partition, which shifted the alignment of the image-0 and image-1 partitions to no longer be on 256KB boundaries. I suspect this breaks the flash loader algorithm debuggers use, since we cannot erase at the 128KB boundary.

I think the partition alignment @blemouzy proposed is right- we need all partitions to be aligned to a 256KB boundary, or I doubt we will be able to run upgrades correctly with MCUBoot.

dleach02, 2024-11-08:
I posted a PR based on @blemouzy. In the sample suggested by @butok it worked on my board.

But if this is a 256KB alignment, seems like we really only need to bump the boot_partition size from 128 to 256?

butok, 2024-11-11:
> I posted a PR based on @blemouzy. In the sample suggested by @butok it worked on my board.
> 
> But if this is a 256KB alignment, seems like we really only need to bump the boot_partition size from 128 to 256?

Hi @danieldegrasse and @dleach02 
For both mimxrt1050 qspi and hyperflash `erase-block-size = 4096`
If _"S26KS512S hyperflash on the RT1050 claims erases may only be performed on 256 KB sectors"_ is true.
the device tree must be fixed to  `erase-block-size = DT_SIZE_K(256)`. Right?

blemouzy, 2024-11-12:
> But if this is a 256KB alignment, seems like we really only need to bump the boot_partition size from 128 to 256?

It looks fine (application starts) with 256 KiB `boot_partition`:

```
partitions {
	compatible = "fixed-partitions";
	#address-cells = <1>;
	#size-cells = <1>;
	boot_partition: partition@0 {
		label = "mcuboot";
		reg = <0x00000000 DT_SIZE_K(256)>;
	};
	slot0_partition: partition@40000 {
		label = "image-0";
		reg = <0x00040000 DT_SIZE_M(3)>;
	};
	slot1_partition: partition@340000 {
		label = "image-1";
		reg = <0x00340000 DT_SIZE_M(3)>;
	};
	storage_partition: partition@640000 {
		label = "storage";
		reg = <0x00640000 (DT_SIZE_M(57) + DT_SIZE_K(768))>;
	};
};
```

Note: is the following comment still relevant?

```c
/* The MCUBoot swap-move algorithm uses the last 11 sectors
 * of the primary slot0 for swap status and move.
 */
```

Do we need the extra `DT_SIZE_K(44)` in `slot0_partition`?

butok, 2024-11-12:
> It looks fine (application starts) with 256 KiB boot_partition

Thanks for the confirmation.

>Do we need the extra DT_SIZE_K(44) in slot0_partition?

No. It was true for the old (wrong) erase size.
It works with equal sized slots, but you may get a warning that the partition sizes are not optimized.
According to my calculations, with the new 256 KB erase block size we only need +2 sectors (512 KB).

BTW: I should get the board by the end of this week, and will test on the real board, and prepare a PR fix if no one does it sooner ;)

andrisk-dev, 2024-11-12:
I did some test runs on mimxrt1060_evkb - first I was able to reproduce the issue. Than I tried the PR linked to this issue "May be fixed by" (https://github.com/zephyrproject-rtos/zephyr/pull/81164) and that build worked.

butok, 2024-11-12:
> test runs on mimxrt1060_evkb - first I was able to reproduce the issue.

OK. It means that this is the common mimxrt1060 and mimxrt1050 issue.

andrisk-dev, 2024-11-12:
> > test runs on mimxrt1060_evkb - first I was able to reproduce the issue.
> 
> OK. It means that this is the common mimxrt1060 and mimxrt1050 issue.

Oh, sorry I meant mimxrt1050_evkb (not 1060)

butok, 2024-11-13:
FYI:
- Received the RT1050 EVKB board
- Reproduced the issue
- Submitted fix PR: https://github.com/zephyrproject-rtos/zephyr/pull/81335

RT1050 EVKB test_mcuboot log:
```
*** Booting MCUboot v2.1.0-rc1-146-gf74b77cf7808 ***
*** Using Zephyr OS build v4.0.0-rc3-4-g65511eacf9d1 ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Image index: 0, Swap type: none
I: Bootloader chainload address offset: 0x40000
I: Image version: v0.0.0
I: Jumping to the first image slot
*** Booting Zephyr OS build v4.0.0-rc3-4-g65511eacf9d1 ***
Launching primary slot application on mimxrt1050_evk
Secondary application ready for swap, rebooting�*** Booting MCUboot v2.1.0-rc1-146-gf74b77cf7808 ***
*** Using Zephyr OS build v4.0.0-rc3-4-g65511eacf9d1 ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=good, swap_type=0x3, copy_done=0x3, image_ok=0x1
I: Boot source: none
I: Image index: 0, Swap type: perm
I: Starting swap using move algorithm.
I: Bootloader chainload address offset: 0x40000
I: Image version: v0.0.0
I: Jumping to the first image slot
*** Booting Zephyr OS build v4.0.0-rc3-4-g65511eacf9d1 ***
Swapped application booted on mimxrt1050_evk
```

@blemouzy 
Thank you for reporting issue. Discovered the testing gap on our side.

@danieldegrasse 
Thank you for pointing to the wrong  S26KS512S write size, it saved a lot of time.

PR #81335 (2024-11-16): boards: nxp: fix s26ks512s0 flash write-block-size
- Sets s26ks512s0 flash write-block-size to correct 256KB.
- Optimizes MCUboot partitions to fit the correct write-block-size.

Fixes #80284
commit a730d9ab (2024-11-16): boards: nxp: fix s26ks512s0 flash write-block-size

- Sets s26ks512s0 flash write-block-size to correct 256KB.
- Optimizes MCUboot partitions to fit the correct write-block-size.

Fixes #80284
commit 371e3a67 (2024-11-22): boards: nxp: fix s26ks512s0 flash write-block-size

- Sets s26ks512s0 flash write-block-size to correct 256KB.
- Optimizes MCUboot partitions to fit the correct write-block-size.

Fixes #80284

(cherry picked from commit a730d9abad9241c0eb3fe0be64f42137692a6285)
closed 2024-11-16 by nashif