zephyrproject-rtos/zephyr #80285: TAINT: TAINT enabled on binaries not actually including blobs

erwango, 2024-10-23:
**Describe the bug**

This is a resurrection of discussion that happened [during #77386 review](https://github.com/zephyrproject-rtos/zephyr/pull/77386#issuecomment-2306641080).

Issue is that `CONFIG_TAINT=y` is set irrespective of the fact that blob is actually included of not in the binary.
This is in contradiction with [tainting documentation](https://docs.zephyrproject.org/latest/contribute/bin_blobs.html#tainting), which specifies that TAINT symbol should be defined based on *inclusion* ("a tainted image will be one that includes binary blobs in it."), not simply that a blob was fetched.

**To Reproduce**

Note: In current main blobs are made available in hal_stm32 in order to provide BLE support on STM32WBA series.
In the steps below, I'm fetching the blobs, but won't be using them.

1. west build -b nucleo_f207zg samples/hello_world/ ; west flash => Not Tainted
2. west blobs fetch hal_stm32
3. west build -b nucleo_f207zg samples/hello_world/ ; west flash => Tainted
4. west blobs clean
5.  west build -b nucleo_f207zg samples/hello_world/ ; west flash => Not Tainted

=> Taint status of an image of a non BLE sample on a non BLE series is impacted by the presence of a blob in the tree.

**Expected behavior**
TAINT should be set according to the documentation, ie when the blob is included in the image.
Otherwise documentation should be fixed (which would need to be reviewed at TSC level as the blob documentation was approved by TSC).

**Impact**
TAINT information becomes useless since not accurate.

erwango, 2024-10-23:
@rettichschnidi FYI

rettichschnidi, 2024-10-23:
Just in case anyone would want to remove the whole TAINT symbol/logic: From my PoV, until we have a perfectly matching solution as requested by this issue/the documentation, it is better to err on the save side, and apply TAINT too often, rather than the opposite.

nordicjm, 2024-10-23:
> Just in case anyone would want to remove the whole TAINT symbol/logic: ~From my PoV, until we have a perfectly matching solution as requested by this issue/the documentation, it is better to err on the save side, and apply TAINT too often, rather than the opposite.~
> 
> IMHO, the `tainted` status should not depend only on the resulting image, but on the presence of the blobs (as is today). Reason: Just by sitting on the file system they can (potentially) induce problems.

Agreed, and this was the same thing said in the PR linked above

rettichschnidi, 2024-10-23:
Heads up: #80314

erwango, 2024-10-23:
> Reason: Just by sitting on the file system they can (potentially) induce problems.

Would you have an example ?

> Agreed, and this was the same thing said in the PR linked above

If this is the consensus then why not, but documentation should be updated.

erwango, 2024-10-23:
@carlescufi Any opinion, as you are the one that wrote the documentation ?

rettichschnidi, 2024-10-23:
> > Reason: Just by sitting on the file system they can (potentially) induce problems.
> 
> Would you have an example ?

I do not have one at hand, but I might be able to create one.

> > Agreed, and this was the same thing said in the PR linked above
> 
> If this is the consensus then why not, but documentation should be updated.

That would then resolve this issue and I can go ahead with other PRs such as #80314?

henrikbrixandersen, 2024-10-23:
I am in strong favor of not marking a build as tainted just because a binary blob is present in the workspace. The rationale for this is that for many developers working on upstream Zephyr, blobs _will_ be present due to testing on platforms requiring these, but this should not tag all bug reports/logs coming for that developer as tainted.

rettichschnidi, 2024-10-23:
> > Would you have an example ?
> 
> I do not have one at hand, but I might be able to create one.

I spent a bit of time and could not come up with an example. Maybe time will. :)

nordicjm, 2024-10-24:
> > Reason: Just by sitting on the file system they can (potentially) induce problems.
> 
> Would you have an example ?

The same one I gave to you before https://github.com/zephyrproject-rtos/zephyr/pull/77386#issuecomment-2307027410

> Since you can load anything through llext it must be set if blobs are present, irrespective if used in a build or not.

erwango, 2024-10-24:
> > > Reason: Just by sitting on the file system they can (potentially) induce problems.
> > 
> > 
> > Would you have an example ?
> 
> The same one I gave to you before [#77386 (comment)](https://github.com/zephyrproject-rtos/zephyr/pull/77386#issuecomment-2307027410)
> 
> > Since you can load anything through llext it must be set if blobs are present, irrespective if used in a build or not.

As as I answered:

> From llext you can load anything, even out of zephyr build system using llext edk, so such mechanism would be largely inefficient in any case.

It would be much more efficient to set TAINT if llext is enabled, if you fear issues with llext.

nordicjm, 2024-10-24:
> As as I answered:
>
>
> It would be much more efficient to set TAINT if llext is enabled, if you fear issues with llext.

Works for me!

rettichschnidi, 2024-10-25:
> > 
> 
> It would be much more efficient to set TAINT if llext is enabled, if you fear issues with llext.

This seems very doable. 👍

What I am more concerned with is: I currently can not see a *reliable* way to detect if any of the binary blobs present on the file system actually end up in the final image (images?). Any ideas?

nordicjm, 2024-10-28:
The taint Kconfig for a particular blob should only be set if it's used? Unless that's the way it uses to detect if the blob *can* be used but to me there should be a Kconfig that is force set if a blob is used with specific naming (then you can just comb through the Kconfigs) and a different way of knowing if it's available to use (which doesn't mean it is used in an image, just that it can be used)

rettichschnidi, 2024-10-29:
> The taint Kconfig for a particular blob should only be set if it's used?

Currently, the pure presence of blobs sets the TAINT flag. No matter whether an image actually uses them. I think this is a feature and not a bug, but people in #80389 do not see it this way.

> Unless that's the way it uses to detect if the blob _can_ be used but to me there should be a Kconfig that is force set if a blob is used with specific naming (then you can just comb through the [...]

Would this be a convention that everyone has to stick to? If so, when not sticking to this convention, on purpose or not, one could prevent builds using a blob from getting flagged as TAINTed?

keith-zephyr, 2024-10-30:
Process WG discussion:

@MaureenHelm - Change made to display TAINT in boot banner.  Release team reverted change until this could be discussed in the Process WG.
@jhedberg - Revert only affected the boot banner, and not the Kconfig setting
@dleach02 - How does one determine what blobs are actually in the build?
@jhedberg - Should the check be for any binary added into the image, even ones not maintained by a Zephyr module.
@dleach02 - CONFIG_TAINT is set automatically when the blob is fetched.
@jhedberg - CI processes the modules to generate the CONFIG_TAINT symbol.  There was a bug that skipped the stm32 hal.
@keith-zephyr - can we scan the output binary?  Consensus - this is too late because the information is needed to set the Kconfig option
@jhedberg - Create a CMake helper function that is mandatory to run when including binaries.  We also need checks to make sure the helper isn't bypassed.  `add_library` could be wrapped to perform these checks.  Need to check if there are uses of `add_library` that outside of modules.
@dleach02 - scanning the final binary has problems in setting the TAINT label into the binary after the fact.h
@dleach02 - a workaround today is to cleanup your repo to remove the blobs from the filesystem.
@keith-zephyr - what about a post build check to scan for binaries?
@dleach02 - this is too much burden and cost for most projects.
@jfischer-no Current [script](https://github.com/zephyrproject-rtos/zephyr/blob/9cce72daa192d954ce462c5680bd8b7e1ed0a780/scripts/zephyr_module.py#L328-L385) for checking Kconfig options in a mode.
@dleach02 - proposal to push back to @erwango to clean local the workspace as a workaround.  `west blobs clean`.
@dleach02 - there is still an open issue to better track whether blobs are actually used.
@stephanosio - Workaround is somewhat brute force, and not particularly developer friendly.  The documentation isn't accurate.
@dleach02 - there are few possible solutions, but we need an owner to drive to a solution.
@jhedberg - Side topic - it would be useful to document which features depend on a blob.  Board metadata could include info as to whether there is blob dependency.  The CI can then skip these boards.
@dleach02 - Would also like to see a streamlined process for approving blobs with TSC involvement.  @keith-zephyr - we could add a maintainers section specifically for blobs.
@stephanosio - this issue should be fixed.
@dleach02 - agreed this is a bug in the build system
@dleach02 - suggest we mark the issue as a bug and reassign to torsten.

jhedberg, 2024-10-30:
> @jhedberg - Side topic - it would be useful to document which features depend on a blob. Board metadata could include info as to whether there is blob dependency. The CI can then skip these boards.

Clarification to this: assuming we'd introduce something like that, I don't think it should be on a board level, rather it should be per-feature (or perhaps "feature A with board B").

nashif, 2024-10-31:
tbh, I question the need for all of this taint labelling in the first place, what exactly is the purpose of it?
If I am building a product that depends on a binary provided by the vendor I certaily do not want to see/show some banner declaring that my product binary is tainted which means contaminated or polluted.

It is a problem if a basic zephyr feature supported and implemented in zephyr causes the contamination, but not if it was a choice made when I selected a vendor that only provide some additional feature on top of zephyr using come binary.

If that is going to happen, we will need an option to disable this as well. Integrating blobs I trust (i.e. I built the binary blob myself and I know what it is) is not tainted for me, maybe for others...

Just copying what linux does with binary modules does not always translate well in the world of zephyr and embedded in general.

nordicjm, 2024-10-31:
> tbh, I question the need for all of this taint labelling in the first place, what exactly is the purpose of it?

Support really, if someone opens a bug, why should we spend time looking into an issue if it's with a private codebase that they do not share using blobs which might be the fundamental cause of it? That's why linux has it and why I think it works here. Likewise the boot banner in NCS is different - which frankly is great because instead of investigating possible non-issues in zephyr people can be told to reproduce with upstream or post the issue to nordic support whereas before you spend the time trying something out, it works fine, you post back and then they say it's with NCS, wasting both yours and theirs time.

jhedberg, 2024-10-31:
> if someone opens a bug, why should we spend time looking into an issue if it's with a private codebase that they do not share using blobs which might be the fundamental cause of it?

Isn't a private code base sufficient by itself? Mixing blobs into such a support situation doesn't really make a difference.

rettichschnidi, 2024-10-31:
> Isn't a private code base sufficient by itself?

Not sure what you are saying. Sufficient reason to not expect upstream to help? I'd say so.

> Mixing blobs into such a support situation doesn't really make a difference.

Blobs are based on a private code base. Even worse - the reporter can not share the source to it, even if they wanted to. Blobs deserve to stick out like a sore thumb.

jhedberg, 2024-10-31:
> > Isn't a private code base sufficient by itself?
> 
> Not sure what you are saying. Sufficient reason to not expect upstream to help? I'd say so.

Yes, that's what I meant.

> Blobs are based on a private code base. Even worse - the reporter can not share the source to it, even if they wanted to. Blobs deserve to stick out like a sore thumb.

I'm not sure that I'm completely following the logic here. If we say that a private code base is sufficient to eliminate expectations of upstream support, then it's sufficient. Yes, blobs in the mix is worse (again, from an upstream perspective), but it seems it's also irrelevant since we've already crossed into the "not supported" territory with the private code base part.

It's also not quite the same to say that there are no expectations of upstream support, and saying that such support doesn't exist. For upstream build configurations where blobs are involved, it's typically employees of the companies that provide the blobs who are also maintainers for the respective Zephyr modules and platforms. I'd therefore expect it to be common case that these maintainers don't mind helping out and investigating issues where blobs are involved.

nashif, 2024-11-01:
> > tbh, I question the need for all of this taint labelling in the first place, what exactly is the purpose of it?
> 
> Support really,

do we have examples of this support burden or overhead to the project already?

nordicjm, 2024-11-04:
> > > tbh, I question the need for all of this taint labelling in the first place, what exactly is the purpose of it?
> > 
> > 
> > Support really,
> 
> do we have examples of this support burden or overhead to the project already?

I didn't save the links but have had various issues over the years which were caused by NCS not zephyr, @henrikbrixandersen sees them in issues too, lots of issues are raised which are relating to blobs the 2 biggest number I've seen are espressif bluetooth and stm32 bluetooth

mathieuchopstm, 2024-11-04:
> * Extend zephyr_link_libraries() to detect when [a string containing "/blobs/"](https://github.com/zephyrproject-rtos/hal_espressif/blob/0f1874284f5dee0d49cb23f44f756e7be404e7b7/zephyr/esp32s2/CMakeLists.txt#L584) gets passed
>   
>   * Would [also need adaption of generate_inc_file_for_target()](https://github.com/zephyrproject-rtos/hal_nxp/blob/4a4741fa2be33f6b638a49e357c5e33bb7ad0544/zephyr/src/rw61x/CMakeLists.txt#L59)
>   * [Would need set_target_properties()](https://github.com/zephyrproject-rtos/hal_stm32/blob/019d8255333f96bdd47d26b44049bd3e7af8ef55/lib/CMakeLists.txt#L61) to be instrumented too
>   * ... an ever growing list of other ways to introduced blobs ...
>   * → this does not work
> 
> Any better ideas?

What if we introduced a `zephyr_link_blob_libraries()` function that modules would be required to use in place of `zephyr_link_libraries()` when linking blobs?

If tainting is to be used as help for issue/support triage, as suggested by @nordicjm (similar to the role of NCS splash), I don't see the issue with making this an opt-in behavior. It should be pretty easy to enforce this in modules under the Zephyr Project umbrella (e.g., vendor HALs), which I think are where most (if not all) blobs come from (?)

On the other hand, if tainting is to be used so issues can be automatically closed under "tainted -> unsupported configuration" rule, as is what I understand some people suggest, I'm not really convinced there's much to be done: what prevents one from just removing `(tainted)` from logs, or removing the whole `config TAINT` machinery from his downstream fork, then ask for support? The only way to "protect" against this would be to request full tree availability, but as said by @jhedberg:
> If we say that a private code base is sufficient to eliminate expectations of upstream support, then it's sufficient. Yes, blobs in the mix is worse (again, from an upstream perspective), but it [...]