zephyrproject-rtos/zephyr #80289: net: code improvement in process_data() for high throughput case

MaochenWang1, 2024-10-23:
**Is your enhancement proposal related to a problem? Please describe.**
In process_data(), there are three net_packet_socket_input() called, looks redundant for running UDP throughput case. 
And can we skip/optimize some of these functions? I locally tried, skip these functions can improve UDP RX for 5 Mbps. 

**Describe the solution you'd like**
can we add some configs to skip some net_packet_socket_input() functions for high throughput case.

jukkar, 2024-10-24:
Just disable `CONFIG_NET_SOCKETS_PACKET` so that the packet socket processing is not done. If you need packet socket support, then the calls are needed to handle raw packet data in sockets.

The three `net_packet_socket_input()` calls are each handling different kind of packet/raw sockets.

jukkar, 2024-10-24:
It is probably possible to optimize the packet socket handling even further, patches are certainly welcome to improve performance.

MaochenWang1, 2024-10-28:
Hi @jukkar  
the CONFIG_NET_SOCKETS_PACKET are needed for hostap enabled case to handle the EAPoL packets.
Any suggestions to optimize the packet socket handling even further?
Thanks

rlubos, 2025-04-23:
We got rid of one of the packet socket input calls (`net_packet_socket_input(pkt, IPPROTO_RAW)`) with the invalid IPPROTO_RAW socket removal: https://github.com/zephyrproject-rtos/zephyr/pull/88044. The alternative (AF_INET/SOCK_RAW) now needs to be explicitly enabled in Kconfig, so it won't affect the overall performance if not used.

However, for the remaining two `net_packet_socket_input()` calls, I don't really see much room for improvement here. If packet sockets are enabled, we need to call this function before and after L2 processing, so it needs to be called twice. I've simplified the packet processing a bit in https://github.com/zephyrproject-rtos/zephyr/pull/88720, so we no longer call the bulky `net_conn_input()` in those cases, but still it's more of a simplification rather than optimization.

So, with all above, I think we should close the issue as no more improvements can be seen from my side. Please reopen if you have any specific ideas on what could be changed/improved if possible.

closed 2025-04-23 by rlubos