zephyrproject-rtos/zephyr #80295: tests: drivers: flash: common.default test Failed

hakehuang, 2024-10-23:
**Describe the bug**
common.default test is Failed on v3.7.0-4826-g372b3a9b8da2 on lpcxpresso54114/lpc54114/m4
testcase path is drivers/flash/common/default/flash_copy

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso54114/lpc54114/m4  --sub-test drivers.flash
```
or
```
# cd tests/drivers/flash/common/default/flash_copy
# west build -b lpcxpresso54114/lpc54114/m4
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build v3.7.0-4826-g372b3a9b8da2 ***
Running TESTSUITE flash_driver
===================================================================
START - test_flash_copy
Test will run on device flash-controller@4009c000
E: Size of CONFIG_FLASH_FILL_BUFFER_SIZE

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:386: test_flash_copy_inner: flash_fill(flash_dev, 0xaa, page_info.start_offset, page_info.size) is non-zero

 FAIL - test_flash_copy in 0.326 seconds
===================================================================
START - test_flash_erase
Test will run on device flash-controller@4009c000
 PASS - test_flash_erase in 0.208 seconds
===================================================================
START - test_flash_fill
Test will run on device flash-controller@4009c000
E: Size of CONFIG_FLASH_FILL_BUFFER_SIZE

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:210: flash_driver_test_flash_fill: (rc not equal to 0)
Fill failed

 FAIL - test_flash_fill in 0.222 seconds
===================================================================
START - test_flash_flatten
Test will run on device flash-controller@4009c000
E: Size of CONFIG_FLASH_FILL_BUFFER_SIZE

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/flash/common/src/main.c:242: flash_driver_test_flash_flatten: (rc not equal to 0)
Fill failed

 FAIL - test_flash_flatten in 0.222 seconds
===================================================================
START - test_flash_page_layout
Test will run on device flash-controller@4009c000
start_offset=0x30000	size=256	index=768
page_count=1024
 PASS - test_flash_page_layout in 0.011 seconds
===================================================================
START - test_read_unaligned_address
Test will run on device flash-controller@4009c000
 PASS - test_read_unaligned_address in 0.213 seconds
===================================================================
TESTSUITE flash_driver failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  50.00% [flash_driver]: pass = 3, fail = 3, skip = 0, total = 6 duration = 1.202 seconds
 - FAIL - [flash_driver.test_flash_copy] duration = 0.326 seconds
 - PASS - [flash_driver.test_flash_erase] duration = 0.208 seconds
 - FAIL - [flash_driver.test_flash_fill] duration = 0.222 seconds
 - FAIL - [flash_driver.test_flash_flatten] duration = 0.222 seconds
 - PASS - [flash_driver.test_flash_page_layout] duration = 0.011 seconds
 - PASS - [flash_driver.test_read_unaligned_address] duration = 0.213 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: 07945c02a70c4cd207bcab1e8f4f9542
PROJECT EXECUTION FAILED

    
```**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v3.7.0-4826-g372b3a9b8da2

butok, 2024-10-30:
Hi @hakehuang, 
I do not have this platform, so far.
Is it posible to bisect to a problematic commit, to see if it's fixable blindly.

dleach02, 2024-10-30:
@hakehuang can you bisect where this failure came in?

hakehuang, 2024-10-31:
> @hakehuang can you bisect where this failure came in?

this fails since the flash test is enabled. and It is report before(closed), and I see @butok starts to fix flash issue, and I report again.

butok, 2024-10-31:
> > @hakehuang can you bisect where this failure came in?
> 
> this fails since the flash test is enabled. and It is report before(closed), and I see @butok starts to fix flash issue, and I report again.

Hi @hakehuang

So far, I have tried the following tests, and fixed issues related to >32 write size for LPC55S69/MCXN947:
- tests/drivers/flash/common (default configuration)
- tests/drivers/flash/erase_blocks
- tests/subsys/storage/flash_map

Did not try the reported drivers/flash/common/default/flash_copylink.
It means that the problem can be there from the beginning.
Can you test other platforms, it can be a common issue.

hakehuang, 2024-11-01:
> Did not try the reported drivers/flash/common/default/flash_copylink.
> It means that the problem can be there from the beginning.
> Can you test other platforms, it can be a common issue.

I will run a full cycle this week, will feedback on those

butok, 2024-11-04:
> cd tests/drivers/flash/common/default/flash_copy

Hi @hakehuang 
The tests/drivers/flash/common/**default/flash_copy** folder does not exist.
Only tests/drivers/flash/common exists.

Please fix the west reproduce steps.

hakehuang, 2024-11-04:
> The tests/drivers/flash/common/**default/flash_copy** folder does not exist.

sorry update the path to drivers.flash.common

butok, 2024-11-15:
> > Did not try the reported drivers/flash/common/default/flash_copylink.
> > It means that the problem can be there from the beginning.
> > Can you test other platforms, it can be a common issue.
> 
> I will run a full cycle this week, will feedback on those

I have tested test drivers/flash/common on many different available platforms and it is passed.
So the problem is specific to LPC54114 (its DTS or Flash drivers).
As the LPC54 board is not available, there is no plan on my side.

decsny, 2025-01-17:
I have access to this board so I will assign myself