zephyrproject-rtos/zephyr #80296: tests: kernel: objects: minimallibc test Timeout

hakehuang, 2024-10-23:
**Describe the bug**
minimallibc test is Timeout on v3.7.0-4826-g372b3a9b8da2 on lpcxpresso54114/lpc54114/m4
testcase path is kernel/objects/minimallibc/pending_timer

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p lpcxpresso54114/lpc54114/m4  --sub-test kernel.objects
```
or
```
# cd tests/kernel/objects/minimallibc/pending_timer
# west build -b lpcxpresso54114/lpc54114/m4
# west flash
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**

```
*** Booting Zephyr OS build v3.7.0-4826-g372b3a9b8da2 ***
Running TESTSUITE pending
===================================================================
START - test_pending_fifo
Testing preemptible threads block on fifos ...
Testing fifos time-out in correct order ...
Testing  fifos delivered data correctly ...
 PASS - test_pending_fifo in 2.013 seconds
===================================================================
START - test_pending_lifo
Testing preemptible threads block on lifos ...
Testing lifos time-out in correct order ...
Testing lifos delivered data correctly ...
 PASS - test_pending_lifo in %u.%03u seconds
TESTSUITE %s succeeded
TESTSUITE %s failed.
ERROR: Test suite '%s' did not run.
ERROR: Test '%s' assigned to test suite '%s' which doesn't exist
Bad stats for %s.%s

------ TESTSUITE SUMMARY START ------

SUITE %s - %3d.%02d%% [%s]: pass = %d, fail = %d, skip = %d, total = %d duration = %u.%03u seconds
FLAKY - %s - [%s.%s] - (Failed %d of %d attempts) - duration = %u.%03u seconds
 - %s - [%s.%s] duration = %u.%03u seconds
------ TESTSUITE SUMMARY END ------

RunID: 9a136b230ae989353c5719e4a4da4a91
FAILEDPROJECT EXECUTION %s
SUCCESSFULsetupbeforetestafterteardownframeworkPASSSKIPpint@40004000syscon@40000000clock_controlWEST_TOPDIR/zephyr/include/zephyr/drivers/gpio/gpio_utils.hcallback	No callback!
callback->handler	No callback handler!
cb->handlergpio@1gpio@0WEST_TOPDIR/zephyr/drivers/pinctrl/pinctrl_lpc_iocon.c(((pin_mux) & 0xC0000) >> 18) <= 0x2flexcomm@40086000WEST_TOPDIR/zephyr/include/zephyr/spinlock.hz_spin_lock_valid(l)	Invalid spinlock %p
z_spin_unlock_valid(l)	Not my spinlock %p
WEST_TOPDIR/modules/hal/nxp/mcux/mcux-sdk/devices/LPC54114/drivers/fsl_clock.c "$&()*,-.01245689:;<=WEST_TOPDIR/modules/hal/nxp/mcux/mcux-sdk/devices/LPC54114/drivers/fsl_reset.cbitPos < 32ULWEST_TOPDIR/modules/hal/nxp/mcux/mcux-sdk/drivers/lpc_gpio/fsl_gpio.cport < ((size_t) (((int) sizeof(char[1 - 2 * !(!__builtin_types_compatible_p(__typeof__(s_gpioClockName), __typeof__(&(s_gpioClockName)[0])))]) - 1) + (sizeof(s_gpioClockName) / sizeof((s_gpioClockName)[0]))))WEST_TOPDIR/modules/hal/nxp/mcux/mcux-sdk/drivers/pint/fsl_pint.cinstance < ((size_t) (((int) sizeof(char[1 - 2 * !(!__builtin_types_compatible_p(__typeof__(s_pintBases), __typeof__(&(s_pintBases)[0])))]) - 1) + (sizeof(s_pintBases) / sizeof((s_pintBases)[0]))))base != ((void *)0)WEST_TOPDIR/modules/hal/nxp/mcux/mcux-sdk/drivers/inputmux/fsl_inputmux.cinstance < ((size_t) (((int) sizeof(char[1 - 2 * !(!__builtin_types_compatible_p(__typeof__(s_inputmuxBases), __typeof__(&(s_inputmuxBases)[0])))]) - 1) + (sizeof(s_inputmuxBases) / sizeof((s_inputmuxBases)[0]))))WEST_TOPDIR/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcomm/fsl_flexcomm.ci < (uint32_t)(8)WEST_TOPDIR/modules/hal/nxp/mcux/mcux-sdk/drivers/flexcomm/usart/fsl_usart.c((void *)0) != base((void *)0) != ringBuffer((void *)0) != config!((((void *)0) == base) || (0U == baudrate_Bps) || (0U == srcClock_Hz))!((((void *)0) == base) || (((void *)0) == config) || (0U == srcClock_Hz))WEST_TOPDIR/zephyr/kernel/device.cdev <= _device_list_end	unexpected list end location
%c: Halting system
Unknown errorunknown%c: >>> ZEPHYR FATAL ERROR %d: %s on CPU %d
%c: Fault during interrupt handling

%c: Current thread: %p (%s)
CPU exceptionUnhandled interruptStack overflowKernel oopsKernel panicWEST_TOPDIR/zephyr/kernel/init.cthread_data <= __static_thread_data_list_endidlemainosWEST_TOPDIR/zephyr/kernel/idle.c_kernel.cpus[0].current->base.prio >= 0WEST_TOPDIR/zephyr/kernel/sem.c((arch_is_in_isr() == 0) || ((timeout).ticks == (((k_timeout_t) {0})).ticks))	
WEST_TOPDIR/zephyr/kernel/work.cwork->queue != ((void *)0)work != ((void *)0)work->handler != ((void *)0)queuestack!flag_test(&queue->flags, K_WORK_QUEUE_STARTED_BIT)WEST_TOPDIR/zephyr/kernel/thread.c((((prio)) == 15 && z_is_idle_thread_entry((entry))) || (((15 - 1) >= ((-16))) && ((prio)) >= ((-16)) && ((prio)) <= (15 - 1)))	invalid priority (%d); allowed range: %d to %d
!arch_is_in_isr()	Threads may not be created in ISRs
WEST_TOPDIR/zephyr/kernel/sched.c!z_is_idle_thread_object(thread)WEST_TOPDIR/zephyr/kernel/include/kthread.h_kernel.cpus[0].current != ((void *)0)thread->base.pended_on!terminate	aborted _current back from dead
[... 193 lines cut ...]
tp
`\
LD	9 84
	 L< L< #[4 $MX
1)#@Z?16K1=\J`S!(&)^3M**#[4 $MX
1)#@Z?16K1=**y]`m`  XA XA `A `A  A A A A A A A A A A A A A A B B B B pp(rqqA  	pA qA qA qA qA qA (n xrdrPr pr  	r( oorpprprpxndnPnrprptq`q\pLqHppB <r8q4p0 , $qB  r 	H 	 	s	

 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~	

 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x40000
E: r0/a1:  0x000000ff  r1/a2:  0x00000016  r2/a3:  0x00000016
E: r3/a4:  0x00002ccd r12/ip:  0x00000000 r14/lr:  0x000073b5
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x00006fce
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20000520 (main)
E: Halting system

    
```**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v3.7.0-4826-g372b3a9b8da2

dleach02, 2024-11-01:
@hakehuang as Anas is pointing out, this test path doesn't exist?

hakehuang, 2025-01-02:
> > cd tests/kernel/objects/minimallibc/pending_timer
> 
> that path does not exist!

sorry, I extra path form the case name, which is wrong, let me update

hakehuang, 2025-01-02:
> @hakehuang as Anas is pointing out, this test path doesn't exist?

updated

EmilioCBen, 2025-02-21:
The twister command doesn't seem to work for me, but I tried the test and it seems to pass, is an error still showing up on your side?

hakehuang, 2025-02-24:
@EmilioCBen , thanks for checking, this issue is fixed in latest main 4.1.0-rc2

closed 2025-02-24 by hakehuang