zephyrproject-rtos/zephyr #80364: logging: Logs with more than 4 '%s' arguments cause out-of-bounds memory access

jp-pacotte, 2024-10-24:
**Describe the bug**

Logs with more than 4 '%s' arguments cause out-of-bounds memory access and may trigger -ENOSPC in the z_cbprintf_cpy function from cbprintf.h.

**To Reproduce**

Set CONFIG_DEBUG_OPTIMIZATIONS=n

``
LOG_INF("Flash board configuration: productId: %s, groupId: %s, id: %s, versionMajor: %s, "
            "versionMinor: %s, variant: %s",
            boardConfiguration->productId, boardConfiguration->groupId, boardConfiguration->id, boardConfiguration->versionMajor,
            boardConfiguration->versionMinor, boardConfiguration->variant);
```
**Expected behavior**

The root cause lies in the size of uint16_t strl[4]; in z_impl_z_log_msg_static_create, which is not validated by the loop in cbprintf_packaged.c during the str_len calculation.

If scpy_cnt > 4, then str_len goes out of bounds of strl.

```
   /* Append strings */
    for (unsigned int i = 0; i < scpy_cnt; i++) {
        uint8_t loc = cpy_str_pos[i];
        const char *str = *(const char **)&buf32[loc];
        uint16_t str_len = strl ? strl[i] : 0;

        rv = cb(&loc, 1, ctx);
        if (rv < 0) {
            return rv;
        }
        out_len += rv;

        rv = append_string(cb, ctx, str, str_len);
        if (rv < 0) {
            return rv;
        }
        out_len += rv;
    }
```

Should an ASSERT or KCONFIG be added to check the size of strl?

**Impact**

The application crashes.

**Logs and console output**

**Environment (please complete the following information):**

Zephyr 3.7

**Additional context**

mmahadevan108, 2024-10-29:
@jp-pacotte can you submit a PR.

jp-pacotte, 2024-10-30:
I don't know what is the right strategy for this bug
 - Keep size limitation at 4 but add an ASSERT
 - Add a KCONFIG for this size

nordic-krch, 2024-10-30:
I've opened the bugfix PR.

nordic-krch, 2024-10-30:
the size should stay as it is. It is just a helper array to avoid calling `strlen` on the same string twice (when package conversion is done in 2 steps: package length calculation followed by actual package conversion). If array is too short then `strlen` shall be called twice on the same string in both steps (and that is acceptable). 4 seems ok since it is that rarely to use more than 4 `%s` arguments that we hit this issue now, few years after bug was in the code.

PR #80616 (2024-11-01): lib: os: cbprintf_packaged: Fix cbprintf_package_convert
 When conversion is done in two steps (output length calculation followed by the actual conversion), it was failing when helper array for holding calculated string length from the first conversion run was shorter than number of strings that were supposed to be appended to the package. Fix by taking into account strl_len to determine if length can be taken from the array or need to be calculated again.

Fixes #80364
closed 2024-11-01 by mmahadevan108