zephyrproject-rtos/zephyr #80383: drivers: flash: spi_nor: values vom DT not correctly used, when `CONFIG_SPI_NOR_SFDP_RUNTIME=y`

maass-hamburg, 2024-10-24:
**Describe the bug**

when `CONFIG_SPI_NOR_SFDP_RUNTIME=y`, configs from the dts like the wp, hold and reset pins are not parsed correctly into the `struct spi_nor_config` and therefore can't be used by the driver.

**To Reproduce**
set `CONFIG_SPI_NOR_SFDP_RUNTIME=y`, when using the jedec,spi-nor flash driver

**Expected behavior**
dts configuration is used correctly

**Impact**
most of the available configuration, that can be done via the dts wont work when `CONFIG_SPI_NOR_SFDP_RUNTIME=y`,

**Logs and console output**

**Environment (please complete the following information):**
NA

**Additional context**
Got introduced with https://github.com/zephyrproject-rtos/zephyr/pull/70415

de-nordic, 2024-10-25:
Bug level set to low, seems to go mainly undetected by users.

alexgoldstone, 2024-10-25:
This just bit me for several hours so its definitely impacting more than just the original reporter.

de-nordic, 2024-10-25:
> This just bit me for several hours so its definitely impacting more than just the original reporter.

Sorry. Have you reported that? It is hard to evaluate user impact without reports.

alexgoldstone, 2024-10-25:
No, it literally just occurred as an issue for us today and once I identified the issues I searched to see if anyone else had the same issue and commented above. Is there somewhere else I should flag this also please?

de-nordic, 2024-10-26:
Ok, so you are the second person to report this since it went in as part of 3.7.0.
This is the right place to report it.

PR #80365 (2024-10-30): drivers: flash: spi_nor: fix config struct
lots of values from the DT where not set corretly, if
`CONFIG_SPI_NOR_SFDP_RUNTIME` was
enabled. This fixes it.

Got introduced with https://github.com/zephyrproject-rtos/zephyr/pull/70415

Fixes #80383
closed 2024-10-30 by dkalowsk