zephyrproject-rtos/zephyr #80386: PM policy: definition of the end of a registered event

hubertmis, 2024-10-24:
**Is your enhancement proposal related to a problem? Please describe.**

There is a set of function intended to be used to register low latency events:
https://github.com/zephyrproject-rtos/zephyr/blob/3c4fcaa2c0e2c525d416ae248574a2fb505eff7b/include/zephyr/pm/policy.h#L161

In the current implementation, the RTOS makes sure that the CPU is in a low latency mode at the time registered by the function. But one cycle later, the idle thread is allowed to enter any sleep state causing any latency.

The user of the API provides the `cycle` value according to its best knowledge. However, because of clocks drifts, asynchronous clocks synchronization, etc it's not always possible to provide the value with 1 cycle accuracy and the caller must add a margin.

When caller adds a margin and idle thread kicks in during the margin, the CPU enters a sleep state, and the event gets handled with unacceptable latency.

**Describe the solution you'd like**

I would like to define in the policy API how long the requested low-latency state is applied. Potentially something like:

```
 * CPU is woken up before the time passed in cycle to prevent the event handling
 * latency. The CPU is kept active until the event is deregistered with
 * @ref pm_policy_event_unregister
```

Following this clarification, the implementation would need to be modified to meet the new requirement.

New procedure should be added around the wake-up sequence: https://github.com/zephyrproject-rtos/zephyr/blob/1e83368d88222ae573fe5ff6ad1b7b8187cee554/subsys/pm/pm.c#L227-L229

Something like:
```
if (woken_up_because_of_a_registered_event()) {
  pm_policy_latency_request_add(&reg_evt_req, 0);
}
```

And another one around https://github.com/zephyrproject-rtos/zephyr/blob/1e83368d88222ae573fe5ff6ad1b7b8187cee554/subsys/pm/policy/policy_events.c#L93-L101

```
if (event_recently_fired_was(evt)) {
  pm_policy_latency_request_remove(&reg_evt_req);
}
```

To fully support this solution, there should be also a system timer associated with the next event, that would call `pm_policy_latency_request_add(...)` if the event is expected to occur while the CPU is active.

**Describe alternatives you've considered**

Alternatively, we could extend `pm_policy_event_register()` with a parameter `duration` and another timer automatically removing the low-latency request. However, it seems more complicated, less-intuitive, and brings no additional value.

**Additional context**

The proposed implementation seems overlapping with #79730

bjarki-andreasen, 2024-10-25:
mention @gmarull @JordanYates

ceolin, 2024-10-26:
@hubertmis that is a good point. We already expect users to unregister an event, I am wondering if instead of requesting more information from the user (like the proposed `duration`) we can just just set the latency `pm_policy_latency_request_add(&reg_evt_req, 0);` before invoke the policy and remove it when the cpu wakes up, do it until the event is unregistered.

The other possibility is to just keep the cpu active until a past event is still registered ? That is the simplest one for sure and is already expected the user to unregister the event ...

bjarki-andreasen, 2024-10-28:
I will put together a PR which attempts:

> The other possibility is to just keep the cpu active until a past event is still registered ? That is the simplest one for sure and is already expected the user to unregister the event ...

bjarki-andreasen, 2024-10-28:
@ceolin what is the reason for using the absolute cycle count for events rather than kernel ticks? `pm_system_suspend()` takes kernel ticks, so the resolution seems to tied to kernel ticks res rather than the cycle res.

I would like to update pm_policy to use absolute kernel ticks for events to avoid dealing with wrapping of the counter, since the wrapping makes it really hard to see if an event happened in the past or will happen later :)

PR #80580 (2024-12-09): PM Policy Event: Update to use uptime ticks and be "sticky"
Registering a pm policy event will now represent a single event in absolute time, which will remain active, preventing the system from suspending, until the event has been handled. To indicate an event has been handled, it must be updated to a time in the future, or unregistered.

To greatly simplify the events and the "sticky" logic, the time primitive used is updated to the monotonic uptime in ticks. This ensures an event can be determined to be in the past, since unlike the HW cycle, uptime can't wrap. Additionally, the kernel work [... cut]
closed 2024-12-09 by kartben