zephyrproject-rtos/zephyr #80409: Zephyr defines POSIX values which already appear in limits.h

keith-packard, 2024-10-25:
**Describe the bug**
Upstream picolibc includes definitions for various minimum values in limits.h that are required by POSIX including `_POSIX_NAME_MAX` and `_POSIX_PATH_MAX`. Zephyr is redefining these in include/zephyr/posix/posix_features.h.

**To Reproduce**
1. Download toolchain built against current picolibc development branch: https://github.com/zephyrproject-rtos/sdk-ng/actions/runs/11503233418/artifacts/2100796287
2.  west twister -p qemu_malta -s tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc
3. See error

**Expected behavior**
Successful test run

**Impact**
Cannot migrate picolibc to upstream development version

**Logs and console output**
```
[81/147] Building C object zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj
FAILED: zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj 
ccache /__w/sdk-ng/sdk-ng/tools/zephyr-sdk-0.17.0-8-g6fd2e1d/mips-zephyr-elf/bin/mips-zephyr-elf-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=1024 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DTC_RUNID=020f12a965888084e31d7f59ce1d46d3 -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/kernel/include -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/arch/mips/include -I/__w/sdk-ng/sdk-ng/test/twister-out/qemu_malta/tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc/zephyr/include/generated/zephyr -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include -I/__w/sdk-ng/sdk-ng/test/twister-out/qemu_malta/tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc/zephyr/include/generated -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/soc/qemu/malta -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/posix -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/soc/qemu/malta/. -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/subsys/testsuite/include -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/subsys/testsuite/coverage -I/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/subsys/testsuite/ztest/include -isystem /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/lib/libc/common/include -Os -DNDEBUG -Wshadow -fno-strict-aliasing -Werror -Os -imacros /__w/sdk-ng/sdk-ng/test/twister-out/qemu_malta/tests/lib/c_lib/thrd/libraries.libc.c11_threads.picolibc/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -EL -msoft-float -G0 -mno-gpopt -fno-pic -pipe --sysroot=/__w/sdk-ng/sdk-ng/tools/zephyr-sdk-0.17.0-8-g6fd2e1d/mips-zephyr-elf/mips-zephyr-elf -imacros /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/tests/lib/c_lib/thrd=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/sdk-ng/sdk-ng/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/sdk-ng/sdk-ng/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -D_POSIX_THREADS -mips32 -std=c99 -MD -MT zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj -MF zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj.d -o zephyr/lib/posix/options/CMakeFiles/lib__posix__options.dir/sysconf.c.obj -c /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/lib/posix/options/sysconf.c
In file included from /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/posix/signal.h:10,
                 from /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/posix/time.h:62,
                 from /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/posix/pthread.h:14,
                 from /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/lib/posix/options/sysconf.c:9:
/__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/posix/posix_features.h:241: error: "_POSIX_AIO_LISTIO_MAX" redefined [-Werror]
  241 | #define _POSIX_AIO_LISTIO_MAX               (2)
      | 
In file included from /__w/sdk-ng/sdk-ng/tools/zephyr-sdk-0.17.0-8-g6fd2e1d/mips-zephyr-elf/picolibc/include/limits.h:6,
                 from /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/kernel_includes.h:22,
                 from /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/kernel.h:17,
                 from /__w/sdk-ng/sdk-ng/zephyrproject/zephyr/include/zephyr/posix/pthread.h:13:
/__w/sdk-ng/sdk-ng/tools/zephyr-sdk-0.17.0-8-g6fd2e1d/mips-zephyr-elf/picolibc/include/sys/syslimits.h:63: note: this is the location of the previous definition
   63 | #define _POSIX_AIO_LISTIO_MAX   2
```
**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK development version

keith-packard, 2024-10-25:
> I would guess that an `ifndef` might not be sufficient here, since that value should probably be configurable via Kconfig (if it isn't already).
> 
> What are your thoughts here @keith-packard ?

All of these constants actually have fixed values defined by POSIX, so `#ifndef` would be completely fine. You probably only need to check one of them?

cfriedt, 2024-10-25:
Just landed 🛬 - this should be easy enough to fix quickly.

keith-packard, 2024-10-25:
as we both try to track the C23 changes in parallel. Thanks for taking a look; hope your travel is going well.

cfriedt, 2025-01-29:
Is this actually a medium priority bug? The fix has been available for 3 months already and has not been merged.

Jannik-M91, 2025-06-03:
I just tried to update my Zephyr toolchain to the latest [release v0.17.1](https://github.com/zephyrproject-rtos/sdk-ng/releases/tag/v0.17.1) which uses Picolibc release 1.8.10.

Building my project now fails with the same messages as seen in the original ticket:

```
zephyr/include/zephyr/posix/posix_features.h:252: error: "_POSIX_AIO_LISTIO_MAX" redefined [-Werror]
  252 | #define _POSIX_AIO_LISTIO_MAX               (2)
      |
In file included from c:\users\mahrjannik\zephyr-sdk-0.17.1\arm-zephyr-eabi\picolibc\include\limits.h:6,
                 from C:/Users/MahrJannik/projects/proton-os-workspace/zephyr/include/zephyr/kernel_includes.h:22,
                 from C:/Users/MahrJannik/projects/proton-os-workspace/zephyr/include/zephyr/kernel.h:17,
                 from C:/Users/MahrJannik/projects/proton-os-workspace/zephyr/include/zephyr/posix/posix_types.h:30,
                 from C:/Users/MahrJannik/projects/proton-os-workspace/zephyr/include/zephyr/posix/unistd.h:9:
c:\users\mahrjannik\zephyr-sdk-0.17.1\arm-zephyr-eabi\picolibc\include\sys\syslimits.h:63: note: this is the location of the previous definition
   63 | #define _POSIX_AIO_LISTIO_MAX   2
```

This makes toolchain release v0.17.1 unusable for me. I am a little bit confused that I could not find any other complains about this yet.

gbmhunter, 2025-07-01:
Ditto, upgrading from `v0.17.0` to `0.17.1` of the SDK gave me the same error(s).

cfriedt, 2025-09-09:
Fixed-by https://github.com/zephyrproject-rtos/zephyr/pull/94612

closed 2025-09-09 by cfriedt