zephyrproject-rtos/zephyr #80413: `drivers/ethernet/phy/phy_mii.c`: logs additional `\n`s

clamattia, 2024-10-25:
**Describe the bug**
`drivers/ethernet/phy/phy_mii.c`: logs additional `\n`s

**To Reproduce**
Run a sample that uses `phy_mii`.

**Expected behavior**
Every log statement on a single line, like the rest of `zephyr` without additional spacing.

**Impact**
Low, mostly aesthetic.

**Logs and console output**
Taken from the docs `samples/boards/nxp/s32/netc/README.rst` :
```
[00:00:00.051,000] <inf> phy_mii: PHY (7) ID 1CC916

[00:00:00.052,000] <inf> nxp_s32_eth_psi: SI0 MAC: 00:00:00:01:02:00
[00:00:00.058,000] <inf> shell_telnet: Telnet shell backend initialized
[00:00:00.058,000] <inf> nxp_s32_netc_sample: Starting sample
[00:00:00.058,000] <inf> nxp_s32_netc_sample: Waiting for iface 1 to come up
[00:00:07.595,000] <inf> phy_mii: PHY (7) Link speed 1000 Mb, full duplex

[00:00:07.595,000] <inf> nxp_s32_netc_sample: Configuring iface 1 (0x318008f0)
[00:00:07.595,000] <inf> nxp_s32_netc_sample: IPv6 address: 2001:db8::1
[00:00:07.595,000] <inf> nxp_s32_netc_sample: IPv4 address: 192.0.2.1
```

**Additional context**
Example:
```
	LOG_INF("PHY (%d) Link speed %s Mb, %s duplex\n",
		cfg->phy_addr,
		PHY_LINK_IS_SPEED_1000M(data->state.speed) ? "1000" :
		(PHY_LINK_IS_SPEED_100M(data->state.speed) ? "100" : "10"),
		PHY_LINK_IS_FULL_DUPLEX(data->state.speed) ? "full" : "half");
```

**Alternatives**
This is intentional and documented.

jukkar, 2024-10-25:
@clamattia would you be able to send a PR fixing this?

clamattia, 2024-10-25:
> @clamattia would you be able to send a PR fixing this?

yes

clamattia, 2024-10-25:
> @clamattia would you be able to send a PR fixing this?

@jukkar done

closed 2024-10-25 by mmahadevan108