zephyrproject-rtos/zephyr #80417: Extend Ethernet HAL for STM32 H5 and H7 MCU series

robi251, 2024-10-25:
**Is your enhancement proposal related to a problem? Please describe.**

I would like to use STM32F4 series MCU in my project. Now the STM32 Ethernet HAL is supporting the H5 and H7 series of STM32 MCU's.

**Describe the solution you'd like**

After review of file `eth_stm32_hal.c` i noticed a lot of compilation conditions like
`#if defined(CONFIG_SOC_SERIES_STM32H7X) || defined(CONFIG_SOC_SERIES_STM32H5X)`

I reviewed this file and it looks, that if I extend it like this:
`#if defined(CONFIG_SOC_SERIES_STM32H7X) || defined(CONFIG_SOC_SERIES_STM32H5X) || defined(CONFIG_SOC_SERIES_STM32F4X`
then my project is compiled and functioning well. Does this extension make sense? 

**Describe alternatives you've considered**

The alternative is to create eth_stm32_hal_f4x.c or similar with help of STM32 Cube MX just for this MCU series

**Additional context**

closed 2024-10-25 by erwango

robi251, 2024-10-26:
I think it's a misunderstand. Maybe I was not clear enough. The thing is to leverage the ability of the MDIO API in the Ethernet communication. This part is available now only for STM32 H5/H7 series. Can we reopen this topic?

erwango, 2024-11-04:
> @erwango I think it's a misunderstand. Maybe I was not clear enough. The thing is to leverage the ability of the MDIO API in the Ethernet communication. This part is available now only for STM32 H5/H7 series. Can we reopen this topic?

Please open a new issue with an accurate title.

robi251, 2024-11-06:
It's not required anymore, since this was noticed already in 3.7 kernel. This can be close.