zephyrproject-rtos/zephyr #80423: samples/sensor/ for trigger cannot run without trigger

FRASTM, 2024-10-25:
**Describe the bug**
The PR https://github.com/zephyrproject-rtos/zephyr/pull/77335  added samples to trig accel and magn sensors.
It appears that the stm32f3_disco board has both 

- lsm303dlhc_accel
- lsm303dlhc_magn

but has no trigger
`#define CONFIG_LIS2DH_TRIGGER_NONE 1
`

Consequently the samples
- samples/sensor/accel_trig/
- samples/sensor/magn_trig/

are not relevant

**To Reproduce**
1. west build  -p always -b stm32f3_disco samples/sensor/accel_trig/
2. west flash
3. See error

**Expected behavior**
Do not execute this  application for this platform even if dt_alias_exists("accel0") or dt_alias_exists("magn0") is true
  filter through the sample.yaml 

**Impact**
CI failure

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-4635-g3978693d474f ***
Could not set trigger

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows) ubuntu 
 - Toolchain (e.g Zephyr SDK, ...) 0.16.8
 - Commit SHA or Version used  dfbcea777a10e27d12539a4fd85b5203f5421151

**Additional context**
```patch
diff --git a/samples/sensor/accel_trig/sample.yaml b/samples/sensor/accel_trig/sample.yaml
index 0cdbd066781..a404f5ec887 100644
--- a/samples/sensor/accel_trig/sample.yaml
+++ b/samples/sensor/accel_trig/sample.yaml
@@ -12,3 +12,4 @@ tests:
           \\(\\s*-?[0-9\\.]*,\\s*-?[0-9\\.]*,\\s*-?[0-9\\.]*\\)$"
     integration_platforms:
       - frdm_k64f                       # fxos8700
+    platform_exclude: stm32f3_disco
diff --git a/samples/sensor/magn_trig/sample.yaml b/samples/sensor/magn_trig/sample.yaml
index 3f010b7099d..f0f808c8fb6 100644
--- a/samples/sensor/magn_trig/sample.yaml
+++ b/samples/sensor/magn_trig/sample.yaml
@@ -12,3 +12,4 @@ tests:
           \\(\\s*-?[0-9\\.]*,\\s*-?[0-9\\.]*,\\s*-?[0-9\\.]*\\)$"
     integration_platforms:
       - frdm_k64f                       # fxos8700
+    platform_exclude: stm32f3_disco
```

mmahadevan108, 2024-10-29:
@FRASTM , can you please submit a PR with your proposal.

FRASTM, 2024-11-04:
> @FRASTM , can you please submit a PR with your proposal.

Here it is https://github.com/zephyrproject-rtos/zephyr/pull/80842

closed 2024-11-22 by kartben