zephyrproject-rtos/zephyr #80424: intel_adsp_ace: tests/kernel/timer/timer_behavior/kernel.timer.timer ACE15/20/30 sims build failure

mdubielx, 2024-10-25:
**Describe the bug**

tests/kernel/timer/timer_behavior build failure for all ace sims
ace15_mtpm/ace20_lnl/ace30_ptl

**To Reproduce**

west twister -p intel_adsp/ace15_mtpm -T tests/kernel/timer/timer_behavior/kernel.timer.timer  --build-only

**Expected behavior**

build pass

**Impact**

kernel time API

**Logs and console output**

```
/opt/toolchains/xtensa/XtDevTools/install/tools/RI-2022.10-linux/XtensaTools/bin/xt-clang -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DTC_RUNID=c604bb2540c1da4d041e2079cf7a2b0d -D__LINUX_ERRNO_EXTENSIONS__ -D__ZEPHYR__=1 -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/twister-out/intel_adsp_ace20_lnl_sim/tests/kernel/timer/timer_behavior/kernel.timer.timer/zephyr/include/generated/zephyr -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/twister-out/intel_adsp_ace20_lnl_sim/tests/kernel/timer/timer_behavior/kernel.timer.timer/zephyr/include/generated -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/soc/intel/intel_adsp -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/soc/intel/intel_adsp/common -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/soc/intel/intel_adsp/common/include -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/soc/intel/intel_adsp/ace -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/soc/intel/intel_adsp/ace/include -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/soc/intel/intel_adsp/ace/include/ace20_lnl -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/subsys/testsuite/include -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/subsys/testsuite/coverage -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/subsys/testsuite/ztest/include -I/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/drivers -isystem /__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/lib/libc/common/include -isystem /__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/twister-out/intel_adsp_ace20_lnl_sim/tests/kernel/timer/timer_behavior/kernel.timer.timer/modules/picolibc/picolibc/include -fno-strict-aliasing -Werror -Os -imacros /__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/twister-out/intel_adsp_ace20_lnl_sim/tests/kernel/timer/timer_behavior/kernel.timer.timer/zephyr/include/generated/zephyr/autoconf.h -fno-common -g -gdwarf-4 -fcolor-diagnostics -fms-extensions -imacros /__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-typedef-redefinition -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wno-sometimes-uninitialized -Wno-shift-overflow -Wno-missing-braces -Wno-self-assign -Wno-address-of-packed-member -Wno-unused-function -Wno-initializer-overrides -Wno-section -Wno-unknown-warning-option -Wno-unused-variable -Wno-format-invalid-specifier -Wno-gnu -Wno-tautological-compare -Werror=implicit-int -fno-asynchronous-unwind-tables -Wno-vla -fmacro-prefix-map=/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/tests/kernel/timer/timer_behavior=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/os.rtos.zephyr.zephyr=WEST_TOPDIR -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT CMakeFiles/app.dir/src/jitter_drift.c.obj -MF CMakeFiles/app.dir/src/jitter_drift.c.obj.d -o CMakeFiles/app.dir/src/jitter_drift.c.obj -c /__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:212:21: error: implicit conversion increases floating-point precision: 'float' to 'double' [-Werror,-Wdouble-promotion]
        double stddev_us = sqrtf(variance_us);
               ~~~~~~~~~   ^~~~~~~~~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:213:22: error: implicit conversion increases floating-point precision: 'float' to 'double' [-Werror,-Wdouble-promotion]
        double stddev_cyc = sqrtf(variance_cyc);
               ~~~~~~~~~~   ^~~~~~~~~~~~~~~~~~~
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:336:15: error: using integer absolute value function 'abs' when argument is of floating point type [-Werror,-Wabsolute-value]
        zassert_true(abs(time_diff_us) < CONFIG_TIMER_TEST_MAX_DRIFT,
                     ^
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:336:15: note: use function 'fabs' instead
        zassert_true(abs(time_diff_us) < CONFIG_TIMER_TEST_MAX_DRIFT,
                     ^~~
                     fabs
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/subsys/testsuite/ztest/include/zephyr/ztest_assert.h:275:41: note: expanded from macro 'zassert_true'
#define zassert_true(cond, ...) zassert(cond, #cond " is false", ##__VA_ARGS__)
                                        ^
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/subsys/testsuite/ztest/include/zephyr/ztest_assert.h:194:14: note: expanded from macro 'zassert'
        _zassert_va(cond, default_msg, COND_CODE_1(__VA_OPT__(1), (__VA_ARGS__), (NULL)))
                    ^
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/subsys/testsuite/ztest/include/zephyr/ztest_assert.h:191:16: note: expanded from macro '_zassert_va'
        _zassert_base(cond, default_msg, msg, ##__VA_ARGS__)
                      ^
/__w/os.rtos.zephyr.zephyr/os.rtos.zephyr.zephyr/subsys/testsuite/ztest/include/zephyr/ztest_assert.h:180:14: note: expanded from macro '_zassert_base'
                        z_zassert(cond, _msg ? ("(" default_msg ")") : (default_msg), __FILE__,    \
                                  ^
3 errors generated.
[18/104] Building C object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/irq_manage.c.obj
[19/104] Building C object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/cpu_idle.c.obj
[20/104] Building C object zephyr/soc/soc/ace20_lnl/common/CMakeFiles/intel_adsp_common.dir/soc.c.obj
[21/104] Building C object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/thread.c.obj
[22/104] Building C object zephyr/arch/common/CMakeFiles/isr_tables.dir/isr_tables.c.obj
[23/104] Building C object zephyr/soc/soc/ace20_lnl/common/CMakeFiles/intel_adsp_common.dir/rimage_modules.c.obj
[24/104] Building C object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/xtensa_backtrace.c.obj
[25/104] Building C object zephyr/soc/soc/ace20_lnl/common/CMakeFiles/intel_adsp_common.dir/boot_complete.c.obj
[26/104] Building C object zephyr/arch/common/CMakeFiles/arch__common.dir/multilevel_irq.c.obj
[27/104] Building C object zephyr/soc/soc/ace20_lnl/common/CMakeFiles/intel_adsp_common.dir/mem_window.c.obj
[28/104] Building C object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/fatal.c.obj
[29/104] Building C object zephyr/lib/libc/common/CMakeFiles/lib__libc__common.dir/source/stdlib/malloc.c.obj
[30/104] Building C object zephyr/soc/soc/ace20_lnl/common/CMakeFiles/intel_adsp_common.dir/multiprocessing.c.obj
[31/104] Building C object zephyr/subsys/testsuite/ztest/CMakeFiles/subsys__testsuite__ztest.dir/src/ztest_error_hook.c.obj
[32/104] Building C object zephyr/lib/libc/picolibc/CMakeFiles/lib__libc__picolibc.dir/libc-hooks.c.obj
[33/104] Building C object zephyr/soc/soc/ace20_lnl/common/CMakeFiles/intel_adsp_common.dir/boot.c.obj
[34/104] Building C object zephyr/soc/soc/ace20_lnl/common/CMakeFiles/intel_adsp_common.dir/ipc.c.obj
[35/104] Building C object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/vector_handlers.c.obj
[36/104] Building C object zephyr/drivers/interrupt_controller/CMakeFiles/drivers__interrupt_controller.dir/intc_dw_ace.c.obj
[37/104] Building C object zephyr/subsys/testsuite/ztest/CMakeFiles/subsys__testsuite__ztest.dir/src/ztest.c.obj
[38/104] Building C object zephyr/subsys/testsuite/ztest/CMakeFiles/subsys__testsuite__ztest.dir/src/ztest_rules.c.obj
[39/104] Building C object zephyr/subsys/testsuite/ztest/CMakeFiles/subsys__testsuite__ztest.dir/src/ztest_defaults.c.obj
[40/104] Building C object zephyr/drivers/console/CMakeFiles/drivers__console.dir/winstream_console.c.obj
ninja: build stopped: subcommand failed.
```

**Environment:**
 
- Linux Ubuntu 22.04
- Toolchain: Zephyr SDK 16.8
- Zephyr version: `v3.7.0-5108-g1a89d4b13e41`

nashif, 2024-10-25:
> west twister -p intel_adsp/ace15_mtpm -T tests/kernel/timer/timer_behavior/kernel.timer.timer --build-only

that command line is wrong.

Using `west twister -p intel_adsp/ace15_mtpm  -T tests/kernel/timer/timer_behavior` on main, builds fine, also on v3.7.0-5108-g1a89d4b13e41

nashif, 2024-10-25:
oh, you also wrongly specify `Toolchain: Zephyr SDK 16.8`, it seems you are using another toolchain.

mdubielx, 2024-10-25:
> oh, you also wrongly specify `Toolchain: Zephyr SDK 16.8`, it seems you are using another toolchain.

copy-paste... it should be "xt-clang"

PR #80534 (2024-10-30): tests: timer_behavior: fix double float mismatch
* Use `sqrt()` instead of `sqrtf()` as arguments and resulting variables are all of doubles.
* Use `fabs()` instead of `abs()` as the argument is of double.

Note that the issue can be seen via LLVM/Clang: `west build -p -b native_sim tests/kernel/timer/timer_behavior/`

Fixes #80424
closed 2024-10-30 by mmahadevan108