zephyrproject-rtos/zephyr #80427: BLE controller doesn't support the size more than 191 Bytes in Advertising extended mode

j-delsol-dev, 2024-10-25:


rruuaanng, 2024-10-25:
That is to say, you expect the size of `adv_data` to be 255.

cvinayak, 2024-10-25:
Please refer to the conf used in this test for your needs : https://github.com/zephyrproject-rtos/zephyr/blob/8cd361a51a4e7954d58ca7f06ed554bd3a7ef076/tests/bsim/bluetooth/host/adv/chain/prj.conf#L8-L20

cvinayak, 2024-10-28:
> @cvinayak , maybe we should set the default of CONFIG_BT_CTLR_ADV_DATA_CHAIN to y when the configured data length is greater than 191?

The feature is experimental in Zephyr Controller, and only tested in the context of Host feature coverage, there are no specific test coverage of the implementation in the Controller. The Controller has design listing upto Direction Finding support with only 191 bytes of maximum Advertising Data, chaining feature only for the purpose of CTE.

Also, with the Kconfig enabled, advertising data > 191 is not accepted in AUX_ADV_IND (to allow room for possible ACAD updates) and will split over to AUX_CHAIN_IND. And I believe thereafter data received over HCI is recombined into additional AUX_CHAIN_IND PDUs.

I believe this feature in Zephyr Controller should remain experimental until more tests be added.

cvinayak, 2024-10-29:
@j-delsol-dev did you try the Observer sample with `-DCONF_FILE=prj_extended.conf`

https://github.com/zephyrproject-rtos/zephyr/blob/9f93dede369c365c68b5618844e3d31e35327bc2/samples/bluetooth/observer/prj_extended.conf#L1-L17

j-delsol-dev, 2024-10-31:
it works with this config, thank you.

cvinayak, 2024-11-02:
@j-delsol-dev Please close this issue if you are able to advertise and scan successfully.

closed 2024-11-04 by j-delsol-dev