zephyrproject-rtos/zephyr #80437: tests/net/wifi/configs/wifi.build.* fail due to build warnings in native_sim//64

aescolar, 2024-10-25:
**Describe the bug**
tests/net/wifi/configs/wifi.build.* fail in twister due to a lot of trivial build warnings in native_sim/native/64

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p native_sim/native/64 -T tests/net/wifi/
4. See error

**Expected behavior**
No build warnings & therefore no twister failures

**Impact**
Theses tests failing in 64 bit platforms

**Logs and console output**
A few tenths of warnings like
```
modules/lib/hostap/hostapd/hostapd_cli_zephyr.c:269:47: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘size_t’ {aka ‘long unsigned int’} [-Wformat=]
  269 |                         wpa_printf(MSG_DEBUG, "Received len: %d, msg_len:%d - %s->END\n",
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
A couple like
```
modules/lib/hostap/wpa_supplicant/wpa_cli_zephyr.c:506:30: error: passing argument 1 of ‘supp_make_argv’ from incompatible pointer type [-Wincompatible-pointer-types]
  506 |         (void)supp_make_argv(&argc, &argv[0], cmd, MAX_ARGS);
      |                              ^~~~~
      |                              |
      |                              int *
```
And 
```
modules/lib/hostap/src/ap/ap_drv_ops.c:272:41: warning: overflow in conversion from ‘long unsigned int’ to ‘int’ changes value from ‘18446744073709551614’ to ‘-2’ [-Woverflow]
  272 |                                      0, ~WPA_STA_AUTHORIZED);
```

**Environment (please complete the following information):**

 - OS: Ubuntu Linux 24.04
 - Toolchain (e.g Zephyr SDK, ...): Host gcc 14.2
 - Commit SHA or Version used: Main as of now 95cc5f53b82cbc87da1e0f8f402bb1430bda9d66 

**Additional context**

jukkar, 2024-10-25:
Weird, I just wonder how come these were not seen earlier when the test PR was merged?
Anyway, could we just disable testing with 64 bit target at this point.

aescolar, 2024-10-25:
> Weird, I just wonder how come these were not seen earlier when the test PR was merged?

The 64 bit native_sim is not a "default" test platform, the 32 bit is. In PRs we only run in the default ones to save time.

> Anyway, could we just disable testing with 64 bit target at this point.

That's fine for me :)

jukkar, 2024-10-25:
> The 64 bit native_sim is not a "default" test platform, the 32 bit is. In PRs we only run in the default ones to save time.

Ok, I did not realize that. I should have not put the 64 bit version in the allowed list in the first place :facepalm:

aescolar, 2024-10-25:
> I should have not put the 64 bit version in the allowed list in the first place 🤦

I think it is a good idea to let code be tested for 64bit platforms :). I don't think you did anything wrong allowing it.

jukkar, 2024-10-25:
Sure, I just thought the test would run with native/64 target automatically, in which case I would have seen these issues myself earlier. I did run twister locally but of course only with 32-bit native-sim board.

PR #80442 (2024-10-25): tests: net: wifi: Disable native/64 from tests
There are issues with 64 bit compilation in hostap so disable native_sim/native/64 board from the tests. The board will be enabled again after updating the hostap from upstream and fixing the issues.

Fixes #80437
commit ea099b03 (2024-10-25): tests: net: wifi: Disable native/64 from tests

There are issues with 64 bit compilation in hostap so disable
native_sim/native/64 board from the tests. The board will be
enabled again after updating the hostap from upstream and fixing
the issues.

Fixes #80437
closed 2024-10-25 by dleach02