zephyrproject-rtos/zephyr #80463: ls0xx: Sharp memory Disaplay cannot run samples/drivers/display

ric96, 2024-10-25:
I am trying to run the Adafruit Sharp memory display 2.7 inch on a STM32H7B0 board but cannot run the display driver sample.

here's the dt snippet I used:

```
&spi2 {
	pinctrl-0 = <&spi2_sck_pb10 &spi2_miso_pc2_c &spi2_mosi_pc1>;
	cs-gpios = <&gpioc 0 GPIO_ACTIVE_HIGH>;
	pinctrl-names = "default";
	status = "okay";

	ls0xx_ls027b7dh01: ls0xx@0 {
		compatible = "sharp,ls0xx";
		spi-max-frequency = <2000000>;
		reg = <0>;
		width = <400>;
		height = <240>;
		extcomin-gpios = <&gpioc 3 GPIO_ACTIVE_HIGH>;
		extcomin-frequency = <60>; /* required if extcomin-gpios is defined */
		disp-en-gpios = <&gpioc 13 GPIO_ACTIVE_HIGH>;
	};

};
```

And here's the error log from console:

```
[00:00:00.241,000] <inf> ls0xx: Configuring DISP pin to OUTPUT_HIGH
[00:00:00.241,000] <inf> ls0xx: Configuring EXTCOMIN pin
[00:00:00.241,000] <inf> usb_dc_stm32: PWR not active yet
*** Booting Zephyr OS build v3.7.0-5324-gae361a9853c0 ***
[00:00:00.351,000] <inf> sample: Display sample for ls0xx@0
[00:00:00.503,000] <err> ls0xx: Width not a multiple of 400
[00:00:00.503,000] <err> ls0xx: Width not a multiple of 400
[00:00:00.503,000] <err> ls0xx: Width not a multiple of 400
[00:00:00.504,000] <err> ls0xx: Width not a multiple of 400
[00:00:00.604,000] <err> ls0xx: Width not a multiple of 400
[00:00:00.704,000] <err> ls0xx: Width not a multiple of 400

```

From what I can tell from basic printf, the demo sends a fraction of the width, in this case its 100px. While that works fine with displays like the `st7735r_160x80`, its failing for `ls0xx_ls027b7dh01`.

mmahadevan108, 2024-10-29:
@erwango , can you help take a look as this is being tried on an ST board.

jhedberg, 2024-10-29:
Looks like the error is coming from the driver's `ls0xx_write()` function:
https://github.com/zephyrproject-rtos/zephyr/blob/022c8ee1afbc6373265ef9c0a198defa3e862247/drivers/display/ls0xx.c#L182-L185

The sample in question simply feeds the driver back what it reports through get_capabilities():
https://github.com/zephyrproject-rtos/zephyr/blob/022c8ee1afbc6373265ef9c0a198defa3e862247/samples/drivers/display/src/main.c#L203
and:
https://github.com/zephyrproject-rtos/zephyr/blob/022c8ee1afbc6373265ef9c0a198defa3e862247/samples/drivers/display/src/main.c#L297-L310

The sample isn't checking for the return value of `display_get_capabilities()`, so it's possible that it's failing, e.g. because there's something wrong with your board configuration or snippet.

fabiobaltieri, 2024-12-10:
@erwango any chance you could take a look? no dedicated maintainer for display :-(

erwango, 2024-12-11:
> @erwango any chance you could take a look? no dedicated maintainer for display :-(

Wrong pick. I'll be useless on display (if ever I am actually useful elsewhere). @JarmouniA maybe ?

asmellby, 2024-12-18:
I came over the same thing on a Silicon Labs dev kit, which uses a LS013 display. The sample app attempts to draw 3 rectangles in the corners of the screen, but this fails with the `ls0xx` driver since the SPI protocol used by this display requires full lines to be written, and the driver does not have an internal framebuffer. Something like this makes the display sample functional (though pretty ugly):
```
diff --git a/samples/drivers/display/src/main.c b/samples/drivers/display/src/main.c
index cdb4c2b4c8d..f926f3ddf9a 100644
--- a/samples/drivers/display/src/main.c
+++ b/samples/drivers/display/src/main.c
@@ -231,6 +231,12 @@ int main(void)
                grey_scale_sleep = 100;
        }
 
+       if (capabilities.screen_info & SCREEN_INFO_X_ALIGNMENT_WIDTH) {
+               rect_w = capabilities.x_resolution;
+       }
+
        buf_size = rect_w * rect_h;
 
        if (buf_size < (capabilities.x_resolution * h_step)) {
```

This is similar to what LVGL does in `lvgl_rounder_cb_mono`. It does lead to a degraded sample app experience (the concept of "top left" and "top right" rectangles becomes a single "top bar"), but a working sample is arguably better than no sample. With this patch, the sample looks like this on a 128x128 display:

https://github.com/user-attachments/assets/4328024e-3ffb-49f7-b24c-6610461ee8d1

PR #83167 (2025-01-31): samples: drivers: display: Support line alignment constrained displays
The sample app draws rectangles in the corner of the display. On devices with the `SCREEN_INFO_X_ALIGNMENT_WIDTH` capability, each draw needs to transmit a complete line. Make the sample work on such displays (e.g. the `ls0xx` driver) by making the rectangles take up the full width of the display.

Fixes #80463
closed 2025-01-31 by kartben