zephyrproject-rtos/zephyr #80476: mgmt: mcumgr: stat_mgmt: stat_mgmt_list always send legacy behaviour

nandojve, 2024-10-27:
**Describe the bug**

The MCUmgr statistics only work correctly when selecting MCUMGR_SMP_LEGACY_RC_BEHAVIOUR. This flag is correctly handled for **stat_mgmt_show** command but it is missing for **stat_mgmt_list** command.

It is easy to compare both functions on this points:

https://github.com/zephyrproject-rtos/zephyr/blob/4b4bba4fa4c0cf5e84433f60a0d0c358a910924c/subsys/mgmt/mcumgr/grp/stat_mgmt/src/stat_mgmt.c#L153-L163

https://github.com/zephyrproject-rtos/zephyr/blob/4b4bba4fa4c0cf5e84433f60a0d0c358a910924c/subsys/mgmt/mcumgr/grp/stat_mgmt/src/stat_mgmt.c#L203-L206

**Expected behavior**

The MCUmgr statistics should work correctly for all protocol versions.

**Impact**

Field devices can not report their groups correctly when using a tool that not allow to use the MCUMGR_SMP_LEGACY_RC_BEHAVIOUR.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK: 0.16.4
 - mainline

**Additional context**

Tested with https://github.com/intercreate/smpmgr

thedjnK, 2024-10-27:
> The MCUmgr statistics only work correctly when selecting MCUMGR_SMP_LEGACY_RC_BEHAVIOUR

Then your tool is broken, this Kconfig exists specifically for the go tool, which had a deprecated warning added to the documented a long time ago, and whose documentation was fully removed earlier this year. If you use another tool and it requires this then the fault is with the tool

nandojve, 2024-10-27:
Can you explain why one command send the RC and the another not?

thedjnK, 2024-10-27:
> @thedjnK ,
> 
> Can you explain why one command send the RC and the another not?

Probably an oversight when it was added, but the extra rc field being there should not breaks tools when rc is 0

nandojve, 2024-10-27:
Ok,

>  response with rc = 0 and response without rc are equivalent, 

This means that if I drop rc field it still comply with a V2 protocol. Is this assumption correct? I mean, a tool that only works with V2 will be able to handle the response. In this case, can I assume that rc field on V2 protocol means always an error ?

thedjnK, 2024-10-27:
> Ok,
>
>
> This means that if I drop rc field it still comply with a V2 protocol. Is this assumption correct? I mean, a tool that only works with V2 will be able to handle the response. In this case, can [...]

No rc has nothing to do with v2, it was added before such a protocol existed, it is because you don't need an error variable if there is no error, it is superfluous.

nandojve, 2024-10-27:
@thedjnK ,

Thank you for point out those info it help me to clarify some concepts since I still green in this area.
In the case of the PR #80477, I think it is fair to move in. At least we ensure behavior for both commands.

JPHutchins, 2024-11-12:
@nandojve Is any action needed at smp/smpclient/smpmgr to fix this?

Looking at the spec I see no rc field in the success response: https://docs.zephyrproject.org/latest/services/device_mgmt/smp_groups/smp_group_2.html

I see no indication that the behavior should be different for V1 or V2.  Presence of rc = 0 should cause validation to fail because that is not a valid success response of the form:

```
{
    (str)"stat_list" :  [
        (str)<stat_group_name>, ...
    ]
}
```

PR #80477 (2024-10-30): mgmt: mcumgr: stat: Fix stat_mgmt_list behavior
The MCUmgr statistics only work correctly when selecting MCUMGR_SMP_LEGACY_RC_BEHAVIOUR option. Checks the option flag on the stat_mgmt_list and fix the behavior.

Fixes: #80476
CC: @otavio