zephyrproject-rtos/zephyr #80489: Zephyr 4.0  security related modules update overview

dkalowsk, 2024-10-28:
Umbrella issue for tracking security related modules which may require an update before Zephyr 4.0.

- [ ] https://github.com/zephyrproject-rtos/mbedtls (@ceolin , @d3zd3z , @tomi-font)
  - Looks to be on upstream latest 3.6.2 LTS https://github.com/Mbed-TLS/mbedtls
- [ ] https://github.com/zephyrproject-rtos/tinycrypt (@ceolin )
  - Is marked as deprecated
- [ ] https://github.com/zephyrproject-rtos/trusted-firmware-m (@d3zd3z , @ithinuel , @tomi-font  )
  - Seems to have been update to the latest upstream 2.1.0 as 2.1.1 is in RC.
  - https://tf-m-user-guide.trustedfirmware.org/releases/index.html
- [ ] https://github.com/zephyrproject-rtos/trusted-firmware-a (@povergoing, @sgrrzhf, @ithinuel )
  - Seems to be behind (2.7) and not on the LTS (2.10).
  -  I thought we updated to 2.10 with https://github.com/zephyrproject-rtos/zephyr/pull/74111 but I don't see those tags in our repo.
- [ ] https://github.com/zephyrproject-rtos/uoscore-uedhoc (@rlubos , @StefanHri )

tomi-font, 2024-10-28:
I'll take care of updating TF-M to 2.1.1 once it is officially released.

rlubos, 2024-10-29:
`uoscore-uedhoc` is up to date with upstream.

dkalowsk, 2024-11-12:
tinycrypt deprecation is moved to 4.1

mmahadevan108, 2024-11-19:
All items addressed.

closed 2024-11-19 by mmahadevan108