zephyrproject-rtos/zephyr #80491: Zephyr 4.0 module update overview

dkalowsk, 2024-10-28:
Umbrella issue for modules which may require an update before Zephyr 4.0.

- [ ] https://github.com/zephyrproject-rtos/acpica (@najumon1980)
- [ ] https://github.com/zephyrproject-rtos/babblesim-manifest (@aescolar)
- [ ] https://github.com/zephyrproject-rtos/canopennode (@henrikbrixandersen)
- [ ] https://github.com/zephyrproject-rtos/chre (@yperess)
- [ ] https://github.com/zephyrproject-rtos/cmsis (@stephanosio, @microbuilder, @povergoing)
- [ ] https://github.com/zephyrproject-rtos/cmsis-dsp (@XenuIsWatching, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/cmsis-nn (@XenuIsWatching, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/edtt (@aescolar, @wopu-ot, @thoh-ot)
- [ ] https://github.com/zephyrproject-rtos/fatfs (@de-nordic)
- [ ] https://github.com/zephyrproject-rtos/hostap (@krish2718, @jukkar)
- [ ] https://github.com/zephyrproject-rtos/liblc3 (@Casper-Bonde-Bose, @MariuszSkamra)
- [ ] https://github.com/zephyrproject-rtos/libmetal (@carlocaione, @arnopo)
- [ ] https://github.com/zephyrproject-rtos/loramac-node (@Mani-Sadhasivam)
- [ ] https://github.com/zephyrproject-rtos/lvgl (@faxe1008, @brgl, @pdgendt)
- [ ] https://github.com/zephyrproject-rtos/lz4 (@parthitce )
- [ ] https://github.com/zephyrproject-rtos/mcuboot (@d3zd3z, @de-nordic, @nordicjm)
- [ ] https://github.com/zephyrproject-rtos/nanopb (@pdgendt)
- [ ] https://github.com/zephyrproject-rtos/open-amp (@carlocaione)
- [ ] https://github.com/zephyrproject-rtos/openthread (@rlubos)
- [ ] https://github.com/zephyrproject-rtos/percepio (@eriktamlin, @aronlander-pe)
- [ ] https://github.com/zephyrproject-rtos/picolibc (@keith-packard, @stephanosio)
- [ ] https://github.com/zephyrproject-rtos/sof (@kv2019i)
- [ ] https://github.com/zephyrproject-rtos/tflite-micro (@laurenmurphyx64)
- [ ] https://github.com/zephyrproject-rtos/thrift (@cfriedt)
- [ ] https://github.com/zephyrproject-rtos/zcbor (@de-nordic)
- [ ] https://github.com/zephyrproject-rtos/zephyr-lang-rust (@d3zd3z)
- [ ] https://github.com/zephyrproject-rtos/zscilib (@microbuilder)

Folks that are tagged please comment and confirm the status of your modules with respect to readiness for the 4.0 release.

- [Zephyr 4.0 Release Checklist #76338](https://github.com/zephyrproject-rtos/zephyr/issues/76338)
- [Zephyr 4.0 security related modules update overview #80489 ](https://github.com/zephyrproject-rtos/zephyr/issues/80489)

aescolar, 2024-10-28:
@dkalowsk EDTT is already the latest version. For bsim, I just qeued a PR with a minor update with some minor improvements. It is not required for 4.0 but does not harm either.

jukkar, 2024-10-28:
@dkalowsk hostap update is planned after 4.0 is released (tracked in Wi-Fi project page https://github.com/orgs/zephyrproject-rtos/projects/50)

microbuilder, 2024-10-28:
There is a PR for zscilib, but waiting on the PR author to comment, and can be updated post 4.0 release.

keith-packard, 2024-10-28:
In picolibc, there are some fixes for clang, arc and xtensa upstream which should probably get pulled in for 4.0. We'll need to sync with the SDK and see when that might get updated.

stephanosio, 2024-10-29:
> In picolibc, there are some fixes for clang, arc and xtensa upstream which should probably get pulled in for 4.0. We'll need to sync with the SDK and see when that might get updated.

@keith-packard Unless those are critical bug fixes, we will likely have to target 4.1 since releasing and integrating another version of SDK will likely take more than a week and, at that point, we will be at rc2 or even rc3 where we cannot risk toolchain-related regressions.

stephanosio, 2024-10-29:
CMSIS (v5) is already at the latest version (5.9.0). As for CMSIS v6, it will likely have to be added as a separate module: https://github.com/zephyrproject-rtos/zephyr/issues/77220.

keith-packard, 2024-10-29:
> @keith-packard Unless those are critical bug fixes, we will likely have to target 4.1 since releasing and integrating another version of SDK will likely take more than a week and, at that point, we will be at rc2 or even rc3 where we cannot risk toolchain-related regressions.

Sounds fine -- none of them are critical.

nordicjm, 2024-10-29:
MCUboot will be updated with a bug fix which is currently pending on the submitted fixing their commit, a sync will then be opened in zephyr

pdgendt, 2024-10-29:
Created #80555 to sync `nanopb` with upstream.

rlubos, 2024-10-29:
No need to update OpenThread.

de-nordic, 2024-10-29:
Regarding following modules:
 - https://github.com/zephyrproject-rtos/fatfs - will not be updated prior to 4.0; there will be update after, bringing fixes in features that are not yet enabled in Zephyr
 - https://github.com/zephyrproject-rtos/littlefs - missing from list, at this point this is at version latest offered by upstream (https://github.com/littlefs-project/littlefs/commits/master/)
 - https://github.com/zephyrproject-rtos/zcbor  - @oyvindronningstad, can you comment here, the project is under your control.

eriktamlin, 2024-10-29:
No need to update percepio module, as far as we can tell.

Best regards
Erik Tamlin

keith-packard, 2024-10-29:
Sounds like we'll leave picolibc alone for 4.0 as the SDK is already set.

nordicjm, 2024-10-30:
MCUboot: https://github.com/zephyrproject-rtos/zephyr/pull/80607

cfriedt, 2024-11-05:
No updates for Thrift yet. I'm following up about why this was reverted and how to get it back in. 
https://github.com/apache/thrift/pull/2871
https://github.com/apache/thrift/pull/2899

nordicjm, 2024-11-06:
MCUboot has been updated to bring in a fix for a user reported issue

MariuszSkamra, 2024-11-06:
No plans to update liblc3

laurenmurphyx64, 2024-11-06:
@JordanYates Since you last updated the module, do you think the Tensorflow Lite Micro module needs an update?

JordanYates, 2024-11-08:
LoRaMAC-node is in maintenance mode and has no updated

JordanYates, 2024-11-08:
> @JordanYates Since you last updated the module, do you think the Tensorflow Lite Micro module needs an update?

Only if required by #80495, which was the reason I had to update it last time.

closed 2024-11-19 by mmahadevan108