zephyrproject-rtos/zephyr #80514: drivers: video: migrate from source=/sensor= to remote-endpoint-label=

josuah, 2024-10-28:
**Describe the bug**

Following https://github.com/zephyrproject-rtos/zephyr/pull/74415, video drivers are in the middle of a migration away from an ad-hoc devicetree API.

The reason for it was https://github.com/zephyrproject-rtos/zephyr/issues/57708 and the workaround is https://github.com/zephyrproject-rtos/zephyr/pull/74415. A final conversion from `remote-endpoint-label` to `remote-endpoint` will need to happen once #57708 is addressed, but will be trivial to propagate.

**To Reproduce**

```
&source_dev {
        status = "okay";
};

&sino_dev {
        source = <&source_dev>;
        /* or */
        sensor = <&source_dev>;
        /* or nothing */
};
```

- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/espressif%2Cesp32-cam.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/nxp%2Cvideo-smartdma.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/ovti%2Cov2640.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/ovti%2Cov2640.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/ovti%2Cov7670.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/st%2Cstm32-dcmi.yaml

**Expected behavior**

```
&source_dev {
        status = "okay";
        port {
                source_dev_ep_out: endpoint {
                        remote-endpoint-label = "sink_dev_ep_in";
                };
        };
};

&sink_dev {
        port {
                sink_dev_ep_in: endpoint {
                        remote-endpoint-label = "source_dev_ep_out";
                };
        };
};
```

- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/aptina%2Cmt9m114.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/galaxycore%2Cgc2145.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/nxp%2Cimx-csi.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/nxp%2Cmipi-csi2rx.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/ovti%2Cov5640.yaml
- https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/video/ovti%2Cov7725.yaml

**Impact**

- Need to also convert the board and shields
- Only one API through the sources, more similar to Linux.
- A stable API that covers complex situations.

**Additional context**

Zephyr v4.0 is in feature-freeze: not possible to rush every vendor to complete the migration of their driver in time.

One possibility is to not convert anything for v4.0, but propagate bugfixes that might be needed:

- https://github.com/zephyrproject-rtos/zephyr/pull/79334

josuah, 2024-10-28:
It might not be required to push any bugfix for `v4.0.0`:

Currently, the sensor drivers do not do anything with the `remote-endpoint-label`, so passing them `remote-endpoint-label = "unused";` will work with MIPI/DVP controllers with the old API: not ideal but not broken.

Also, as pointed-out [here](https://github.com/zephyrproject-rtos/zephyr/pull/79334/files#r1817894995), adding the `remote-endpoint-label` is not expected to break the compilation.

ngphibang, 2024-10-28:
Thanks @josuah for creating the issue. I made some modifications in the description. This is not considered as "bug" so I removed the bug label.

`remote-endpoint` is declared in some dts but not actually used in any Zephyr drivers. And while we don't declare a binding for the endpoint, declaring or not the `remote-endpoint` or anything else doesn't matter, it still compiles.

However, migration to use the video-interfaces binding is strongly recommended because of the following benefits:

- We can actually use the `remote-endpoint-label` to retrieve the peer remote device object in the driver without the need of direct phandle reference.
- We can use the common endpoint properties in the video-interfaces binding to avoid declaring so many vendor custom properties in each driver.
- We can avoid some sort of chicken-egg issue as described [here](https://github.com/zephyrproject-rtos/zephyr/issues/80304).

ngphibang, 2025-04-26:
Reopen because there are still esp32 dvp and stm32 dcmi drivers that need to be done.

josuah, 2025-05-08:
Arducam Nicla Vision needs conversion from `remote-endpoint` left as a palceholder to `remote-endpoint-label` enforced from #89571.

ngphibang, 2025-05-14:
Reopen because there are still some drivers that need to be done.