zephyrproject-rtos/zephyr #80524: Flash API:flash_mmap API call for obtaining flash device mapping in CPU address space

de-nordic, 2024-10-28:
## Introduction

The proposed change brings optional `flash_driver_api` callback `mmap` with supporting `flash_mmap` API call.
Function is provided to obtain, from supporting driver, information on where flash device is mapped, if that is the case in CPU memory map.

### Problem description

There are two issues being solved here:
 1)  Nearly all SoC devices have internal flash mapped to address space which allows quick access and direct casting of storage device to structures used in code, allowing significant improvements in performance and reduced stack usage where reads, and potentially writes, can be done via standard C functions like memcpy, rather than calling entire chain of calls to flash driver.
 2) Devices have now various coprocessors that have direct access to address space and can operate on data directly very efficiently, without need to transfer the data to RAM first - this also saves on RAM usage in some scenarios; such coprocessors can be used for direct SHA calculation, AES decryption/encryption. direct signature validation on storage, etc. These coprocessors, to work, need to obtain direct address on device they are going to operate on so that they can program DMA or other peripheral to directly get data from storage.

### Proposed change

Proposed change here is implementation of Flash API call, followed by Flash Map API call, that will allow to obtain address and size of mapping a device has within CPU address space.

The proposed API call looks like this:
```
int flash_mmap(const struct device *dev, void **base, uint64_t *size, uint32_t flags)
```
where `base` is returned base address of device, `size` is variable for size of mapping and `flags` is one or more flags that describe requested access.

## Detailed RFC

Drivers that support such change will have to implement `mmap` callback returns mapping of device to address space.
If driver does not implement the function, -ENOTSUP if will be returned by the `flash_mmap` call.

### Dependencies

None, change is independent from other parts of code and function may or may not be implemented and may and may not be used.

### Concerns and Unresolved Questions

There is certain risk in allowing direct access to memory as output of mmap may not be filtered, at this point, by driver to limit mapping to only area that is supposed to be visible by thread. This gives two potential risks: 1) areas protected by readout will fault device if access is attempted and 2) unprotected areas that should not be read may be mapped and read directly.

Both issues actually are resident within the code anyway, because any pointer can trigger them.

Nevertheless the coprocessors can not work effectively without direct access to storage via address space, which means that without such convenient access as asking device driver directly, users will have to resolve to hardcoding addresses and offsets, potentially falling into above issues anyway.

So, all you can do is test your stuff.

## Alternatives
Hardcoding offsets/addresses and making maintaining the user code more painful than needed.
Directly accessing information from DTS, which in turn makes problems for projects like MCUboot, where you have to potentially serve each device/vendor differently instead of just asking device for values to use.

de-nordic, 2024-12-18:
@FRASTM @dleach02 you may be interested in this feature as it is also intended to allow MCUboot directly pass memory regions to coprocessors capable of running SHA and encryption functions via DMA, bypassing CPU.

swift-tk, 2025-02-04:
Could this applicable to non internal flash devices with XIP regions? These regions are also within the CPU address space. I’m curious why getting it from the DTS not a better option?

carlescufi, 2025-02-04:
Architecture WG

- @bjarki-andreasen likes the idea but is worried about concurrent access. 
- @danieldegrasse asks whether we get out of this API, because this will create a two-tier system of devices, one that uses mmap and is faster, and another one that uses standard `flash_read()`. @de-nordic responds that it's not only about reading, but also being able to point hardware accelerators directly to the place in flash where a particular data item is stored, instead of having to load it into RAM first

de-nordic, 2025-02-04:
> 
> Would we need to be able to handle noncontinuous flash regions from the same driver?

This is good question to how we want to treat such regions as Zephyr overall.
If vendor decides that they want to see flash as continuous space where read/write/erase operations re-align when crossing between banks or over holes, then they would probably not be able to use the flash_mmap, or be limited to only support certain bank.
If vendor is able to map flash at specific location, via DTS for example, and make it continuous then there will be no such problem,

Still there is question whether other devices that can DMA from address space, can use such device.

I guess that this will be a choice here, if continuous space can not be obtained: supporting only part of space or just not supporting flash_mmap at all.
Note also that such devices are problematic for users because user may decide to directly get info how the device is mmaped to provide such info to accelerator, for example by parsing the DTS, but this then creates problem where that mapping is not in pair with what flash API sees and how to map these things together.

de-nordic, 2025-02-04:
> Architecture WG
> 
>     * [@bjarki-andreasen](https://github.com/bjarki-andreasen) likes the idea but is worried about concurrent access.

Regarding this problem: even though I was thinking about providing some API to allow sync when mmaped area is "entered" and "left" by some thread, at what range and for what reason, I wonder what wold be the case?
In the end the access will be synchronized by some subsystem that takes certain area exclusively. For example FS will take certain partition and access to that partition is synchronized by that FS, and that FS can also decide to switch between mmaped and API access.
Another example would be DFU: here we write image of application and then verify it; in this case neither upload nor verification overlap and have to be synchronized by DFU.
What about sharing some data over part of flash: that needs to be done via some subsystem that will do the synchronization anyway.
I may be wrong here and miss some use cases, but in all of above what I see is that Flash API access is very low API that does not support directly overlapping of interest from other APIs.
The only problem I see now is that some drivers can not do writes concurrently across any part of device.

> * [@danieldegrasse](https://github.com/danieldegrasse) asks whether we get out of this API, because this will create a two-tier system of devices, one that uses mmap and is faster, and another one [...]

That is important issue, that we also have met. For example we have nrf54l15 devices that support ed25519, where they can in ms verify signature over internal memory, but when user decides to use SPI connected flash, this no longer works and we have to switch to ed25519PH. Still the internal memory is quicker because we can at least pass the internal partition to sh512 accelerator, so we get some advantage here anyway.

@danieldegrasse You have also mentioned important thing: catching. How do we deal with that? Do we need extra call?
For example mmaping of a file system to RAM and directly accessing via pointers would significantly improve read access times, but when the same file system performs write, do we need to invoke some call to re-fresh that read cache? Or do we drop that on driver to do with write implementation.

Another important question is: do we assume that mapping is only done via DTS? If there is a case for run-time mapping, can we assume that semantic of the flash_mmap is as here - just to expose the already done mapping to RAM - and the actual mapping is operation reserved only for drivers?

de-nordic, 2025-02-06:
> Could this applicable to non internal flash devices with XIP regions? These regions are also within the CPU address space. I’m curious why getting it from the DTS not a better option?

In theory yes, but that depends on your chip. Read access would certainly work, sync with write may not, but as I have stated you - as a vendor and owner of chip - will know the details and what you can offer via this call to an user.

The reason it is not done via DTS is that this is intended to find out how flash device maps to memory. This would propagate to partitions too. So for example if you have some partition over internal, or external, flash/rram/mram and want to know where it is in your address space, you can do that via this call.

butok, 2025-02-10:
So, you are going to circumvent the flash device .read API abstraсtion that can have additional checks.
For most platforms it should be OK. But a problem can be for a flash using ECC, e.g. LPC55Sxx soc flash.

de-nordic, 2025-02-11:
> So, you are going to circumvent the flash device .read API abstraсtion that can have additional checks. 

Yes. But, things that pick up data directly, like crypto co-processors/accelerators, will ignore the checks too.

> For most platforms it should be OK. But a problem can be for a flash using ECC, e.g. LPC55Sxx soc flash.

How things that can pick data directly via DMA behave when ECC is not ok? For example on that LPC55Sxx?

That is why we have the issue here to point out things like that, and figure how we deal with them.

So the question is then: should the `flash_mmap` call then do the check whether area is readable and report error, forcing software into `flash_read` path or other mitigation?

Or should we expose the "can that be read" API call?

The first would have to check entire mmaped device and return error if anything in range is not ok; problems:
1) entire range needs to be checked
2) even if something that will not be accessed causes problem we are returning error from flash_mmap

If we expose the "can that be read" API call, then we can use it to verify if the limited range we want to pass can be used.

In favor  of exposing the "can that be read" API call: this can also be used by Flash Map, to check the range of partition only.

de-nordic, 2025-11-06:
> Find the RFC when working with STM32 internal flash. Also need such flash mmap feature both for code simplicity and RAM efficiency.

If there is any input you can provide regarding the API extension and any risks you see, from your experience, it would be nice, because so far the discussion is rather stale here.