zephyrproject-rtos/zephyr #80531: Twister broken with external tree

jerome-pouiller, 2024-10-28:
The [`zephyr-silabs`][1] repository [use zephyr as an external tree][2]. `zephyr-silabs` recently [update from Zephyr commit a2ac676a1 to 9f93dede3][3]. Since then, [`zephyr-silabs` CI is broken][4] (I also wonder why twister fail but the job success, but it is out of the scop f this bug report).

I am able to reproduce the issue by running the same commands than the [`silabs-zephyr` CI][5].

Using git bisect, I have found the behavior of twister has changed with the commit f1bf24e4d ("boards: other: stm32_min_dev: use one twister.yaml"). I am also able to fix the issue by reverting these commits:

    "boards: other: stm32_min_dev: use one twister.yaml"
    "twister: rework board handling"
    "boards: intel_adsp: use one twister.yaml"
    "twister: test udpates for new board handling"
    "twister: --device-testing always implies --filter runnable"
    "twister: fix platform filter when loading plan from file"
    "twister: also convert platform names from the hardwaremap"

jhedberg, 2024-10-28:
Pasting here the relevant part of the CI failure logs:
```
Run west twister -s sample.bluetooth.peripheral_hr -p siwx917_rb4338a -v --inline-logs -K
Renaming output directory to /home/runner/work/zephyr-silabs/zephyr-silabs/zephyr-silabs/twister-out.1
INFO    - Using Ninja..
INFO    - Zephyr version: 9f93dede369c
INFO    - Using 'zephyr' toolchain.
Traceback (most recent call last):
  File "/opt/hostedtoolcache/Python/3.11.10/x64/bin/west", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/opt/hostedtoolcache/Python/3.11.10/x64/lib/python3.11/site-packages/west/app/main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "/opt/hostedtoolcache/Python/3.11.10/x64/lib/python3.11/site-packages/west/app/main.py", line 244, in run
    self.run_command(argv, early_args)
  File "/opt/hostedtoolcache/Python/3.11.10/x64/lib/python3.11/site-packages/west/app/main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "/opt/hostedtoolcache/Python/3.11.10/x64/lib/python3.11/site-packages/west/app/main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/opt/hostedtoolcache/Python/3.11.10/x64/lib/python3.11/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/home/runner/work/zephyr-silabs/zephyr-silabs/zephyr/scripts/west_commands/twister_cmd.py", line 61, in do_run
    ret = main(options, default_options)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/work/zephyr-silabs/zephyr-silabs/zephyr/scripts/pylib/twister/twisterlib/twister_main.py", line 120, in main
    tplan.discover()
  File "/home/runner/work/zephyr-silabs/zephyr-silabs/zephyr/scripts/pylib/twister/twisterlib/testplan.py", line 185, in discover
    self.add_configurations()
  File "/home/runner/work/zephyr-silabs/zephyr-silabs/zephyr/scripts/pylib/twister/twisterlib/testplan.py", line 424, in add_configurations
    known_boards = list_boards.find_v2_boards(lb_args)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/work/zephyr-silabs/zephyr-silabs/zephyr/scripts/list_boards.py", line 329, in find_v2_boards
    systems = list_hardware.find_v2_systems(root_args)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/runner/work/zephyr-silabs/zephyr-silabs/zephyr/scripts/list_hardware.py", line 238, in find_v2_systems
    for root in unique_paths(args.soc_roots):
  File "/home/runner/work/zephyr-silabs/zephyr-silabs/zephyr/scripts/list_hardware.py", line 204, in unique_paths
    yield from dict.fromkeys(map(Path.resolve, paths)).keys()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/hostedtoolcache/Python/3.11.10/x64/lib/python3.11/pathlib.py", line 997, in resolve
    p = self._from_parts((s,))
        ^^^^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute '_from_parts'
Error: Process completed with exit code 1.
```

jhedberg, 2024-10-28:
@jerome-pouiller looking at the above log, is it possible that this might be a python version issue?

jhedberg, 2024-10-28:
@jerome-pouiller I'm not able to reproduce this with https://github.com/zephyrproject-rtos/example-application

The following command works through `example-application` but fails with the same python error on `zephyr-silabs`:
```
❯ west twister -s sample.basic.helloworld -p qemu_x86
INFO    - Using Ninja..
INFO    - Zephyr version: v3.7.0-5545-g9f93dede369c
INFO    - Using 'zephyr' toolchain.
Traceback (most recent call last):
  File "/Users/johedber/.venv/bin/west", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/johedber/.venv/lib/python3.12/site-packages/west/app/main.py", line 1085, in main
    app.run(argv or sys.argv[1:])
  File "/Users/johedber/.venv/lib/python3.12/site-packages/west/app/main.py", line 244, in run
    self.run_command(argv, early_args)
  File "/Users/johedber/.venv/lib/python3.12/site-packages/west/app/main.py", line 505, in run_command
    self.run_extension(args.command, argv)
  File "/Users/johedber/.venv/lib/python3.12/site-packages/west/app/main.py", line 654, in run_extension
    self.cmd.run(args, unknown, self.topdir, manifest=self.manifest,
  File "/Users/johedber/.venv/lib/python3.12/site-packages/west/commands.py", line 194, in run
    self.do_run(args, unknown)
  File "/Users/johedber/src/silabs/zephyr/scripts/west_commands/twister_cmd.py", line 61, in do_run
    ret = main(options, default_options)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/johedber/src/silabs/zephyr/scripts/pylib/twister/twisterlib/twister_main.py", line 120, in main
    tplan.discover()
  File "/Users/johedber/src/silabs/zephyr/scripts/pylib/twister/twisterlib/testplan.py", line 185, in discover
    self.add_configurations()
  File "/Users/johedber/src/silabs/zephyr/scripts/pylib/twister/twisterlib/testplan.py", line 424, in add_configurations
    known_boards = list_boards.find_v2_boards(lb_args)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/johedber/src/silabs/zephyr/scripts/list_boards.py", line 329, in find_v2_boards
    systems = list_hardware.find_v2_systems(root_args)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/johedber/src/silabs/zephyr/scripts/list_hardware.py", line 238, in find_v2_systems
    for root in unique_paths(args.soc_roots):
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/johedber/src/silabs/zephyr/scripts/list_hardware.py", line 204, in unique_paths
    yield from dict.fromkeys(map(Path.resolve, paths)).keys()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/johedber/.local/share/uv/python/cpython-3.12.5-macos-aarch64-none/lib/python3.12/pathlib.py", line 1240, in resolve
    s = self._flavour.realpath(self, strict=strict)
        ^^^^^^^^^^^^^
AttributeError: 'str' object has no attribute '_flavour'
```

So to me this smells like something may be wrong with the zephyr-silabs tree.

jhedberg, 2024-10-28:
If I comment out `board_root` and `soc_root` in `zephyr/module.yml` I'm able to run `west twister -s sample.basic.helloworld -p qemu_x86` in the `zephyr-silabs` tree.

jhedberg, 2024-10-28:
If I add `soc_root = .` to `example-application/zephyr/module.yml` I get the same error too

jerome-pouiller, 2024-10-28:
Yeah, I suspect these lines are not correct anymore:
```
def find_v2_boards(args):
    root_args = argparse.Namespace(**{'soc_roots': args.soc_roots})
    systems = list_hardware.find_v2_systems(root_args)
```

nashif, 2024-10-28:
try this fix 80537

PR #80537 (2024-10-30): twister: oot soc: set soc_root using Path
Set soc_root using Path to avoid wrong generated path in the list of
soc_roots and other roots read from module.yml file

Fixes #80531
commit 2c43d7fe (2024-10-30): twister: oot soc: set soc_root using Path

Set soc_root using Path to avoid wrong generated path in the list of
soc_roots and other roots read from module.yml file

Fixes #80531
closed 2024-10-30 by mmahadevan108