zephyrproject-rtos/zephyr #80558: Samples: Bluetooth: peripheral_hids fails to set security

Thalley, 2024-10-29:
**Describe the bug**
The sample has
```c
	if (bt_conn_set_security(conn, BT_SECURITY_L2)) {
		printk("Failed to set security\n");
	}
```

in the `connected` callback, but it seems to fail to do encryption when connected to an Android Phone.

**To Reproduce**
1. Build the sample with CONFIG_SAMPLE_BT_USE_AUTHENTICATION=n
2. Flash the sample
3. Connect an Android Phone
4. See error

**Expected behavior**
Expect the security to be set to level 2. 

**Impact**
Depends. If it was something that used to work and stopped working at some point, it is caused by a change in the stack that may not be documented. 

**Logs and console output**
```
Advertising successfully started
Connected 51:9E:D8:12:F4:4F (random)
Security failed: 51:9E:D8:12:F4:4F (random) level 1 err (2)
Disconnected from 51:9E:D8:12:F4:4F (random), reason 0x13 
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: afce5329f772a4306af699c7a06efddf93cadca0

**Additional context**
Tested on a nRF5340 DK for the Zephyr sample with the `hci_ipc` sample with the  `nrf5340_cpunet_iso-bt_ll_sw_split.conf` overlay for the network core. 
Tested on an Pixel 8 running Android 14

alwa-nordic, 2024-10-29:
Would you be able to get an HCI pcap? Would be great if you could test with an older version of Zephyr too.

Thalley, 2024-10-29:
> Would you be able to get an HCI pcap? Would be great if you could test with an older version of Zephyr too.

Trying to, but I have some issues getting the log with `btmon`. Not sure what the cause is yet.

jhedberg, 2024-10-29:
Shouldn't the `bt_conn_set_security()` call be excluded when `CONFIG_SAMPLE_BT_USE_AUTHENTICATION=n`? I.e. is the sample simply missing an #ifdef?

Thalley, 2024-10-29:
> Shouldn't the `bt_conn_set_security()` call be excluded when `CONFIG_SAMPLE_BT_USE_AUTHENTICATION=n`? I.e. is the sample simply missing an #ifdef?

Security level 2 can be done without authentication

jhedberg, 2024-10-30:
> Security level 2 can be done without authentication

Right, and I checked that `smp.c` should correctly fall back to just-works when there are no authentication callbacks registered.

Thalley, 2025-01-06:
Reran the sample on the nRF5340 DK again today to check if this is still an issue. There is still an issue, but now I'm getting 
```
Advertising successfully started
Connected 46:CD:72:5C:D6:6C (random)
Security failed: 46:CD:72:5C:D6:6C (random) level 1 err (9)
Disconnected from 46:CD:72:5C:D6:6C (random), reason 0x13 
```

The sample is also not restarting advertising after disconnects.

Thalley, 2025-03-10:
As of today, I can no longer reproduce this issue. Closing

closed 2025-03-10 by Thalley