zephyrproject-rtos/zephyr #80574: Meta-IRQ does not preempt cooperative threads when woken from IRQ  (cortex_m)

Ronvanl, 2024-10-29:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Impact**

**Environment (please complete the following information):**

 - Zephyr 3.7.0

cvinayak, 2024-10-31:
@wopu-ot Could someone in your team comment here, I assume Meta-IRQ are used by some of your implementations?

wopu-ot, 2024-11-01:
Yes, we use Meta-IRQs in our implementation of the controller, but this issue seems to be specific to ARM and therefore does not apply. The test in tests/kernel/sched/metairq tests some of the Meta-IRQ behavior; maybe it could be extended to test for the issue described above.

ithinuel, 2024-11-01:
Hi, [this](https://github.com/zephyrproject-rtos/zephyr/pull/65071/commits/131ca8a5bee25706926faccb3d126b8cd059d2d5) was introduced as a fix to [#61761](https://github.com/zephyrproject-rtos/zephyr/issues/61761).

From [here](https://docs.zephyrproject.org/latest/kernel/services/threads/index.html#meta-irq-priorities):

> […] there is a special subclass of cooperative priorities at the highest (numerically lowest) end of the priority space: meta-IRQ threads. These are scheduled according to their normal priority, but also have the special ability to preempt all other threads (and other meta-IRQ threads) at lower priorities, even if those threads are cooperative and/or have taken a scheduler lock.

and

> Note that because this breaks the promise made to cooperative threads by the Zephyr API (namely that the OS won’t schedule other thread until the current thread deliberately blocks)

@Ronvanl does replacing `(_kernel.cpus->current->base.prio >= 0)` with the following snippet helps ?
```c
((_kernel.cpus->current->base.prio >= 0) || (_kernel.cpus->current->base.prio < (-CONFIG_NUM_COOP_PRIORITIES)))
```

Ronvanl, 2024-11-01:
@ithinuel, I replacing the check with the snippet and it doesn't work as it assumes the current thread is a meta-irq thread? Replacing it with the following snippet does work as it checks the cached thread but I am wondering if this check is supposed to be here anyway. I believe the administration is done in `update_cache()`

```
((_kernel.cpus->current->base.prio >= 0) || thread_is_metairq(_kernel.ready_q.cache))
```

Looking at https://github.com/zephyrproject-rtos/zephyr/issues/61761, it doesn't take in account meta-irq threads which according to the documentation (below) can preempt all threads even if they have taken a scheduler lock.

_These are scheduled according to their normal priority, but also have the special ability to preempt all other threads (and other meta-IRQ threads) at lower priorities, even if those threads are cooperative and/or have taken a scheduler lock._

Ronvanl, 2024-11-04:
Can we confirm this is a regression and if so will it be fixed and backported to 3.7.0 and if so when?

In the meantime, it there a way to overwrite the z_arm_exc_exit fucntion from application code? We are using workspace application where zephyr is cloned from official repo and we do not have our own fork (yet).

ithinuel, 2024-11-05:
> Can we confirm this is a regression 

I think so yes.

> and if so will it be fixed and backported to 3.7.0

I think it should be backported.

> and if so when?

once a fixed has been proposed & merged.

> In the meantime, is there a way to overwrite the z_arm_exc_exit functionfrom application code? We are using workspace application where zephyr is cloned from official repo and we do not have our own fork (yet).

I don't think so.
Does the snippet you shared in your previous message fix the issue?
If so, I think the fastest would be for you to open a PR with the fix you could validate, ideally with extra tests that would catch future regressions of that feature.

Ronvanl, 2024-11-05:
The snippet indeed fixes the issue but I think reverting the changes in [42036cd](https://github.com/zephyrproject-rtos/zephyr/commit/42036cdbcae270d00b5bd08e805e33bbd3b5787f) should be the way forward however there is a reason why its added right?

There is a sample which demonstrates the constant latency of the meta-irq threads https://docs.zephyrproject.org/latest/samples/kernel/metairq_dispatch/README.html. Maybe this can be included in the tests with a well-defined criteria

ithinuel, 2024-11-06:
The reason for this change is that it is a fix to https://github.com/zephyrproject-rtos/zephyr/issues/61761.

I don't know if zephyr is able to validate timing accuracy (this highly depends on the simulator/device).

ithinuel, 2024-11-13:
@Ronvanl Would you mind creating a PR with that change you could validate or do you prefer me to create it?

Ronvanl, 2024-11-14:
Would appreciate if you could do it, especially if any (unit) testing is required as I'm not familiar with the test framework.

andyross, 2024-11-14:
Driving by: scheduler logic really doesn't belong in arch code, and this is a good example why.  It's long since time to migrate cortex-m to arch_switch(), which puts this logic under a z_get_next_switch_handle() API.  I got started in this space a week and a half ago (in the context of "remove PendSV" and not switch/swap per se) and might have something to push soonish.

Ronvanl, 2024-12-05:
@andyross I see 3.7.1 is released, any chance a fix is included here already?

andyross, 2024-12-05:
Not yet, I'd expect that work at best to be slow-rolled into the default configs.  For this bug we want to address it in the existing code; the comment above was a design nitpick, basically.

kietavainen, 2025-01-27:
There is now a test case for this issue in #84509.

PR #84509 (2025-02-04): arch: arm: cortex_m: Fix meta-IRQ unable to preempt from interrupt
This reverts commit 42036cdbcae270d00b5bd08e805e33bbd3b5787f.

Architecture specific code should not do preemption checking before context switch. This is already handled by the scheduler, so duplicating it would be redundant and error prone. These checks used to be necessary, but the scheduler has been rewritten since then and the checks were removed in 3a0cb2d35d9 (kernel: Remove legacy preemption checking, 2018-05-23).

The added checks tried to address #61761, and were motivated by this in the Architecture Porting Guide:
ht [... cut]
commit d929b8a9 (2025-02-04): Revert "arch: arm: cortex_m: Only trigger context switch if thread is preemptible"

This reverts commit 42036cdbcae270d00b5bd08e805e33bbd3b5787f.

Architecture specific code should not do preemption checking before context
switch. This is already handled by the scheduler, so duplicating it would
be redundant and error prone. These checks used to be necessary, but the
scheduler has been rewritten since then and the checks were removed in
3a0cb2d35d9 (kernel: Remove legacy preemption checking, 2018-05-23).

The check this reverts was also incorrect, as it didn't take scheduler
locking nor meta-I [... cut]
PR #85139 (2025-07-03): [Backport v3.7-branch] arch: arm: cortex_m: Fix meta-IRQ unable to preempt from interrupt
Backport 52b8e3d3706f7503d6e4e36413c2f78a2ac95480~2..52b8e3d3706f7503d6e4e36413c2f78a2ac95480 from #84509.

Fixes #80574
Fixes https://github.com/zephyrproject-rtos/zephyr/issues/92413
commit e011c605 (2025-07-03): Revert "arch: arm: cortex_m: Only trigger context switch if thread is preemptible"

This reverts commit 42036cdbcae270d00b5bd08e805e33bbd3b5787f.

Architecture specific code should not do preemption checking before context
switch. This is already handled by the scheduler, so duplicating it would
be redundant and error prone. These checks used to be necessary, but the
scheduler has been rewritten since then and the checks were removed in
3a0cb2d35d9 (kernel: Remove legacy preemption checking, 2018-05-23).

The check this reverts was also incorrect, as it didn't take scheduler
locking nor meta-I [... cut]
closed 2025-02-04 by kartben