zephyrproject-rtos/zephyr #80578: Display driver GC9X01X - Color inversion enabled by default on GC9A01A.

sudhishu, 2024-10-29:
**Describe the bug**

While using display driver display_gc9x01x with Waveshare GC9A01A round display module,  the color inversion is set on by default. Absence or explicit deletion of "display_inversion" node to set it false from device-tree overlay did not give desired result. 

Setup:
 - HDK = nRF7002-DK + Waveshare GC9A01A round display module.
 - SDK = nRF Connect SDK  version 2.7.0 and LVGL library for screen.
 - Zephyr OS = Version 3.6.99 ,  build v3.6.99-ncs2
 - IDE  = Visual Studio Code with Nordic nRF connect plugin installed.
 
Workaround or attempted fix:

Modified the driver implementation in Zephyr/drivers/display/display_gc9x01x.c
Explicitly set the GC9X01X_CMD_INVON (0x21U) register to 0 during device and register init calls. Proper colors were displayed after applying the fix.
<code>
        ret = gc9x01x_transmit(dev, GC9X01X_CMD_INVON, 0, 1);
	if (ret < 0) {
		return ret;
	}
</code>

**To Reproduce**

Possibly this is the case with any development board which is using Galaxycore GC9x01x driver. This looks like a registry initialization requirement for the display chip. In this case, the 240x240 px round LCD display using GC9A01A chip is provided by Waveshare.

1. prj.conf
      - Enable CONFIG_DISPLAY and CONFIG_GC9X01X at prj.conf
2. Device tree overlay: 
<code>
chosen {
        zephyr,display = &gc9a01a_lcd;
};
...
...
...
&spi2  {
    status = "okay";
    compatible = "nordic,nrf-spim";
    pinctrl-0  = <&spi2_default>;
    pinctrl-1  = <&spi2_sleep>;
    pinctrl-names = "default", "sleep";
    cs-gpios = <&gpio1 12 GPIO_ACTIVE_LOW>;

    gc9a01a_lcd: gc9a01a_lcd@0 {
        compatible = "galaxycore,gc9x01x";
        status = "okay";
        reg     = <0>;
        width  = <240>;
        height = <240>;
        pixel-format      = <(0x1 << 4)>;
        spi-max-frequency = <100000000>;
        cmd-data-gpios     = <&gpio1 11 GPIO_ACTIVE_HIGH>;
        reset-gpios          = <&gpio0 25 GPIO_ACTIVE_LOW>;
        /delete-property/ display-inversion;
      };
};
</code>
Note that the <code>/delete-property/ display-inversion </code> was an attempt to set the node explicitly false in the device configuration, but that did not work. Hence not necessary.

**Expected behavior**

The display should show the desired colors on the screen as set using the LVGL library. 
**Impact**

The entire color scheme on the display was inverted. Failed to create the desired screen.
**Logs and console output**

None.
**Environment (please complete the following information):**
 - OS  = Fedora Linux - 6.5.10-200.fc38.x86_64
 - HDK = nRF7002-DK + Waveshare GC9A01A round display module.
 - SDK = nRF Connect SDK  version 2.7.0 and LVGL library for screen.
 - Zephyr OS = Version 3.6.99 ,  build v3.6.99-ncs2
 - IDE  = Visual Studio Code with Nordic nRF connect plugin installed.
 
**Additional context**

kartben, 2024-10-29:
ping @MrMarteng @joelguittet for awareness

sudhishu, 2024-10-30:
> @sudhishu can you help submit a PR with your fix
1) Did you mean "pull-request" by PR ?  I am new to the process of Zephyr development.  Need time to go through the contribution guidelines doc. I welcome any additional guidance, how-to pointers etc. 

2) The fix I attempted was a straight code-insertion according to my understanding of the problem. I see a "default init sequence" looping over a set of commands inside the <code> gc9x01x_regs_init() </code> function. May be, including the GC9X01X_CMD_INVON register value into that command-set  (struct default_init_regs array) could be the correct approach to comply with the design of the current implementation. I would like to verify and confirm that first.

sudhishu, 2024-10-30:
The correct approach to fix the code according to the implementation is : 
Add the following command entry into the <code> struct default_init_regs </code> array. 
 - Ref: zephyr/drivers/display/display_gc9x01x.c  - line 63
 <code>
{
        .cmd = 0x21U,
        .len = 1U,
        .data = {0x00U}
 }
</code>
This value will be picked by <code> gc9x01x_regs_init() </code> and the color-inversion register will be initialized accordingly.  This fix is tested successfully on Waveshare GC9A01A 240x240 px round lcd implementation. 
Note that the datasheet says that this registry is set to 0 by default, hence the source of problem could be the chip's internal default settings and by this fix, the driver is compensating it.

fabiobaltieri, 2024-12-10:
@danieldegrasse could you take a look by any chance?

danieldegrasse, 2024-12-20:
I don't have a module with this display controller to test on, but I'm unclear why sending the `INVON` command like this would work:
```
ret = gc9x01x_transmit(dev, GC9X01X_CMD_INVON, 0, 1);
if (ret < 0) {
    return ret;
}
```

The signature of the `gc9x01x_transmit` function looks like so:
```
static int gc9x01x_transmit(const struct device *dev, uint8_t cmd, const void *tx_data, size_t tx_len)
```

So the call you make would try to write 1 byte from 0x0. This may work without causing a fault on some platforms, but it isn't a stable fix- effectively what you are doing is writing a dummy byte of data after the INVON command. This should not be required, as the INVON command should not need a parameter: 
[image]

Could you possibly look at the data being sent on a signal analyzer? Perhaps there is some odd behavior on your system when transmitting single byte commands to the display

vishwamartur, 2025-02-25:
I apologize for the inconvenience. I'm unable to resolve this issue @danieldegrasse

joelguittet, 2025-02-26:
Hello,
Does it need somebody with a hardware to test ?
I have the ESP32-S3-Touch-LCD-1.28. Last time I wanted to test the HEAD of zephyr the support was broken, not sure why, but I may be able to have a look.
Joel

joelguittet, 2025-04-28:
Thanks for the quick answer.
Waveshare provides the register set: https://files.waveshare.com/upload/5/5e/GC9A01A.pdf
I disagree with sending some bytes with the 0x20 and 0x21 commands. It is explicitly indicated there is no parameters. Sending more bytes probably overflows somewhere we don't know and even if it seems to work for you, this is not an acceptable solution at the moment.

What about this kind of modification in the driver:

```
	/* Display inversion mode. */
	if (config->inversion) {
		ret = gc9x01x_transmit(dev, GC9X01X_CMD_INVON, NULL, 0);
		if (ret < 0) {
			return ret;
		}
	} else {
		ret = gc9x01x_transmit(dev, GC9X01X_CMD_INVOFF, NULL, 0);
		if (ret < 0) {
			return ret;
		}
        }
```

Can you also confirm your screen has the exact reference GC9A01A? Not a derivative that is similar but different?

Else maybe need clarification from Waveshare about the screen:
- Confirmation of the exact chipset.
- Why a screen shipped with default setting with inversion ON while it is not required?
- If sending the ON command with 1 byte is allowed (maybe this is something not indicated by the datasheet but which is permitted) - if yes an update of the datasheet would be appreciated.

Joel

joelguittet, 2025-04-29:
> As we have already seen, if I do
ret = gc9x01x_transmit(dev, GC9X01X_CMD_INVON, NULL, 0);
regardless of config->inversion state, the display works with proper
colors.

So, if you add the inversion property to the device tree, this should be working. Isn't it ? (as this is exactly what the driver does when this property is added).

joelguittet, 2025-05-05:
@sudhishu ping :-)

If you are not sure can you share your device tree source?

sudhishu, 2025-05-06:
Hi,
Following is the major portion of the DTS we are using where the relevant
parts are shown.
Setting "display-inversion" in DTS will enable inversion, correct ? To my
understanding, the presence of that property means a "true" and absence is
"false". What I had to do is to explicitly "unset" or in other words, set
display-inversion to false, as if it has been set to true by default. I
could not find out whether I am doing any configuration mistakes or not.
Board is "b_u585i_iot02a"
----------------------------------------------------------------------------------------------------------
/* SPDX-License-Identifier: Apache-2.0 */

#include <zephyr/dt-bindings/mipi_dbi/mipi_dbi.h>
#include <zephyr/dt-bindings/display/panel.h>
#include <zephyr/dt-bindings/lvgl/lvgl.h>

/delete-node/ &storage_partition;

/{
    chosen {
        zephyr,display = &gc9a01a;
        zephyr,rtc = &rv8803;
    };

    fstab {
        compatible = "zephyr,fstab";
        lfs1: lfs1 {
            compatible = "zephyr,fstab,littlefs";
            mount-point = "/lfs1";
            partition = <&lfs1_part>;
            automount;
            read-size = <16>;
            prog-size = <16>;
            cache-size = <64>;
            lookahead-size = <32>;
            block-cycles = <512>;
      };
  };

  mipi_dbi_spi{
      status      = "okay";
      compatible  = "zephyr,mipi-dbi-spi";
      dc-gpios    = <&gpioc 1  (GPIO_ACTIVE_HIGH)>;
      reset-gpios = <&gpiof 13 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
      spi-dev     = <&spi1>;
      write-only;
      #address-cells = <1>;
      #size-cells    = <0>;

      gc9a01a: ***@***.*** {
          status     = "okay";
          compatible = "galaxycore,gc9x01x";
          reg        = <0>;
          width      = <240>;
          height     = <240>;
          orientation = "normal";
          pixel-format = <PANEL_PIXEL_FORMAT_RGB_565>;   // <(0x1 << 4)>;
          mipi-max-frequency = <DT_FREQ_M(32)>;
          mipi-mode = "MIPI_DBI_MODE_SPI_4WIRE";
      };
  };

  buttons {
      compatible = "gpio-keys";
      button0: button_0 {
          gpios = <&gpiod 15 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
          label = "Push button 1";
          };
      button1: button_1 {
          gpios = <&gpioe 7 (GPIO_PULL_UP | GPIO_ACTIVE_LOW)>;
          label = "Push button 2";
      };
  };

   aliases{
      sw0 = &button0;
      sw1 = &button1;
   };
};

dut: &usart3 {
    dmas = <&gpdma1 0 29 STM32_DMA_PERIPH_TX
    &gpdma1 1 28 STM32_DMA_PERIPH_RX>;
    dma-names = "tx", "rx";
};

/* spi1: ***@***.*** */
&spi1  {
    status = "okay";
    cs-gpios = <&gpioe 12 (GPIO_ACTIVE_LOW| GPIO_PULL_DOWN)>;
};

&timers4 {
  status = "okay";
  st,prescaler = <0>;
  pwm4: pwm {
    status = "okay";
    pinctrl-0 = <&tim4_ch1_pb6>;
    pinctrl-names = "default";
   };
};
----------------------------------------------------------------------------------------------------------

- Sudhish.

joelguittet, 2025-05-06:
Whatever it is the default, whatever it is set or unset, this is just toggling the color inversion setting.

You indicated:

> ret = gc9x01x_transmit(dev, GC9X01X_CMD_INVON, NULL, 0);
> 
> - regardless of config->inversion state, the display works with proper
> colors.

I think this is just misunderstanding. The default of the controller is color inversion NOT set, but the screen itself NEEDS color inversion to be SET. As for the Waveshare ESP32-S3 Touch LCD board: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/waveshare/esp32s3_touch_lcd_1_28/esp32s3_touch_lcd_1_28_esp32s3_procpu.dts#L72-L81

Just do:

```
      gc9a01a: ***@***.*** {
          status     = "okay";
          compatible = "galaxycore,gc9x01x";
          reg        = <0>;
          width      = <240>;
          height     = <240>;
          orientation = "normal";
          pixel-format = <PANEL_PIXEL_FORMAT_RGB_565>;   // <(0x1 << 4)>;
          display-inversion;
          mipi-max-frequency = <DT_FREQ_M(32)>;
          mipi-mode = "MIPI_DBI_MODE_SPI_4WIRE";
      };
```

This will call `gc9x01x_transmit(dev, GC9X01X_CMD_INVON, NULL, 0);` command in the driver as you do manually, and it should be fine. Don't think to the color logic and default in the controller, just activate inversion.

Or this is myself misunderstanding something?

Joel

sudhishu, 2025-05-07:
> @sudhishu <https://github.com/sudhishu> good ! So I guess this issue can
> be closed now ?

Yes, I think it is okay to close this issue with a note, saying there are
gc9x01x chipsets out there which may display with color inversion by
default and the workaround in such a situation is to specify
"display-inversion" in the DTS entry.

Thankyou Joel, for your time and effort to follow-up this issue, and
bringing clarity to it.

-Sudhish

joelguittet, 2025-05-07:
I don't have the permission to close.
@danieldegrasse I guess you can.

closed 2025-05-07 by kartben