zephyrproject-rtos/zephyr #80598: zephyr system hang when reset

GaofengZhangNXP, 2024-10-30:
zephyr system hang and can't  input any command

nordicjm, 2024-10-30:
> zephyr system hang and can't input any command

Based on the very limited information provided (i.e. none) I disagree, it works fine here?

danieldegrasse, 2024-10-30:
@GaofengZhangNXP please expand this issue to list (at a minimum) the following:
- What platform and SHA of Zephyr you can reproduce the issue on
- Specific steps (IE a build and flash command) to reproduce the issue
- Any specific logs that would help understand the issue

The guidelines within the bug template are helpful for this, usually I try to fill all of them out. For an example of an issue with this information, take a look here: https://github.com/zephyrproject-rtos/zephyr/issues/80590

dkalowsk, 2024-11-05:
No details.  Marking as closed.

closed 2024-11-07 by GaofengZhangNXP