zephyrproject-rtos/zephyr #80618: Different behavior with and without --sysbuild

gwa-simon, 2024-10-30:
I am porting my application from zephyr 3.5 to 3.7 and I have updated the boards structure according to the new standards. My project folder structure looks something like the following:

```
<home>
├── boards/plank
|	├── board.yml
|	├── plank_0_5_0_defconfig        
|	├── plank_0_5_0.overlay          
|	├── plank.dts    
|	├── plank.yml  
|	├── Kconfig.defconfig
|	├── Kconfig.plank
|	└── plank_defconfig 
|
├── app
	├── CMakeLists.txt
	├── prj.conf
	├── sysbuild.conf
	└── sysbuild
    		└── mcuboot.conf

```

I have revisions for my board and since it allows to add configuration linked to boards on the boards folder, I have done so by moving some config that I had previously in prj.conf to plank_defconfig and plank_0_5_0_defconfig. 

If I run ``west build -p always -b plank@0.5.0 \app`` it works perfectly, even if I select other revisions.

But my problem starts when I try running it with **sysbuild**. If I run ``west build -p always -b plank@0.5.0 \app  --sysbuild`` the compilation works for mcuboot, but fails for my application because the Kconfigs get messed up. It complains that s dependencies for configs in the board folder are missing (this dependencies are defined in the prj.conf).

Therefore my question is, what is the right procedure to use sysbuild while having board revisions and sub-configs defined at boards folder?

nordicjm, 2024-10-30:
The board folder like that only works for applications, not sysbuild (since sysbuild itself is actually an application), you would need to add your board root in a module file e.g. see example application here https://github.com/zephyrproject-rtos/example-application/blob/main/zephyr/module.yml#L16

gwa-simon, 2024-10-31:
Hello @nordicjm, thanks for the fast answer.

Although I already have my board as a module imported by west and in the app module.yml I have the config as following, since my boards modules is 2 levels above the app:

```

build:
  settings:
    board_root: ..
    dts_root: ..
```

The dts and overlay it can find, it just can not link both **prj.conf** and **board specific configs** (plank_0_5_0_defconfig) to the final configs at the same time.

nordicjm, 2024-10-31:
the only place a module.yml file can go is `zephyr/module.yml` from the root directory of the git repo, and it needs to be either the manifest repo itself or a module referenced in a west manifest

gwa-simon, 2024-10-31:
Yes, I have the module.yml inside a zephyr folder for the main project and on my modules.

[image]

Hmm, now I noticed I don't have it on the boards module because I point to it in the CMakeLists. Should I have a zephyr/module.yml also in my boards folder?

nordicjm, 2024-10-31:
it should be in your boards repo yes, since the folder has to be named boards for the build system to find it and having a boards folder inside of a boards folder isn't ideal, you should have this in your board folder `zephyr/module.yml` file:
```
``

gwa-simon, 2024-10-31:
Thanks for the tip, I did it and I was able to remove an include in my CMakeLists. But I still have my initial problem. 
An example, I have CONFIG_SENSOR=y in my prj.conf file and in my boards module I have my sensors enabled on the plank_defconfig (e.g. CONFIG_SHT3XD=y).

With normal application it builds fine, but with sysbuild it does not see CONFIG_SENSOR=y when including CONFIG_SHT3XD=y

```
warning: SHT3XD (defined at drivers/sensor/sensirion/sht3xd/Kconfig:6) was assigned the value 'y'
but got the value 'n'. Check these unsatisfied dependencies: SENSOR (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SHT3XD and/or look up SHT3XD in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.
```

Although, if I look in the autoconf.h, it is there:

[image]

Aaaa, now I notice something, the warning is while building the mcuboot, where the prj.conf is not included, but the plank_defconfig (I guess), generating the dependency error, even dough I don't need this on the mcuboot.

gwa-simon, 2024-10-31:
So it turns out that the dependency warning is only for the mcuboot build, and does not really affect it. My real problem is that while sysbuild builds the app my own modules are not finding the CONFIGs definitions

[image]

But CONFIG_SENSOR_LOG_LEVEL is present on the app/autoconf.h, so it looks the module is not including this definitions

nordicjm, 2024-10-31:
Having a sensor enabled in a board defconfig is frankly wrong, if a project needs it, the project should enable it in the prj.conf (or `boards/<board>.conf`)