zephyrproject-rtos/zephyr #80631: Multiprotocol Support for BLE and Thread on Silicon Labs Devices

swapnilcchougule, 2024-10-30:
Hello,

I'm exploring the feasibility of implementing a multiprotocol feature to run both **Bluetooth Low Energy (BLE) and Thread protocols concurrently** on a Silicon Labs platform using Zephyr. Specifically, I’m interested in understanding the current level of support for this setup on Silicon Labs SoCs (EFR32) with Zephyr, especially in **multiprotocol** scenarios.

My typical use case involves running a RAIL-based proprietary protocol alongside the Silicon Labs Bluetooth stack on the same chip simultaneously.

Here is a specific question I have:
Are there any sample applications or example configurations within Zephyr that demonstrate BLE and other wireless protocols running concurrently on Silicon Labs devices?

Thanks in advance for any insights or resources!

Best regards,
Swapnil

asmellby, 2024-10-31:
This is currently not supported. As of today, Zephyr only integrates the single-protocol variant of the RAIL library. To enable multiprotocol, `librail_multiprotocol_*.a` would need to take the place of `librail_*.a` to enable the use of multiple RAIL handles.

swapnilcchougule, 2024-10-31:
Thank you for the information!

I have added librail_multiprotocol_efr32xg24_gcc_release.a and initialized the BLE stack with bt_enable();. Additionally, sl_btctrl_init_multiprotocol(); initializes successfully, indicating that the multiprotocol library is functioning.

However, I’m encountering an issue where RAIL_Init() returns a NULL pointer, which suggests it’s unable to create a second RAIL instance. For reference, I am using this example from Silicon Labs: [Simple TX/RX Sample](https://github.com/SiliconLabsSoftware/zephyr-silabs/tree/main/samples/simple_txrx).

If you have any insights on what might be causing issue on enabling multiple RAIL instances effectively, I’d greatly appreciate your help!

Thanks again for your assistance.