zephyrproject-rtos/zephyr #80658: Zephyr as PIC

dev-72-Z, 2024-10-31:
**Is your feature request related to a problem? Please describe.**
I need to build Zephyr as position independent code. 

**Describe the solution you'd like**
To have PIC option in linker scripts, or to be able to set it with not to much intervention in the code.

**Describe alternatives you've considered**
Tried to have .rodata section not dependent on absolute addresses set in .dts files.
Didn't manage to move the initlevel section.  

**Additional context**

henrikbrixandersen, 2024-11-01:
> I need to build Zephyr as position independent code.

Please describe the use case a bit more.

dev-72-Z, 2024-11-01:
We have memory addresses "hardcoded" in `.dts` files.
Based on `.dts` file ~/include/generated/devicetree_generated.h is generated early in build process.
The result binary, has `.rodata` section whose content depends on the absolute addresses from .`dts` file.
Assuming we have several platforms for the same SOC, but with different `.dts` files. After any change in code
we have to build binary for each one of them.
If we had position independent code, i.e. no absolute addresses in .rodata, one binary could be used 
for all platforms.

ycsin, 2025-04-22:
Another usecase would be the A/B seamless system update (with a compatible bootloader)?

achech, 2025-04-24:
The question is where to start from, kobject hashing could be also affected.