zephyrproject-rtos/zephyr #80679: Bluetooth: HCI: More info in prints

koffes, 2024-10-31:
Booting a sample with the Softdevice Controller, output may look like this:

```
<inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
<inf> bt_hci_core: HW Variant: nRF53x (0x0003)
<inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version XX.XXXXX Build XXXXXXXXX
<inf> bt_hci_core: HCI: version 6.0 (0x0e) revision 0x5041, manufacturer 0x0059
<inf> bt_hci_core: LMP: version 6.0 (0x0e) subver 0x5041
```

And the Zephyr controller looks like this:
```
<inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
<inf> bt_hci_core: HW Variant: nRF53x (0x0003)
<inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.7 Build 99
<inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x0059
<inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
```

Unless one knows the controller version or build number, it is difficult to see which controller is being used.
Also, it is easy to think that since the HCI version has manufacturer 0x0059, that the controller is actually SDC.

Recommend:

1. Add manufacturer ID after "Firmware:"
2. Rename "Firmware" to "Controller"
3. Rename "Standard Bluetooth Controller" to actually reflect the name of the vendor.

@theob-pro , @alwa-nordic

cvinayak, 2024-11-01:
> Rename "Firmware" to "Controller"

The variants recognized by the Zephyr Vendor Specific HCI commands are:
https://github.com/zephyrproject-rtos/zephyr/blob/193eeaef0c04570b80a954b58ccd98075150fb11/subsys/bluetooth/host/hci_core.c#L3888-L3893

Action:
* Change `Standard Bluetooth Controller`  to `Zephyr(R) Bluetooth Controller`; vendor specific controllers shall use the `fw_variant` value `1`.
* Change `Vendor specific controller` to `Vendor specific Bluetooth Controller`
  * Enhancement, get a string from Kconfig or Devicetree node?

Rest of the information are correctly represented. Manufacturer ID, revision and subversion, are part of HCI Read Local Version Information Command Response. Firmware version and build are obtained by the HCI Vendor Specific Read Version Information Command Response.

alwa-nordic, 2024-11-01:
> Unless one knows the controller version or build number, it is difficult to see which controller is being used.

This is the real story, right?

> As someone reading some logs given to me, I want it to be obvious which of the ZLL and SDC is used.

Maybe we could issue [Zephyr Read Build Information Command](https://github.com/zephyrproject-rtos/zephyr/blob/193eeaef0c04570b80a954b58ccd98075150fb11/doc/connectivity/bluetooth/api/hci.txt#L501) when info-logging is enabled and print the response. The response is an arbitrary UTF-8 string, so the Controller could introduce itself however it wants.

closed 2026-08-21 by kartben