zephyrproject-rtos/zephyr #80686: The receive buffer provided by the spim_xfer_rx function in the nrfwifi module is empty

hongquan-prog, 2024-10-31:
**Describe the bug**
When I try to use mimxrt1061 to drive the nrf7002, the terminal always tells me that the spi communication timed out!

**To Reproduce**
- Add a wlan node to devicetree
```
/ {
	model = "NXP MIMXRT1060-EVK board";
	compatible = "nxp,mimxrt1062";

	aliases {
		led0 = &green_led;
		watchdog0 = &wdog0;
	};

	chosen {
		zephyr,sram = &ocram2;
		zephyr,itcm = &itcm;
		zephyr,dtcm = &dtcm;
		zephyr,console = &lpuart1;
		zephyr,shell-uart = &lpuart1;
		zephyr,canbus = &flexcan3;
		zephyr,wifi = &wlan0;
	};
};

&lpspi3 {
	status = "okay";
	/* DMA channels 2 and 3, muxed to LPSPI3 RX and TX */
	dmas = <&edma0 2 15>, <&edma0 3 16>;
	dma-names = "rx", "tx";
	pinctrl-0 = <&pinmux_lpspi3>;
	pinctrl-names = "default";
    cs-gpios = <&gpio1 28 GPIO_ACTIVE_LOW>;
	transfer-delay = <60>;

    nrf70: nrf7002@0 {
		status = "okay";
		compatible = "nordic,nrf7002-spi";
		reg = <0>;
		spi-max-frequency = <10000000>;

		/* D0 */
		iovdd-ctrl-gpios = <&gpio1 1 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>;
		/* D1 */
		bucken-gpios = <&gpio1 20 (GPIO_ACTIVE_HIGH | GPIO_PULL_DOWN)>;
		/* D7 */
		host-irq-gpios = <&gpio1 27 GPIO_ACTIVE_HIGH>;
		/* Short-range (SR) co-existence */

		/* Maximum TX power limits for 2.4 GHz */
		wifi-max-tx-pwr-2g-dsss = <21>;
		wifi-max-tx-pwr-2g-mcs0 = <16>;
		wifi-max-tx-pwr-2g-mcs7 = <16>;

		/* List of interfaces */
		wlan0: wlan0 {
			compatible = "nordic,wlan";
		};

		wifi-max-tx-pwr-5g-low-mcs0 = <9>;
		wifi-max-tx-pwr-5g-low-mcs7 = <9>;
		wifi-max-tx-pwr-5g-mid-mcs0 = <11>;
		wifi-max-tx-pwr-5g-mid-mcs7 = <11>;
		wifi-max-tx-pwr-5g-high-mcs0 = <13>;
		wifi-max-tx-pwr-5g-high-mcs7 = <13>;
	};
};
```
- Compiling a wifi sample application with west
```
west build -b mimxrt1060_evk zephyr/samples/net/wifi
```
- Burn the programme and wait for the system to boot up
- Add a log in the following judgement, you will see the error message
```c
bool LPSPI_CheckTransferArgument(LPSPI_Type *base, lpspi_transfer_t *transfer, bool isEdma)
{
    assert(transfer != NULL);
    uint32_t bitsPerFrame  = ((base->TCR & LPSPI_TCR_FRAMESZ_MASK) >> LPSPI_TCR_FRAMESZ_SHIFT) + 1U;
    uint32_t bytesPerFrame = (bitsPerFrame + 7U) / 8U;
    uint32_t temp          = (base->CFGR1 & LPSPI_CFGR1_PINCFG_MASK);
    /* If the transfer count is zero, then return immediately.*/
    if (transfer->dataSize == 0U)
    {
        return false;
    }

    /* If both send buffer and receive buffer is null */
    if ((NULL == (transfer->txData)) && (NULL == (transfer->rxData)))
    {
        printk("txData and rxData are null\n");
        return false;
    }
}
```
**Expected behavior**
NIC initialised successfully

**Impact**

**Logs and console output**
[image]

**Environment (please complete the following information):**

 - OS: MacOS
 - Toolchain zephyr-sdk-0.16.4
 - Commit SHA 124aae3876842154964726ba6e89565040648f9a

**Additional context**
This problem does not exist in Nordic's spi driver, only in some spi driver，
```c
static int spim_xfer_rx(unsigned int addr, void *data, unsigned int len, unsigned int discard_bytes)
{
	uint8_t hdr[] = {
		0x0b, /* FASTREAD opcode */
		(addr >> 16) & 0xFF,
		(addr >> 8) & 0xFF,
		addr & 0xFF,
		0 /* dummy byte */
	};

	const struct spi_buf tx_buf[] = {
		{.buf = hdr,  .len = sizeof(hdr) },
		{.buf = NULL, .len = len },
	};

	const struct spi_buf_set tx = { .buffers = tx_buf, .count = 2 };

	const struct spi_buf rx_buf[] = {
		{.buf = NULL,  .len = sizeof(hdr) + discard_bytes},
		{.buf = data, .len = len },
	};

	const struct spi_buf_set rx = { .buffers = rx_buf, .count = 2 };

	return spi_transceive_dt(&spi_spec, &tx, &rx);
}
```
This problem does not exist in Nordic's spi driver, it only exists in some spi drivers(eg: spi_mcux_lpspi.c ), when nrf7002 initialises and calls the spim_xfer_rx function, it sends 5 bytes and ignores the n bytes it receives after that, n is the length that needs to be ignored, and 5 is the length of the requested hader, which requires a total of n+5 bytes of buffer, but spim_ xfer_rx only provides 5 bytes of send buffer, after 5 bytes are sent, there is no extra buffer for storing the n bytes received, the nxp hal library does not allow the receive buffer and the send buffer to be null pointers at the same time, so when the error occurs it chooses to return directly and will not continue to receive data.
```c
bool LPSPI_CheckTransferArgument(LPSPI_Type *base, lpspi_transfer_t *transfer, bool isEdma)
{
    assert(transfer != NULL);
    uint32_t bitsPerFrame  = ((base->TCR & LPSPI_TCR_FRAMESZ_MASK) >> LPSPI_TCR_FRAMESZ_SHIFT) + 1U;
    uint32_t bytesPerFrame = (bitsPerFrame + 7U) / 8U;
    uint32_t temp          = (base->CFGR1 & LPSPI_CFGR1_PINCFG_MASK);
    /* If the transfer count is zero, then return immediately.*/
    if (transfer->dataSize == 0U)
    {
        return false;
    }

    /* If both send buffer and receive buffer is null */
    if ((NULL == (transfer->txData)) && (NULL == (transfer->rxData)))
    {
        return false;
    }
}
```
There are two solutions to this problem, the first is to provide a receive buffer for in spim_xfer_rx, the second is that nxp's lpspi driver supports both receive and send buffers to be empty at the same time, I chose the former, so that I can avoid the same problem again with the difference in SPI drivers.

Here is my solution：
```c
static int spim_xfer_rx(unsigned int addr, void *data, unsigned int len, unsigned int discard_bytes)
{
	uint8_t hdr[] = {
		0x0b, /* FASTREAD opcode */
		(addr >> 16) & 0xFF,
		(addr >> 8) & 0xFF,
		addr & 0xFF,
		0 /* dummy byte */
	};
	uint8_t discard[sizeof(hdr) + 2 * 4];

	const struct spi_buf tx_buf[] = {
		{.buf = hdr,  .len = sizeof(hdr) },
		{.buf = NULL, .len = len },
	};

	const struct spi_buf_set tx = { .buffers = tx_buf, .count = 2 };

	const struct spi_buf rx_buf[] = {
		{.buf = discard,  .len = sizeof(hdr) + discard_bytes},
		{.buf = data, .len = len },
	};

	const struct spi_buf_set rx = { .buffers = rx_buf, .count = 2 };

	if (rx_buf[0].len > sizeof(discard)) {
		LOG_ERR("Discard bytes too large");
		return -EINVAL;
	}

	return spi_transceive_dt(&spi_spec, &tx, &rx);
}
```

rruuaanng, 2024-11-02:
If you like, please raise a PR.

krish2718, 2024-11-02:
Thanks @hongquan-prog for the fix, I don't see any issues with that, please go ahead and raise a PR.

hongquan-prog, 2024-11-02:
> Thanks @hongquan-prog for the fix, I don't see any issues with that, please go ahead and raise a PR.

OK

hongquan-prog, 2024-11-02:
> please go ahead and raise a PR

I'm very sorry for the somewhat late reply, I'm out hiking today and I'll be submitting a PR at a later date.

hongquan-prog, 2024-11-02:
> please go ahead and raise a PR

I'm very sorry for the late reply, I'm out hiking today and I'll be submitting a PR at a later date.

PR #80787 (2024-11-08): drivers/wifi/nrfwifi: Add buffer for discard bytes
Some spi drivers do not allow the send buffer
and receive buffer to be empty at the same time,
if this happens it will cause the spi to be unable to
communicate with the nrf7002, so add the receive 
buffer for the discard byte in the spim_xfer_rx.

Fix #80686
commit c5fa9af2 (2024-11-08): drivers/wifi/nrfwifi: Add buffer for discard bytes

Some spi drivers do not allow the send buffer
and receive buffer to be empty at the same time,
if this happens it will cause the spi to be unable
to communicate with the nrf7002, so add the receive
buffer for the discard byte in the spim_xfer_rx.

Fix #80686
closed 2024-11-08 by mmahadevan108