zephyrproject-rtos/zephyr #80714: Triage: Apply to be a triage in the classifier

rruuaanng, 2024-11-01:
# Nomination

## GitHub User

Provide the following information about the GitHub user:

My name is James Roy, and my GitHub username is as you see. I hope to become a trigger for zephyr, assisting maintainers with tasks such as labeling issues and merging branches. Currently, I am active in the CPython project, where my main development focus is on free-threading and the C-API. If the project plans to release Python API compatibility in the future, I believe I can be of great help.

My recent plan is to continue developing a minimal libc for zephyr. These contributions are not currently listed because I am still working on the patches. While my contributions may not be very prominent at the moment, I am confident that I will do better in the future.

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pull/80511
https://github.com/zephyrproject-rtos/zephyr/pull/80450
https://github.com/zephyrproject-rtos/zephyr/pull/80467
https://github.com/zephyrproject-rtos/zephyr/pull/80104
https://github.com/zephyrproject-rtos/zephyr/pull/80267

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

rruuaanng, 2024-11-01:
@stephanosio

stephanosio, 2024-11-01:
"Triage" is not a role. I assume you mean "Contributor role with Triage permission level," as mentioned in this very template, which you modified? ...

https://github.com/zephyrproject-rtos/zephyr/blob/193eeaef0c04570b80a954b58ccd98075150fb11/.github/ISSUE_TEMPLATE/006_nomination.md?plain=1#L20-L22

> If the project plans to release Python API compatibility in the future

I do not see how this can possibly work. Do you mean something like [MicroPython](https://micropython.org/)? If so, I can confirm that we have no plan to reinvent MicroPython ourselves.

> My recent plan is to continue developing a minimal libc for zephyr

Please do not add anything new to the minimal libc -- it is only intended for testing purposes. The default and recommeded-for-production-use libc for Zephyr is Picolibc, which is a fully featured libc.

rruuaanng, 2024-11-01:
> "Triage" is not a role. I assume you mean "Contributor role with Triage permission level," as mentioned in this very template, which you modified? ...

Oh sorry. That's what I'm trying to say, contributors with categorical permissions :)

> I do not see how this can possibly work. Do you mean something like [MicroPython](https://micropython.org/)? If so, I can confirm that we have no plan to reinvent MicroPython ourselves.

Cpython exposes a number of APIs for embedded interpreters, which means that you can run python in C code, or embed C code into python code.

> Please do not add anything new to the minimal libc -- it is only intended for testing purposes. The default and recommeded-for-production-use libc for Zephyr is Picolibc, which is a fully featured libc.

Oh, I see. I will add new features to Picolibc.

stephanosio, 2024-11-01:
@rruuaanng has been added as a Contributor with Triage permission level.

closed 2024-11-01 by stephanosio