zephyrproject-rtos/zephyr #80738: Improve SPI driver documentation with guidelines handling SPI buffers

scottwcpg, 2024-11-01:
**Is your enhancement proposal related to a problem? Please describe.**
The current SPI driver transfer API has parameters for an array of struct spi_buf for transmits and another for receive.
The SPI driver is based on a model of a full-duplex bus where bits are transmitted and received on the same (may be on different edges).  This implies struct spi_buf must be processed in pairs and are the same length.  A new test of was added to tests/drivers/spi/spi_loopback which tests a TX buffer with length less than the corresponding RX buffer.  The test assumes the driver will transmit 0 bytes while the remaining bytes of the RX buffer are received.  Some SPI HW requires transmit of some data value to generate clocks.  It is not documented what value to transmit.  I did see the "overrun-character" property in spi-controller.yaml bindings.  The test code is not using this value  and only one vendor is using this property.   Test code should use this property.

Next, "dummy" or tri-state I/O's while generating clocks.  The SPI documentation does state a TX buffer with data a NULL data pointer and non-zero length should be interpreted as generating dummy clocks.  This is HW specific.  Some SPI controllers cannot generate clocks without driving the TX line whereas others can.  Any loopback tests must take this into account and ignore data captured during "dummy" clocks.

Next up Extended modes optional feature to support dual, quad, or octal.  How are the TX and RX buffers to be processed?  Some SPI controllers can split their I/O lines.  I believe there is one SPI controller which can split 8 lines into 4 TX and 4 RX.  Other controllers operate half-duplex in dual and quad mode.  The split bus designs could process SPI buffers in a full duplex fashion.  For those controllers implementing dual and quad as half-duplex there are choices:
1. Do all TX buffers first, then do all RX buffers. 
2. Limit driver to only TX or RX buffers being passed.  If both are passed return error.
3. For each pair of (TX, RX) to TX first then RX and move to next pair.  Does this make sense?  If the target SPI device interprets TX data as a command based on CS assertion/de-assertion then this option will not work (SPI flashes).

**Describe the solution you'd like**
More guidelines for implementing a SPI driver based on various SPI HW implementation.
More guidelines for processing SPI buffers.

**Describe alternatives you've considered**
Give up SPI DMA and other performance enhancements (dual/quad) and implement the most simple byte-by-byte transfer to match the test cases.
De-emphasize the SPI driver and develop custom flash drivers and try the mspi driver for my quad SPI controller.

malto101, 2024-11-19:
I would actually love if this gets done,
The current docs for SPI driver dev are very vague

tbursztyka, 2024-11-19:
Will try to find some time for this. I have noticed many times there is a confusion on how SPI buffers are meant to work.