zephyrproject-rtos/zephyr #80824: The net_if carrier is not properly initialised in the nxp_enet driver

anhuba, 2024-11-04:
On power up the carrier flag (NET_IF_LOWER_UP) is set,
although there is no carrier (phy is down, ETH_CARRIER_UP flag not set)

When comparing with the other ethernet drivers, I think the following patch would fix the problem:
[Proposed Patch](/anhuba/zephyr/compare/main...anhuba-patch-1)

Or maybe both net_X_carrier_off functions need to be called?

decsny, 2024-11-06:
net_eth_carrier_X implementation in L2 ethernet code causes call to net_if_carrier_X 

although you're not wrong that other drivers (including the eth_mcux which is the old driver for the same hardware) seem to call net_if_carrier_off after ethernet_init, and I'm not actually sure what's the reason for this

@jukkar do you know about this ?

jukkar, 2024-11-06:
I think the original idea was that it lets the Ethernet driver to do carrier detection properly. By default network interfaces are marked "ON" / "UP", unless `NET_IF_NO_AUTO_START` interface flag is set. So by calling the `net_if_carrier_off` in the driver, we basically mark the interface not usable for communication at startup. Then the driver can call carrier on when cable is detected during initialization. It is probably better not to call `net_eth_carrier_off` at interface init. It will certainly call `net_if_carrier_off` but it also turns off the carrier forcibly and sends event about it which is perhaps not intended at this point.

decsny, 2024-11-06:
> I think the original idea was that it lets the Ethernet driver to do carrier detection properly. By default network interfaces are marked "ON" / "UP", unless `NET_IF_NO_AUTO_START` interface flag is set when the device boots. So by calling the `net_if_carrier_off` in the driver, we basically mark the interface not usable for communication at startup. Then the driver can call carrier on when cable is detected during initialization. It is probably better not to call `net_eth_carrier_off` at interface init. It will certainly call `net_if_carrier_off` but it also turns off the carrier forcibly and sends event about it which is perhaps not intended at this point.

The contracts of the net_eth_carrier_X vs net_if_carrier_X I guess are just not very clear to me. Also, should ethernet_init() be calling net_if_carrier_off , if this is the most common case ?

decsny, 2024-11-06:
> > And it feels like by calling the net_if_X we bypass the ethernet l2 which is supposed to manage the net interface ? Also, should ethernet_init() be calling net_if_carrier_off , if this is the most common case ?
> 
> There are, or at least there was, some drivers that did not support cable detection. There also was no capability to detect that in L2 (perhaps there should be), so Ethernet L2 was not able to know how the interface/carrier on/off should be done. So atm this is let for the driver to do.

yes, but, what I mean is , why would we not do the opposite, where the driver should explicitly call net_if_carrier_on if it doesn't support cable detection

decsny, 2024-11-12:
reopen because the fix broke the specific case of link already being up before iface init

decsny, 2024-11-12:
closing because there is another issue #81274

PR #81020 (2024-11-08): drivers: eth_nxp_enet: use net_if_carrier_off init
Use net_if_carrier_off during iface init instead of net_eth_carrier_off, to immediately mark net if as down

Fixes #80824
closed 2024-11-12 by decsny