zephyrproject-rtos/zephyr #80829: samples: subsys: ipc: ipc_service: icmsg: sample.ipc.icmsg.nrf54l15* fail

katgiadla, 2024-11-04:
**Describe the bug**
The samples:
- `samples/subsys/ipc/ipc_service/icmsg/sample.ipc.icmsg.nrf54l15`
- `samples/subsys/ipc/ipc_service/icmsg/sample.ipc.icmsg.nrf54l15_remote_no_multithreading`
- `samples/subsys/ipc/ipc_service/icmsg/sample.ipc.icmsg.nrf54l15_no_multithreading` 

fails.

Observed for `nrf54l15dk/nrf54l15/cpuapp`

**To Reproduce**
Steps for reproducing:
1. have `nrf54l15dk/nrf54l15/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T samples/subsys/ipc/ipc_service/icmsg -p nrf54l15dk/nrf54l15/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
Logs for `samples/subsys/ipc/ipc_service/icmsg/sample.ipc.icmsg.nrf54l15`:
```
*** Booting Zephyr OS build v3.7.0-4790-g6f64f7755796 ***
[00:00:00.432,298] <inf> host: IPC-service HOST demo started
[00:00:00.000,000] <inf> host: Ep bounded
[00:00:00.000,000] <inf> host: Perform sends for 1000 [ms]
```

Logs for `samples/subsys/ipc/ipc_service/icmsg/sample.ipc.icmsg.nrf54l15_no_multithreading` :
```
*** Booting Zephyr OS build v3.7.0-4790-g6f64f7755796 ***
I: IPC-service HOST demo started
I: Ep bounded
ASSERTION FAIL [packet->data[0] == expected_message] @ zephyr/samples/subsys/ipc/ipc_service/icmsg/src/main.c:48
Unexpected message. Expected A, got 
E: r0/a1:  0x00000004  r1/a2:  0x00000030  r2/a3:  0x00000001
E: r3/a4:  0x00000004 r12/ip:  0x00000000 r14/lr:  0x0000079b
E:  xpsr:  0x29000000
E: Faulting instruction address (r15/pc): 0x00004b60
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Halting system
```

Logs for `samples/subsys/ipc/ipc_service/icmsg/sample.ipc.icmsg.nrf54l15_remote_no_multithreading`:
```
 *** Booting Zephyr OS build v3.7.0-4790-g6f64f7755796 ***
[00:00:00.000,000] <inf> host: IPC-service HOST demo started
```

**Environment (please complete the following information):**
- OS: Ubuntu 20.04.1 LTS
- Toolchain Zephyr SDK 0.16.8
- Commit SHA or Version used: `v3.7.0-3952-gb6aed5c505e6`

nordic-piks, 2024-11-07:
Triaged internally

nordic-piks, 2025-02-19:
Fixed (cannot reproduce)

closed 2025-02-19 by nordic-piks