zephyrproject-rtos/zephyr #80835: nvs: zephyr is blocked in nvs_recover_last_ate while loop

spayet2024, 2024-11-04:
**Describe the bug**

**Impact**

[NVS_corrupted_memory.docx](https://github.com/user-attachments/files/17617967/NVS_corrupted_memory.docx)

henrikbrixandersen, 2024-11-04:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

nordicjm, 2024-11-04:
If you are reporting a bug you need to fill in all the required information https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md?plain=1 I have removed your doc link as it poses a possible security issue

spayet2024, 2024-11-04:
hello, 

Sorry, i getting start on git hub. how can i edit my post? When i commited my post, i loose my description.

Laczen, 2024-11-04:
Hi @spayet2024, could it be that the device is detecting a double ecc error (when reading from the last sector) and is generating a NMI? This kind of error (that can occur on your device) is not handled by any flash driver. If this is the case there is nothing that nvs can do to correct this.

spayet2024, 2024-11-05:
Hi,
Thanks for your reply.
In fact, it could be that. I will investigate this lead. So, is zephyr remains blocked when 'nvs_flash_ate_rd' is called, and more particulary when call 'flash_read' or is it blocked beacause it can't go out of the while loop of 'nvs_recover_last_ate' ?

Laczen, 2024-11-05:
> Hi, Thanks for your reply. In fact, it could be that. I will investigate this lead. So, is zephyr remains blocked when 'nvs_flash_ate_rd' is called, and more particulary when call 'flash_read' or is it blocked beacause it can't go out of the while loop of 'nvs_recover_last_ate' ?

If my assumption is right it would be "blocked" because it can't do a `flash_read` without generating a NMI. You could test this by not mounting nvs but instead just perform a read of all data in the storage area. If the flash has a problem it should halt. From your data I suspect the problem is in the last sector.

spayet2024, 2024-11-11:
Hi,

You are right, the problem occurs in the last sector. Is there a way to recover of ecc double error when use zephyr? if i understand, this flag is triggered when you read the corrupted flash, but it's too late and the device is blocked.

Laczen, 2024-11-11:
> Hi,
> 
> You are right, the problem occurs in the last sector. Is there a way to recover of ecc double error when use zephyr? if i understand, this flag is triggered when you read the corrupted flash, but it's too late and the device is blocked.

Hi, so you can confirm this problem also occurs when doing only flash_read ? At the moment there is no way to recover from this problem. The flash drivers would need to be modified to support this behavior of a flash device.

closed 2024-12-10 by de-nordic