zephyrproject-rtos/zephyr #80836: Win32 (MSVC/MinGW) Port (equivalent of posix native sim)

Ju1He1, 2024-11-04:
Well the title is pretty self explaining.
Is it possible to provide a basic port with natively runs on windows?
I don't care if it's a MSVC or MinGW port (although MSVC would be preferred as it has much better debugging)

If you are a windows user and you want to debug some tests we are forced to do this in WSL or some docker environment. Although this is working it is a much poorer debugging experience compared to native windows debugging with VS.

Now I'm pretty sure that some of the linux specific stuff can not be ported to windows (especially when it comes to QEmu).
That's why I'm only asking for a **basic** port which includes

- Threads
- Mutexes/Semaphores
- Task Notifications/Eventgroups 
- Timer

Everything beyond that will probably not work on windows.

Now I'm not sure how the posix native sim is implemented, but at some point the K_Thread must be wrapped into a pthread.
Would it be possible instead calling the pthread api directly to use std::thread and std::mutex etc. instead. This way you would not need to differentiate between linux/windows or even mac.
Since zephyr uses cmake under the hood it should be possible to add a minimal win32 board right?

aescolar, 2024-11-04:
@Ju1He1
I'm not a windows person myself, but as far as I understand both visual studio and visual studio code support developing in WSL2:
https://learn.microsoft.com/en-us/cpp/build/walkthrough-build-debug-wsl2?view=msvc-170
https://code.visualstudio.com/docs/remote/wsl
Have you considered that?

Ju1He1, 2024-11-04:
hey @aescolar  thx for your answer.
Yes I'm already using the WSL setup + docker on my windows machine. 
But debugging inside WSL is much slower compared to native windows debugging. Also the debugger provides much less info on variables compared with native MSVC debugging. :)

aescolar, 2024-11-04:
Ok. Unfortunately an architecture port is much more elaborate than just mapping Zephyr threads into host OS threads. That's why there is not even a mac port even though we have a Linux port. There is no current plan for a windows port.
This request can remain open, but given that WSL2 exists and MS has been quite committed to improving it over time, the cost vs benefit of building a native windows solution would seem less likely to justify somebody spending the time.

Ju1He1, 2024-11-04:
can you show me where the actual implementation of posix native sim is? I haven't found it :)