zephyrproject-rtos/zephyr #80850: load mode GPIO bitbang for iCE40 is broken

benediktibk, 2024-11-04:
**Describe the bug**
Trying to load a bitstream into an iCE40 with the GPIO bitbang mode results in a MPU fault.

**To Reproduce**
1. select load mode GPIO bitbang in the devicetree: `load-mode = <1>;`
2. execute fpga_load on this instance
3. MPU fault

    <err> os: ***** MPU FAULT *****
    <err> os:   Data Access Violation
    <err> os:   MMFAR Address: 0x9f85fd3b
    <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x000a0000
    <err> os: r3/a4:  0x000a0000 r12/ip:  0x000a0000 r14/lr:  0x08007d03
    <err> os:  xpsr:  0x41000000
    <err> os: Faulting instruction address (r15/pc): 0x08007cdc
    <err> os: >>> ZEPHYR FATAL ERROR 19: Unknown error on CPU 0
    <err> os: Current thread: 0x20021280 (unknown)
    <err> os: Halting system

The SPI mode is working just fine.

**Expected behavior**
It should be possible to use the GPIO bitbang load mode.

**Impact**
It is not possible to use the iCE40 in combination with low-end microcontrollers.

**Environment:**
- Zephyr: 23fb3b77a2b2cf53eff1697888f443236069b30c
- Hardware: nucleo_f767zi

cfriedt, 2024-11-05:
@benediktibk - can you please provide additional reproduction details? E.g. board, sample, configuration, etc?

Are you attempting to run this from userspace?

benediktibk, 2024-11-05:
main.c:

	/*
	* Copyright (c) 2024 SILA Embedded Solutions GmbH
	*
	* SPDX-License-Identifier: Apache-2.0
	*/

	#include <stdint.h>
	#include <zephyr/drivers/fpga.h>
	#include <zephyr/kernel.h>

	const struct device *ice40 = DEVICE_DT_GET(DT_NODELABEL(ice40));

	uint8_t buffer[104090] __attribute__((aligned(4)));

	int main(void)
	{
		return fpga_load(
			ice40,
			(uint32_t *)buffer,
			ARRAY_SIZE(buffer));
	}

prj.conf:

	CONFIG_COMPILER_OPT="-Wall -Werror"
	CONFIG_DEBUG_OPTIMIZATIONS=y
	CONFIG_LOG=y
	CONFIG_LOG_BUFFER_SIZE=4096
	CONFIG_GPIO=y
	CONFIG_SPI=y
	CONFIG_FPGA=y
	CONFIG_ICE40_FPGA=y

nucleo_f767zi.overlay:

	&spi1 {
		status = "okay";
		pinctrl-0 = <&spi1_sck_pa5>,
			<&spi1_miso_pa6>,
			<&spi1_mosi_pa7>;
		pinctrl-names = "default";
		cs-gpios = <&gpiod 14 GPIO_ACTIVE_LOW>;

		ice40: ice40@0 {
			compatible = "lattice,ice40-fpga";
			status = "okay";
			spi-max-frequency = <1000000>;
			reg = <0x0>;
			creset-gpios = <&gpiob 8 GPIO_ACTIVE_LOW>; // CRESET_B
			cdone-gpios = <&gpiob 9 GPIO_ACTIVE_HIGH>;  // CDONE
			load-mode = <1>;
		};
	};

CMakeLists.txt:

	# SPDX-License-Identifier: Apache-2.0

	cmake_minimum_required(VERSION 3.20.0)

	find_package(Zephyr REQUIRED HINTS $ENV{ZEPHYR_BASE})
	project(ice40_programmer)

	target_sources(app PRIVATE 
		src/main.c)

log output:

	*** Booting Zephyr OS build v4.0.0-rc2-3-g23fb3b77a2b2 ***
	[00:00:00.026,000] <err> os: ***** BUS FAULT *****
	[00:00:00.026,000] <err> os:   Precise data bus error
	[00:00:00.026,000] <err> os:   BFAR Address: 0x1f85fd3b
	[00:00:00.026,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x00000000  r2/a3:  0x000a0000
	[00:00:00.026,000] <err> os: r3/a4:  0x000a0000 r12/ip:  0x000a0000 r14/lr:  0x08007c9b
	[00:00:00.026,000] <err> os:  xpsr:  0x41000000
	[00:00:00.026,000] <err> os: Faulting instruction address (r15/pc): 0x08007c74
	[00:00:00.026,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
	[00:00:00.026,000] <err> os: Current thread: 0x20021280 (unknown)
	[00:00:00.093,000] <err> os: Halting system

cfriedt, 2024-11-05:
What line of code is 0x08007c74? running addr2line should say

benediktibk, 2024-11-05:
I did not intent to go that deep into this bug at this moment, as it is not bothering me. But as an answer to your question I've started the debugger, and this is the call stack:

    z_arm_usage_fault() at fault_s.S:80 0x8002530	
    <signal handler called>() at 0xfffffffd	
    z_impl_gpio_pin_configure() at gpio.h:1,011 0x8007c74	
    gpio_pin_configure() at gpio.h:62 0x8007c9a	
    gpio_pin_configure_dt() at gpio.h:1,031 0x8007c9a	
    fpga_ice40_load_gpio() at fpga_ice40.c:233 0x80033b6	
    fpga_ice40_load() at fpga_ice40.c:443 0x8007dac	
    fpga_load() at fpga.h:111 0x80004ee	
    main() at main.c:11 0x80004ee	

The cause for the fault is therefore the missing configuration of the additional GPIOs, e.g. clk-gpios. As these properties are not required I think a few null-checks would be reasonable.

commit 035b139f (2024-11-07): drivers: fpga: add checks for optional properties of iCE40

Add checks in the GPIO bitbang mode to avoid a fault for missing
configuration in the devicetree.
Fixes #80850
closed 2024-11-07 by mmahadevan108