zephyrproject-rtos/zephyr #80852: Enabling Bluetooth on nRF5340 fails

sjanc, 2024-11-04:
**Describe the bug**
 Call to bt_enable() on nRF5340 fails on bttester with -11 error:

[00:00:02.163,299] <err> bt_hci_core: HCI driver open failed (-11)

**To Reproduce**
Build and flash bttester on nRF5340

**Expected behavior**
Bluetooth is enabled on BTP GAP service registration.

**Impact**
Blocker for running qualification tests on nRF5340

**Logs and console output**
*** Booting Zephyr OS build v4.0.0-rc2 ***
[00:00:00.253,356] <dbg> bttester: tester_init: Initializing tester
[00:00:06.055,450] <dbg> bttester: cmd_handler: cmd service 0 opcode 3 index 255
[00:00:07.055,694] <err> bt_hci_driver: Endpoint binding failed with -11
[00:00:07.055,694] <err> bt_hci_core: HCI driver open failed (-11)
[00:00:07.055,725] <err> bttester_gap: Unable to enable Bluetooth: -11
[00:00:07.055,725] <dbg> bttester: tester_rsp_with_index: service 0 opcode 3 index 255 status 1

First bad commit appears to be 
    samples: Bluetooth: hci_uart hci_ipc configuration update

    Updates to project configuration extras to build with
    Zephyr Controller.

    Configure for 2 broadcast audio sources in 2 groups,
    2 unicast audio source in 1 group and, 2 audio sinks
    for 1 group for both broadcast and unicast usecases.

    Signed-off-by: Vinayak Kariappa Chettimada <vich@nordicsemi.no>

Thalley, 2024-11-04:
@sjanc I'm not seeing issues with flashing the HCI IPC sample to the network core and the peripheral sample to app core. Seems to mainly/only affect the BT tester, at least on 23fb3b77a2b2cf53eff1697888f443236069b30c

Thalley, 2024-11-04:
Just tried with https://github.com/zephyrproject-rtos/zephyr/commit/23fb3b77a2b2cf53eff1697888f443236069b30c with the BT tester built as `-DEXTRA_CONF_FILE=overlay-le-audio.conf` for the nRF5340 ADK and it works fine with the HCI IPC sample. 

This looks like some configuration gone wrong

sjanc, 2024-11-04:
This is build from autopts repository using autopts bot, all configs are from upstream

reverting 23f03dcd84381e14e28c8a4ca1b21933bf14ec97 fix the issue for me

I'm checking which config is causing issues for me

Thalley, 2024-11-04:
> This is build from autopts repository using autopts bot, all configs are from upstream

It is still unclear for other exactly which configs are being used. Please provide the build commands you are using if you want to someone else to fix this issue :)

sjanc, 2024-11-04:
networking core is build with nrf5340_cpunet_iso-bt_ll_sw_split.conf and tests/bluetooth/tester/nrf5340_hci_ipc_cpunet.conf overlays

https://github.com/auto-pts/auto-pts/blob/master/autopts/ptsprojects/boards/nrf53.py

cvinayak, 2024-11-04:
> I'm bit puzzled cause this is now also set from nrf5340_cpunet_iso-bt_ll_sw_split.conf

No, it is commented out in the `samples/bluetooth/hci_ipc` sample:
https://github.com/zephyrproject-rtos/zephyr/blob/23fb3b77a2b2cf53eff1697888f443236069b30c/samples/bluetooth/hci_ipc/nrf5340_cpunet_iso-bt_ll_sw_split.conf#L140-L141

cvinayak, 2024-11-04:
https://github.com/auto-pts/auto-pts/blob/7e33c059df24ccfb843891aec3c19289d2c3cdb1/autopts/ptsprojects/boards/nrf53.py#L47-L49

You have to use CONF_FILE, not OVERLAY_CONFIG.

```
cmd = ['west', 'build', '-b', 'nrf5340dk/nrf5340/cpunet', '--',
           f'-DCONF_FILE=\'nrf5340_cpunet_iso-bt_ll_sw_split.conf '
           f'-DOVERLAY_CONFIG=../../../tests/bluetooth/tester/nrf5340_hci_ipc_cpunet.conf\'']```

sjanc, 2024-11-04:
alright, is this something that changed recently?

cvinayak, 2024-11-04:
No, same for ~2 years now.

https://github.com/zephyrproject-rtos/zephyr/blame/23fb3b77a2b2cf53eff1697888f443236069b30c/samples/bluetooth/hci_ipc/sample.yaml#L78

Thalley, 2024-11-04:
`OVERLAY_CONFIG` was deprecated and replaced with `EXTRA_CONF_FILE` in Zephyr 3.4: https://docs.zephyrproject.org/latest/releases/release-notes-3.4.html#build-system-and-infrastructure

sjanc, 2024-11-05:
it fails for me if I use EXTRA_CONF_FILE too (resulting config is same as with OVERLAY_FILE)

I think the problem is CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y (not sure why it would affect bt_enable() though..)

cvinayak, 2024-11-05:
Could send a PR to auto pts repo, have a sanity run? This way we can review the exact west build command line being used

Thalley, 2024-11-05:
@cvinayak I think this is related to RAM. I tried 3 different builds locally. 

1) CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=n (no changes to the build). This works and uses the following memory: 
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      226844 B       256 KB     86.53%
             RAM:       65492 B        64 KB     99.93%
           SRAM1:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
```

2) CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y (only change to the build). This does not work and uses the following memory: 
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      227020 B       256 KB     86.60%
             RAM:       65524 B        64 KB     99.98%
           SRAM1:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
```

3) CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y and CONFIG_BT_MAX_CONN=2 (only changes to the build). This works and uses the following memory
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      227084 B       256 KB     86.63%
             RAM:       63732 B        64 KB     97.25%
           SRAM1:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
```

4) CONFIG_BT_CTLR_ISOAL_PSN_IGNORE=y and CONFIG_BT_CTLR_CONN_ISO_STREAMS=1 (only changes to the build). This works and uses the following memory
```
Memory region         Used Size  Region Size  %age Used
           FLASH:      226796 B       256 KB     86.52%
             RAM:       65276 B        64 KB     99.60%
           SRAM1:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
```

I wonder if the RAM calculations are incorrect, and we are actually over budget in 2) or what is going on.

cvinayak, 2024-11-05:
This is a possibility, I will test running our audio application with full RAM used (have dummy array) tomorrow and let you know.

nordic-piks, 2024-11-07:
Triaged internally

jhedberg, 2024-11-08:
> Triaged internally

What does this mean?

closed 2024-11-08 by mmahadevan108

Thalley, 2024-11-08:
Might be worth considering that https://github.com/zephyrproject-rtos/zephyr/pull/80990 is only really a workaround and not a proper solution to the bug here.

PR #80990 (2024-11-08): tests: Bluetooth: tester: Workaround Bus Fault in nRF53x using full RAM
Workaround failing autopts weekly run due to tester application having Bus Fault in the nRF53 NET core hci_ipc firmware at startup.

Having atleast 32 bytes free in nRF53x hci_ipc sample to avoid Bus Fault when Zephyr Kernel does SYSINIT calls.

Reduce supported ISO Broadcaster instance and ISO TX buffer counts in the Controller hci_ipc firmware.

Add sysbuild support for tester application so that the build is CI verified.

sysbuild commandline for LE audio PTS testing:

```
  # west build --sysbuild -b nrf5340_audio_dk/n [... cut]