zephyrproject-rtos/zephyr #80857: Initialisation timeout always generated when initialising the Quectel Lx6G GNSS modem

j-delsol-dev, 2024-11-04:
**Bug description**
How does the zephyr chat modem work?
I use the Qectuel modem with interrupts enabled on the uart. The module starts up fine but goes into timeout during initialization even though I am receiving navigation data in the callbacks (latitude, longitude...).

henrikbrixandersen, 2024-11-04:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

bjarki-andreasen, 2024-11-05:
Are you using a newer version of zephyr? there was a bug which caused timeouts due to utilizing spinlocks (which disable interrupts) for locking which was changed to semaphores, which could describe your issue :)

bjarki-andreasen, 2024-11-05:
https://github.com/zephyrproject-rtos/zephyr/pull/72242 https://github.com/zephyrproject-rtos/zephyr/pull/72246

j-delsol-dev, 2024-11-05:
Thank you.
I use Zephyr 3.7 
In drivers, the semaphore is used (gnss_publish.c).

j-delsol-dev, 2024-11-05:
The problem stems from the compatibility of the controls between the LCx6 and Lx6 series. We can't use the generic Quectel drivers with the Lx6 series.
Thank you for your support!