zephyrproject-rtos/zephyr #80861: BAP Unicast Client only discovers 1 of each type of PAC record

Chris-StJohn-Bose, 2024-11-04:
https://github.com/zephyrproject-rtos/zephyr/blob/23fb3b77a2b2cf53eff1697888f443236069b30c/subsys/bluetooth/audio/bap_unicast_client.c#L4405

PACS servers may have multiple PAC characteristic of each type source/sink:

> **PACS v1.0.1 Section 3.1**
> (..)
> When the server supports reception of audio data, the decision to expose all supported PAC records
> within a single Sink PAC characteristic or multiple Sink PAC characteristics is left to the implementation.
> 

`unicast_client_pac_discover_cb()` always returns `BT_GATT_ITER_STOP` so only the first one is found.

Thalley, 2024-11-05:
This looks like a duplicate of https://github.com/zephyrproject-rtos/zephyr/issues/70826. Can you confirm @Chris-StJohn-Bose ?

Chris-StJohn-Bose, 2024-11-06:
> This looks like a duplicate of #70826. Can you confirm @Chris-StJohn-Bose ?

Yes I think this is a duplicate @Thalley

Thalley, 2024-11-06:
Closing as duplicate then :)

closed 2024-11-06 by Thalley

Chris-StJohn-Bose, 2024-11-07:
thanks @Thalley I'll submit a PR if I get around to fixing it and tag #70826

Thalley, 2024-11-07:
Sounds good. There are also some planned effort from @jthm-ot to make make the PACS client usable for the broadcast assistant as well, so just be aware that the PACS client will likely need to be moved out from the BAP unicast client and into a standalone client usable from both BAP Unicast Client and BAP Broadcast Assistant (or have it's own public API). It does not prevent the above from being added, but just be aware of any changes that may cause merge conflicts