zephyrproject-rtos/zephyr #80885: PLL-I2S clock misconfiguration on STM32F446

arthurbabu, 2024-11-04:
**Describe the bug**

I'm using a custom board with an STM32F446 inside and a 24MHz crystal oscillator for HSE. I want to generate a 25MHz via MCO2 output. To do so, i want to use pll-i2s clock with this configuration:
- DIV_M = 12
- MULT_N = 100
- DIV_R = 2
This creates a 100MHz output. With MCO's clock divider = 4 it must generate a 25MHz clock.
Instead i have a 36Mhz output. I've took a look on RCC config register and values are not consistent with dts configuration file. Register values are:
- DIV_M = 16
- MULT_N = 192
- DIV_R = 2
Those values are consistent with experimental 36MHz clock. Furthermore, MCO_2 pin slew_rate is not set to very_high.

**To Reproduce**

DTS file:

```dts
 /dts-v1/;
 #include <st/f4/stm32f446Xe.dtsi>
 #include <st/f4/stm32f446r(c-e)tx-pinctrl.dtsi>
 #include <zephyr/dt-bindings/input/input-event-codes.h>
 
 / {
	 model = "Custom board test";
	 compatible = "st,stm32f446re-nucleo";
 
	 chosen {
		 zephyr,console = &uart4;
		 zephyr,shell-uart = &uart4;
		 zephyr,sram = &sram0;
		 zephyr,flash = &flash0;
	 };
 
	 leds: leds {
		compatible = "gpio-leds";
		green_led: led_0 {
			gpios = <&gpioa 9 GPIO_ACTIVE_HIGH>;
			label = "User LD1";
		};
		en_vusb: led_1 {
			gpios = <&gpioc 3 GPIO_ACTIVE_HIGH>;
			label = "Enable USB alim";
		};
		en_vana: led_2 {
			gpios = <&gpioc 13 GPIO_ACTIVE_HIGH>;
			label = "Enable Analog alim";
		};
	};

	aliases {
		led0 = &green_led;
		led1 = &en_vusb;
		led2 = &en_vana;
	};
 };
 
 &clk_lsi {
	 status = "okay";
 };
 
 &clk_hse {
	clock-frequency = <DT_FREQ_M(24)>; /* 24MHz clock */
	status = "okay";
 };
 
 &pll {
	 div-m = <12>;
	 mul-n = <96>;
	 div-p = <2>;
	 div-q = <4>;
	 clocks = <&clk_hse>;
	 status = "okay";
 };

 &plli2s {
	mul-n = <100>;
	div-r = <2>;
	clocks = <&clk_hse>;
	status = "okay";
};
 
 &rcc {
	 clocks = <&pll>;
	 clock-frequency = <DT_FREQ_M(96)>;
	 ahb-prescaler = <1>;
	 apb1-prescaler = <2>;
	 apb2-prescaler = <1>;
 };
 
 &usart2 {
	 pinctrl-0 = <&usart2_tx_pa2 &usart2_rx_pa3>;
	 pinctrl-names = "default";
	 current-speed = <460800>;
	 status = "okay";
 };

 &uart4 {
	pinctrl-0 = <&uart4_tx_pa0 &uart4_rx_pa1>;
	pinctrl-names = "default";
	current-speed = <460800>;
	status = "okay";
};

&i2s1 {
	pinctrl-0 = <&i2s1_ck_pa5 &i2s1_sd_pa7 &i2s1_ws_pa4>;
	// clocks = <&rcc STM32_SRC_PLL2_R I2S_SEL(0)>;
	pinctrl-names = "default";
	status = "okay";
};

&i2s2 {
	pinctrl-0 = <&i2s2_ck_pb13 &i2s2_sd_pb15 &i2s2_ws_pb12>;
	// clocks = <&rcc STM32_SRC_PLL2_R I2S_SEL(0)>;
	pinctrl-names = "default";
	status = "okay";
};

&mco2 {
	clocks = <&rcc STM32_SRC_PLLI2S_R MCO2_SEL(1)>;
	prescaler = <MCO2_PRE(6)>;
	pinctrl-0 = <&rcc_mco_2_pc9>;
	pinctrl-names = "default";
	status = "okay";
};

 zephyr_udc0: &usbotg_fs {
	pinctrl-0 = <&usb_otg_fs_dm_pa11 &usb_otg_fs_dp_pa12>;
	pinctrl-names = "default";
	maximum-speed = "full-speed";
	phys = <&otgfs_phy>;
	status = "okay";
};

&sdmmc1 {
	compatible = "st,stm32-sdmmc";
	pinctrl-0 = <&sdio_d0_pc8 &sdio_d1_pb0
				&sdio_d2_pb1 &sdio_d3_pc11
				&sdio_d4_pb8 &sdio_d5_pb9
				&sdio_d6_pc6 &sdio_d7_pc7
				&sdio_ck_pb2 &sdio_cmd_pd2>;
	// resets = <&gpiog 6 GPIO_ACTIVE_LOW>;
	// reset-names = "default";
	clk-div = <16>;
	pinctrl-names = "default";
	bus-width = <8>;
	status = "okay";
};
 
 &backup_sram {
	 status = "okay";
 };
 
 &flash0 {
 
	 partitions {
		 compatible = "fixed-partitions";
		 #address-cells = <1>;
		 #size-cells = <1>;
 
		 boot_partition: partition@0 {
			 label = "mcuboot";
			 reg = <0x00000000 DT_SIZE_K(64)>;
			 read-only;
		 };
 
		 /*
		  * The flash starting at 0x00010000 and ending at
		  * 0x0001ffff (sectors 16-31) is reserved for use
		  * by the application.
		  */
		 storage_partition: partition@10000 {
			 label = "storage";
			 reg = <0x00010000 DT_SIZE_K(64)>;
		 };
 
		 slot0_partition: partition@20000 {
			 label = "image-0";
			 reg = <0x00020000 DT_SIZE_K(128)>;
		 };
		 slot1_partition: partition@40000 {
			 label = "image-1";
			 reg = <0x00040000 DT_SIZE_K(128)>;
		 };
		 scratch_partition: partition@60000 {
			 label = "image-scratch";
			 reg = <0x00060000 DT_SIZE_K(128)>;
		 };
	 };
 };
```

**Expected behavior**

Expected behavior is a 25MHz clock on MCO2 pin.

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

 - OS:  Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: Zephyr OS build v3.7.0-4773-g2f23313e37e4

**Additional context**

arthurbabu, 2024-11-04:
I found that this [definition of pll-i2s clock](https://github.com/zephyrproject-rtos/zephyr/blob/4bb945300bf92e4708404102d2a3b54e540bfd6c/dts/arm/st/f4/stm32f446.dtsi#L12-L16) is redundant with [this one ](https://github.com/zephyrproject-rtos/zephyr/blob/4bb945300bf92e4708404102d2a3b54e540bfd6c/dts/arm/st/f4/stm32f401.dtsi#L10-L16)

Removing the first definition on dts/arm/st/f4/stm32f446.dtsi file seems to solve this issue.
For slew_rate does it exist a way to set it into board's dts file ?

FRASTM, 2024-11-05:
In the pinctrl [stm32f446r(c-e)tx-pinctrl.dtsi](https://github.com/zephyrproject-rtos/hal_stm32/blob/main/dts/st/f4/stm32f446r(c-e)tx-pinctrl.dtsi) file, the mco2 pin is defined as very high-speed:
```
			/omit-if-no-ref/ rcc_mco_2_pc9: rcc_mco_2_pc9 {
				pinmux = <STM32_PINMUX('C', 9, AF0)>;
				slew-rate = "very-high-speed";
			};
```

erwango, 2024-11-05:
> I found that this [definition of pll-i2s clock](https://github.com/zephyrproject-rtos/zephyr/blob/4bb945300bf92e4708404102d2a3b54e540bfd6c/dts/arm/st/f4/stm32f446.dtsi#L12-L16) is redundant with [...]

@gautierg-st Should be able to help verify this code is actually correct or not.

arthurbabu, 2024-11-05:
> In the pinctrl [stm32f446r(c-e)tx-pinctrl.dtsi](https://github.com/zephyrproject-rtos/hal_stm32/blob/main/dts/st/f4/stm32f446r(c-e)tx-pinctrl.dtsi) file, the mco2 pin is defined as very high-speed:
> 
> ```
> 			/omit-if-no-ref/ rcc_mco_2_pc9: rcc_mco_2_pc9 {
> 				pinmux = <STM32_PINMUX('C', 9, AF0)>;
> 				slew-rate = "very-high-speed";
> 			};
> ```
> 
> If you do not see the slew-rate correctly defined, could you please try with this change :
> 
> ```diff
> diff --git a/soc/st/stm32/stm32f4x/soc.c b/soc/st/stm32/stm32f4x/soc.c
> index 941f0091997..d2a9f30acee 100644
> --- a/soc/st/stm32/stm32f4x/soc.c
> +++ b/soc/st/stm32/stm32f4x/soc.c
> @@ -33,6 +33,8 @@ void soc_early_init_hook(void)
>  	/* Update CMSIS SystemCoreClock variable (HCLK) */
>  	/* At reset, system core clock is set to 16 MHz from HSI */
>  	SystemCoreClock = 16000000;
> +
> +	LL_APB2_GRP1_EnableClock(LL_APB2_GRP1_PERIPH_SYSCFG);
>  #if CONFIG_PM
>  	stm32_power_init();
>  #endif
> ```

My zephyr repo wasn't up to date. Now MCO pin is in high_speed slew rate.

arthurbabu, 2024-11-12:
> @gautierg-st Should be able to help verify this code is actually correct or not.

@FRASTM  no, PLL-I2S clock configuration issue is still present.

FRASTM, 2024-11-13:
the stm32F446 PLL for the I2S is of type `compatible = "st,stm32f412-plli2s-clock"` : Stm32F446 has the same PLLI2S scheme as the stm32F412
It must have a **div-m** property [2..63] divider on the HSI or HSE PLL clock source.

[image]

FRASTM, 2024-11-14:
@arthurbabu, 
 once this pllI2s is defined by the DTS, in the &mco2 definition above, 
prescaler = <MCO2_PRE(6)>; is not a vaild value as the prescaler is in range 1..5 (clock_control\Kconfig.stm32)
Can you try with `prescaler = <2>; `  in your DTS for the mco2 node

arthurbabu, 2024-11-14:
I've taken a look at STM32F446's reference manual. Prescaler is not directly the clock divider value:
[image]
prescaler = <MCO2_PRE(6)>; writes 6 into RCC->RCC_CFGR->MCO2PRE (I've verified this with ST-Link debugging probe to see registers values) equals division by 4.
`prescaler = <2>;` doesn't set RCC->RCC_CFGR->MCO2PRE register (I see 0 while debugging)

FRASTM, 2024-11-14:
investigating, seems that CONFIG_CLOCK_STM32_MCO2_SRC_PLLI2S=y is defined through the Kconfig even if deprecated

FRASTM, 2024-11-18:
I put a PR https://github.com/zephyrproject-rtos/zephyr/pull/81460  that you could test, which can
-  configure the PLLi2S 
-  set the MCO by the DTS
-  remove the Kconfig

closed 2024-12-06 by nashif