zephyrproject-rtos/zephyr #80915: mutex_api_error testsuite fails with CONFIG_USERSPACE=n

RobinKastberg, 2024-11-05:
**Describe the bug**

Running twister tests on nrf52840dk with the parameter `CONFIG_USERSPACE=n` fails.

**To Reproduce**
```
user@2910b946914f:/workdir/zephyr$ scripts/twister -p nrf52840dk/nrf52840 -T tests/kernel/mutex/mutex_error_case/ -x CONFIG_USERSPACE=n --hardware-map mapnrf.yml --device-testing --verbose
INFO    - 1/1 nrf52840dk/nrf52840       tests/kernel/mutex/mutex_error_case/kernel.mutex.error  FAILED Failed (device: 001050277191, 2.720s)
ERROR   - see: /workdir/zephyr/twister-out/nrf52840dk_nrf52840/tests/kernel/mutex/mutex_error_case/kernel.mutex.error/handler.log
```

**Expected behavior**
Tests pass or get filtered.
When I debugged the code, it seemed like getting errors from the mutex call relies on `CONFIG_USERSPACE=y`, and I don't expect the test to be easily salvageable to a `CONFIG_USERSPACE=n` situation, and therefore I proposed filtering the test in https://github.com/zephyrproject-rtos/zephyr/pull/80851.

**Impact**
This is annoying when trying to get test coverage.

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-4912-gc54e26b758d6 ***
Running TESTSUITE mutex_api_error
===================================================================
START - test_mutex_init_invalid_obj
current case is 1
 FAIL - test_mutex_init_invalid_obj in 0.002 seconds
===================================================================
START - test_mutex_init_null
current case is 0
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x8
E: r0/a1:  0x00000000  r1/a2:  0x00007207  r2/a3:  0x200000b0
E: r3/a4:  0x00000000 r12/ip:  0x20000524 r14/lr:  0x000001e3
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x00006b56
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200000b0 (unknown)
Caught system error -- reason 0 1
Fatal error expected as part of test case.

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mutex/mutex_error_case/src/test_mutex_error.c:45: ztest_post_fatal_error_hook: reason not equal to K_ERR_KERNEL_OOPS

 FAIL - test_mutex_init_null in 0.057 seconds
===================================================================
START - test_mutex_lock_invalid_obj
current case is 3
 FAIL - test_mutex_lock_invalid_obj in 0.002 seconds
===================================================================
START - test_mutex_lock_null
current case is 2
 FAIL - test_mutex_lock_null in 0.002 seconds
===================================================================
START - test_mutex_unlock_invalid_obj
current case is 5
 FAIL - test_mutex_unlock_invalid_obj in 0.002 seconds
===================================================================
START - test_mutex_unlock_null
current case is 4
 FAIL - test_mutex_unlock_null in 0.002 seconds
===================================================================
TESTSUITE mutex_api_error failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -   0.00% [mutex_api_error]: pass = 0, fail = 6, skip = 0, total = 6 duration = 0.067 seconds
 - FAIL - [mutex_api_error.test_mutex_init_invalid_obj] duration = 0.002 seconds
 - FAIL - [mutex_api_error.test_mutex_init_null] duration = 0.057 seconds
 - FAIL - [mutex_api_error.test_mutex_lock_invalid_obj] duration = 0.002 seconds
 - FAIL - [mutex_api_error.test_mutex_lock_null] duration = 0.002 seconds
 - FAIL - [mutex_api_error.test_mutex_unlock_invalid_obj] duration = 0.002 seconds
 - FAIL - [mutex_api_error.test_mutex_unlock_null] duration = 0.002 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: cab5567bc0bbd99039ef68892e728ef3
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**

WSL docker container with zephyr dev container
zephyr: 23fb3b77a2b2cf53eff1697888f443236069b30c

RobinKastberg, 2024-11-05:
test also fails on mimxrt1060_evk and I don't think is platform or hardware related.

cfriedt, 2024-11-05:
Yeah, just looking at `testcase.yaml` and `prj.conf`, this is tagged to also run userspace tests, but there is no test configuration that even enables userspace.

It's a bit confusing especially because faults are ignored here as well.

Ideally we would be validating that faults do occur when userspace is configured and an MPU / MMU is present.

cfriedt, 2024-11-05:
> @cfriedt I guess you were looking at the `mutex_api` test, while this issue is using `mutex_error_case`. I am amending the title to reflect that.

Ah, sorry - I was looking for `CONFIG_USERSPACE`, but should have noticed `CONFIG_TEST_USERSPACE`.

Makes sense

nordic-piks, 2024-11-07:
Triaged internally

RobinKastberg, 2024-11-07:
@nordic-piks thanks I appreciate all the help. But just to be clear I don't think this is a problem with Nordic hardware or software in any way. Just trying to get the test filtered out and I have a PR for it.

dcpleung, 2025-01-07:
As iterated in https://github.com/zephyrproject-rtos/zephyr/pull/80851 that the test should never be run without userspace enabled. What described above is not a supported configuration, and thus is not a bug. Closing...

closed 2025-01-07 by dcpleung