zephyrproject-rtos/zephyr #80932: zephyr building TF-M gives Segmentation fault on b_u585i_iot02a/stm32u585xx/ns

FRASTM, 2024-11-05:
Zephyr CI testing reports a build error b_u585i_iot02a/stm32u585xx/ns tests/benchmarks/sched_queues 
when building the tests/benchmarks/sched_queues on the stm32u585 disco kit
(during test  campaign of the https://github.com/zephyrproject-rtos/zephyr/pull/79797)
whatever the option is -DCONFIG_SCHED_DUMB=y  or -DCONFIG_SCHED_SCALABLE=y

```

FAILED: platform/CMakeFiles/platform_s.dir/ext/target/stm/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.o 
...
/__w/zephyr/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/stm/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ReplaceNode_Head':
/__w/zephyr/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/stm/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.c:4719:1: internal compiler error: Segmentation fault
 4719 | }
      | ^
```

 Zephyr version: 4.0.0-rc2 
 toolchain: zephyr 0.16.8 
CMake version: 3.22.1

trusted-firmware-m : 8134106ef9cb3df60e8bd22b172532558e936bd2 or a11cd27905aecc4416cfc85552bfc3b997375056

**To Reproduce**

**Expected behavior**

**Impact**
CI failure 

**Logs and console output**
```
FAILED: platform/CMakeFiles/platform_s.dir/ext/target/stm/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.o 
...
~/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/stm/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ReplaceNode_Head':
~/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/stm/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.c:4719:1: internal compiler error: Segmentation fault
 4719 | }
      | ^
0x7b24d824251f ???
	./signal/../sysdeps/unix/sysv/linux/x86_64/libc_sigaction.c:0
0x7b24d8229d8f __libc_start_call_main
	../sysdeps/nptl/libc_start_call_main.h:58
0x7b24d8229e3f __libc_start_main_impl
	
```

mathieuchopstm, 2024-11-05:
This is missing one line of context:
```
during GIMPLE pass: evrp
~/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/stm/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ReplaceNode_Head':
~/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/stm/common/stm32u5xx/hal/Src/stm32u5xx_hal_dma_ex.c:4719:1: internal compiler error: Segmentation fault
 4719 | }
      | ^
0x72996f44251f ???
        ./signal/../sysdeps/unix/sysv/linux/x86_64/libc_sigaction.c:0
0x72996f429d8f __libc_start_call_main
        ../sysdeps/nptl/libc_start_call_main.h:58
0x72996f429e3f __libc_start_main_impl
        ../csu/libc-start.c:392
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://github.com/zephyrproject-rtos/sdk-ng/issues> for instructions.
```

This error is actually a red herring, the true problem seems related to `DMA_List_CheckNodesBaseAddresses`:
* adding `__attribute__((no_inline))` to `DMA_List_CheckNodesBaseAddresses` allows the build to pass
* building at `-O1` allows the build to pass
  * one may think that the `tree-evrp` pass being disabled at `-O1` is the reason...
  * ...but `DMA_List_CheckNodesBaseAddresses` is actually not inlined at `-O1`
    * (this can be seen by dumping tree using `-fdump-tree-fre1` - AFAIU, `fre1` is the pass right before `evrp`)
* __using GCC 10.3.0 (i.e., building with Zephyr SDK 0.14.2) allows the build to pass__
* __using GCC 12.1.0 (i.e., building with Zephyr SDK 0.15.0) crashes the same way as 12.2.0__

mathieuchopstm, 2024-11-06:
The following can be used to reproduce the issue:
<details>
<summary>Minimal reproduction code derived from HAL</summary>

```c

#include <stddef.h>
#include <stdint.h>

/**
 * Any value other than 0 and 0xFFFFFFFF causes ICE.
 */
#define MASK (0xA5A5A5A5)

/**
 * Not marking the function as static prevents it from being inlined.
 * Without inlining, the ICE doesn't seem to trigger.
 */
static uint32_t callee(void *a1, void *a2) {
        uint32_t temp = (((uint32_t)a1 | (uint32_t)a2) & MASK);
        uint32_t ref = 0U;

        if (a1 != NULL) {
                ref = (uint32_t)a1;
        } else if (a2 != NULL) {
                ref = (uint32_t)a2;
        }
        /* else: ref = 0U */

        if (temp != (ref & MASK)) {
        /* Removing `& MASK` here prevents ICE */
                return 1U;
        }

        return 0U;
}

int caller(void *a1, void *a2) {
        if ((a1 == NULL) || (a2 == NULL)) {
                return 0;
        }

        if (callee(a1, a2) != 0U) {
                return 1;
        }

        return 0;
}
```

</details>

This code builds with GCC 10.5 and 13.1, but hangs with any version of GCC 11 and 12: see https://godbolt.org/z/hcv7zhhc3
Note: using `-O` merely makes the segfault happen during `dom` pass instead of `evrp`.

The following GCC ticket might be related: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=106878 (found thanks to symbols in x86-64 GCC used on Compiler Explorer)

erwango, 2024-11-06:
Summary: 

Issue linked to compilation of file `HAL_DMAEx_List_ReplaceNode_Head.c` using a GCC version comprised between v11.x and v12.x.
Issue is seen only with TF-M build config `RelWithDebInfo` (With `MinSizeRel` config, this file is not built).

closed 2025-02-04 by kartben